espressif/esp-idf #9719: Mqtt function "esp_mqtt_client_enqueue()" sending no data when parameter 'len' = 0 for auto size calculation

Xtensa2C, 2022-09-05:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

**IDF version.**
v4.4.2

**Operating System used.**
Windows

**How did you build your project?**
VSC  Extension

**If you are using Windows, please specify command line type.**
No response

**What is the expected behavior?**
Function "esp_mqtt_client_enqueue()" automatically calculating the payload size if parameter 'len' is set to '0'

**What is the actual behavior?**

Function "esp_mqtt_client_enqueue()" dispatching messages with 0 byte payload size if parameter 'len' is set to '0'.
 Setting the 'len' parameter to the right payload size circumvents this issue.

**Steps to reproduce.**

Take the basic MQTT websocket example and replace "esp_mqtt_client_publish()" with "esp_mqtt_client_enqueue()" and set the 'len' parameter to '0'.

**Code snippet:**
```
esp_mqtt_client_enqueue(client, "/topic/enqueue_auto", "enqueue", 0, 1, 0,true);
esp_mqtt_client_publish(client, "/topic/publish_auto", "publish", 0, 1, 0);
esp_mqtt_client_enqueue(client, "/topic/enqueue_man", "enqueue", 7, 1, 0,true);
esp_mqtt_client_publish(client, "/topic/publish_man", "publish", 7, 1, 0);
```

**Mosquitto log:**

```
1662403409: Received PUBLISH from ESP (d0, q1, r0, m14871, '/topic/publish_auto', ... (7 bytes))
1662403409: Sending PUBACK to ESP (m14871, rc0)
1662403409: Received PUBLISH from ESP (d0, q1, r0, m6288, '/topic/publish_man', ... (7 bytes))
1662403409: Sending PUBACK to ESP (m6288, rc0)
1662403409: Received PUBLISH from ESP (d0, q1, r0, m2615, '/topic/enqueue_auto', ... (0 bytes))
1662403409: Sending PUBACK to ESP (m2615, rc0)
1662403409: Received PUBLISH from ESP (d0, q1, r0, m18074, '/topic/enqueue_man', ... (7 bytes))
1662403409: Sending PUBACK to ESP (m18074, rc0)
```

**Possible Fix:**
Simple fix is to add the automatic length calculation to function "esp_mqtt_client_enqueue()"  as already done in "esp_mqtt_client_publish()" 

```
int esp_mqtt_client_enqueue(esp_mqtt_client_handle_t client, const char *topic, const char *data, int len, int qos, int retain, bool store)
{
    if (!client) {
        ESP_LOGE(TAG, "Client was not initialized");
        return -1;
    }
    MQTT_API_LOCK(client);

   /* Acceptable publish messages:
        data == NULL, len == 0: publish null message
        data valid,   len == 0: publish all data, payload len is determined from string length
        data valid,   len >  0: publish data with defined length
     */
    if (len <= 0 && data != NULL) {
        len = strlen(data);
    }

    int ret = mqtt_client_enqueue_priv(client, topic, data, len, qos, retain, store);
    MQTT_API_UNLOCK(client);
    if (ret == 0 && store == false) {
        // messages with qos=0 are not enqueued if not overridden by store_in_outobx -> indicate as error
        return -1;
    }
    return ret;
}
```

AxelLin, 2022-09-06:
https://www.esp32.com/viewtopic.php?f=13&t=29633

euripedesrocha, 2022-09-12:
Hi @Xtensa2C, thanks for reporting it. It was fixed in latest version. We'll back port it to 4.4.

Alvin1Zhang, 2022-12-06:
Thanks for reporting, sorry for slow turnaround, fix on release/4.4 is available at https://github.com/espressif/esp-idf/commit/9aa7c83e19193cfd54b0fdf713b0de4b5d0f997d, feel free to reopen.

closed 2022-12-06 by Alvin1Zhang