espressif/esp-idf #9722: Encrypt a library

francis2tm, 2022-09-06:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I'm developing a product when it is expected users to flash via UART their custom firmware. However, I want to provide a proprietary library that is encrypted and it can only be used in the hardware that I sold. Is this achievable? and if so, would the library be vulnerable?

Alvin1Zhang, 2022-09-09:
Thanks for reporting, please help create a post on forum https://esp32.com. Thanks.

closed 2022-09-09 by Alvin1Zhang

francis2tm, 2022-09-09:
Hello @Alvin1Zhang ,
I posted here because of my 4 different posts in the forum, none of them were ever answered..
I'm considering building a product with ESP32, knowing if I can encrypt a library with a key fused in the eFuse is really important.

mahavirj, 2022-09-12:
@francis2tm 

>  I posted here because of my 4 different posts in the forum, none of them were ever answered..

Can you please provide the pointers? I will check and reply

>  I'm considering building a product with ESP32, knowing if I can encrypt a library with a key fused in the eFuse is really important.

Yes, this should be possible. You can use `EFUSE_BLK3` to store custom credentials like application specific encryption key.

francis2tm, 2022-09-12:
Hey @mahavirj, thanks for the reply
> Yes, this should be possible. You can use EFUSE_BLK3 to store custom credentials like application specific encryption key.
And a user cannot read the encryption key with via uart?
How would the Decryption work in this case? Is there an example I can use as a baseline?

mahavirj, 2022-09-13:
> And a user cannot read the encryption key with via uart?

You may enable secure ROM DL mode or can entirely disable download mode as well https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/kconfig.html#config-secure-uart-rom-dl-mode.

>  How would the Decryption work in this case? Is there an example I can use as a baseline?

This could be standard symmetric encryption scheme (e.g., AES GCM). Please refer to mbedTLS APIs (and samples) for this.