espressif/esp-idf #9724: Build failure when both CONFIG_ESP_TASK_WDT and CONFIG_FREERTOS_ENABLE_TASK_SNAPSHOT are disabled

tangzz98, 2022-09-06:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

master

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### What is the expected behavior?

Build should succeed when both CONFIG_ESP_TASK_WDT and CONFIG_FREERTOS_ENABLE_TASK_SNAPSHOT are disabled

### What is the actual behavior?

Build failed

### Steps to reproduce.

1. idf.py menuconfig
2. Disable Component config -> ESP System Settings -> Initialize Task Watchdog Timer on startup
3. Disable Component config -> FreeRTOS -> Port -> Enable task snapshot functions
4. idf.py build

### Build or installation Logs.

```plain
/Users/tangyi/esp-idf/components/esp_system/task_wdt.c:430:5: error: unknown type name 'TaskSnapshot_t'
  430 |     TaskSnapshot_t snapshot = { 0 };
      |     ^~~~~~~~~~~~~~
/Users/tangyi/esp-idf/components/esp_system/task_wdt.c:435:11: error: implicit declaration of function 'vTaskGetSnapshot'; did you mean 'vTaskGetInfo'? [-Werror=implicit-function-declaration]
  435 |     ret = vTaskGetSnapshot(xTaskGetCurrentTaskHandle(), &snapshot);
      |           ^~~~~~~~~~~~~~~~
      |           vTaskGetInfo
/Users/tangyi/esp-idf/components/esp_system/task_wdt.c:449:36: error: request for member 'pxTopOfStack' in something not a structure or union
  449 |     void *frame = (void *) snapshot.pxTopOfStack;
      |                                    ^
cc1: some warnings being treated as errors
ninja: build stopped: subcommand failed.
HINT: The header file task_snapshot.h is no longer included as part of freertos/task.h. Users need to include freertos/task_snapshot.h explicitly.
HINT: The header file task_snapshot.h is no longer included as part of freertos/task.h. Users need to include freertos/task_snapshot.h explicitly.
ninja failed with exit code 1, output of the command is in the /Users/tangyi/esp-idf/examples/get-started/hello_world/build/log/idf_py_stderr_output_77816 and /Users/tangyi/esp-idf/examples/get-started/hello_world/build/log/idf_py_stdout_output_77816
```

AxelLin, 2023-04-21:
Cannot reproduce it with current master tree, maybe this is already fixed?

ESP-Marius, 2023-04-21:
Seems like building `task_wdt.c` now depends on `CONFIG_ESP_TASK_WDT_EN` so this should no longer be an issue neither on master nor 5.0 (fixed after 53c7dd4efc619342ddebd3c938547044506bef84)

Thanks for following up on this one @AxelLin, seems like this can be closed.

closed 2023-04-21 by ESP-Marius