espressif/esp-idf #9725: esp_wifi_scan_start returns undocumented values

ashesman, 2022-09-06:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

V4.4.2 

### Operating System used.

Windows

### How did you build your project?

Eclipse IDE

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

ESP32C3 MINI

### Power Supply used.

USB

### What is the expected behavior?

When esp_wifi_scan_start is called while ESP32 is trying to connect to a station, esp_wifi_scan_start should return ESP_ERR_WIFI_STATE.

As per function documentation:  https://docs.espressif.com/projects/esp-idf/en/v4.4.2/esp32/api-reference/network/esp_wifi.html

### What is the actual behavior?

esp_wifi_scan_start returns -1 (ESP_FAIL).  esp_wifi_scan_start emits the follwoing log message during execution, so is aware of its failure due to currently connecting, but still returns the wrong failure type.

"W (40794) wifi:sta_scan: STA is connecting, scan are not allowed!"

### Steps to reproduce.

Setup wifi as a station, set incorrect AP SSID and call esp_wifi_connect().  The station will never connect to AP. 

In WIFI_EVENT_STA_DISCONNECTED event, call esp_wifi_connect() again to keep repeatedly trying to connect.

At some arbitrary time, call esp_wifi_scan_start and observe the result code

wifi_scan_config_t scanConf = {
   .ssid = NULL,
   .bssid = NULL,
   .channel = 0,
   .show_hidden = true
};
esp_err_t err = esp_wifi_scan_start(&scanConf, false);

### Debug Logs.

```plain
I (40444) wifi:mode : sta (a0:76:4e:6c:6a:44)
I (40444) wifi:enable tsf
I (40444) wifi:set country: cc=NZ schan=1 nchan=13 policy=0
I (40444) phy: chan:1,max_power:80
I (40444) phy: chan:2,max_power:80
I (40444) phy: chan:3,max_power:80
I (40444) phy: chan:4,max_power:80
I (40444) phy: chan:5,max_power:80
I (40444) phy: chan:6,max_power:80
I (40444) phy: chan:7,max_power:80
I (40444) phy: chan:8,max_power:80
I (40444) phy: chan:9,max_power:80
I (40444) phy: chan:10,max_power:80
I (40444) phy: chan:11,max_power:80
I (40444) phy: chan:12,max_power:80
I (40444) phy: chan:13,max_power:80
I (40444) phy: chan:14,max_power:80
I (40444) wifi:new:<11,0>, old:<1,0>, ap:<255,255>, sta:<11,0>, prof:1
I (40454) wifi:state: init -> auth (b0)
W (40794) wifi:sta_scan: STA is connecting, scan are not allowed!
```

### More Information.

Because we have no way of detecting the current WiFi state, it is hard to know if we are currently connecting or not.  The correct return value of esp_wifi_scan_start could help us in determining how to react.

esp_err_t err = esp_wifi_scan_start(&cfg, false);
if (err == ESP_OK)
{
  return true;
}
else if (err == ESP_ERR_WIFI_STATE)
{
  // Currently connecting, stop connecting
  esp_wifi_disconnect();
  return esp_wifi_scan_start(&cfg, false) == ESP_OK;
}
return false;

zhangyanjiaoesp, 2022-09-07:
@ashesman We will update the return values

KonssnoK, 2022-10-17:
Interested in this issue.
Having the same result of library sending the wrong return code.

When do you foresee an update in the return value?
@zhangyanjiaoesp 

Thanks

zhangyanjiaoesp, 2022-10-18:
will raise the priority to merge the fix.

Alvin1Zhang, 2023-01-12:
Thanks for reporting and sharing the updates, sorry for slow turnaround, fix on master branch is available at https://github.com/espressif/esp-idf/commit/e6c57c0d14ab134be77fc5f7f477395b81cdeacb and on release/4.4 is https://github.com/espressif/esp-idf/commit/1c0d0e505d9fc50045bce4625a8c95aaf55af362. Feel free to reopen.

closed 2023-01-12 by Alvin1Zhang