espressif/esp-idf #9726: Information obtained by WPS cannot distinguish between 2.4GHz AP and 5GHz AP

sakamoto330, 2022-09-06:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

## Environment

- Chip : esp32S3
- IDF version : v4.4.2-1b16ef6cfc2479a08136782f9dc57effefa86f66

## Problem Description

Information obtained by WPS cannot distinguish between 2.4GHz AP and 5GHz AP.

### Issue
The following information can be obtained by executing WPS with reference to the sample code.

```c
/** Argument structure for WIFI_EVENT_STA_WPS_ER_SUCCESS event */
typedef struct {
    uint8_t ap_cred_cnt;                        /**< Number of AP credentials received */
    struct {
        uint8_t ssid[MAX_SSID_LEN];             /**< SSID of AP */
        uint8_t passphrase[MAX_PASSPHRASE_LEN]; /**< Passphrase for the AP */
    } ap_cred[MAX_WPS_AP_CRED];                 /**< All AP credentials received from WPS handshake */
} wifi_event_sta_wps_er_success_t;
```

esp32S3 doesn't support WiFi 5GHz, but some routers can get 5GHz AP information as well as 2.4GHz AP.
Is there a way to distinguish this 2.4GHz AP from a 5GHz AP?

kapilkedawat, 2022-09-06:
Hi @sakamoto330 , we take care of this by saving all the credentials and then try to connect to them one by one. 

This is already handled in wps example code and you can take reference from that.
For credentials save: https://github.com/espressif/esp-idf/blob/master/examples/wifi/wps/main/wps.c#L91 , connection retry to next network: https://github.com/espressif/esp-idf/blob/master/examples/wifi/wps/main/wps.c#L70

sakamoto330, 2022-09-06:
Thanks! @kapilkedawat 

> For credentials save: https://github.com/espressif/esp-idf/blob/master/examples/wifi/wps/main/wps.c#L91 

Above we can get two information of 5GHz AP and 2.4GHz AP, but we can get only SSID and passphrase.
Is there a way to determine which is the 5GHz AP and which is the 2.4GHz AP?

kapilkedawat, 2022-09-06:
Hi @sakamoto330 
AP can tell the station on which channels its operating on by setting ap_channel attribute in wps configuration. Unfortunately what we have observed that APs don't fill this info or it's not correctly filled, therefore we use all the available configurations to connect.

sakamoto330, 2022-09-07:
Thanks! @kapilkedawat 

>AP can tell the station on which channels its operating on by setting ap_channel attribute in wps configuration. 

Is there a way to get channels with WPS?
Is it in the sample code?
For example, is it possible to get channel information without calling esp_wifi_connect() when WIFI_EVENT_STA_WPS_ER_SUCCESS is signaled?

>Unfortunately what we have observed that APs don't fill this info or it's not correctly filled, therefore we use all the available configurations to connect.

Does it mean that channel information cannot be got correctly due to the behavior of the router (AP)?

kapilkedawat, 2022-09-09:
Hi @sakamoto330 , As I said, the channel Information may or may not present and it may not be correct even when it is present. Do you see any issues when you use the current logic to connect?

sakamoto330, 2022-09-09:
1. Calling esp_wifi_connect() to 5GHz AP fails.
This is because the esp32S3 doesn't support WiFi 5GHz.
2. On the other hand, calling esp_wifi_connect() to a 2.4GHz AP succeeds.
Now we can connect without any problem.

However, I was looking for a good way to identify the 5GHz AP before calling esp_wifi_connect() in step 1 and failing.
Do you have a good identification method?

kapilkedawat, 2022-09-12:
Hi @sakamoto330 , there is actually no good way to find out to which band credentials belong to.

However after getting the wpa done event, you can issue a scan and then issue_connect to the SSID to which station is able to scan.

sakamoto330, 2022-09-12:
Thanks! @kapilkedawat

I understood.

Also, thanks for the scan suggestion.
I would like to do it that way.

kneko715, 2022-09-13:
Hi, @kapilkedawat @sakamoto330 
Many AP don't fill the ap_channel attribute, but it fill the rf_bands attribute.
So, modifying the source as follow, you can check rf_bands and reject 5GHz credential.

```patch
diff --git a/components/wpa_supplicant/src/wps/wps_enrollee.c b/components/wpa_supplicant/src/wps/wps_enrollee.c
index a124713e45..cfcf96d442 100644
--- a/components/wpa_supplicant/src/wps/wps_enrollee.c
+++ b/components/wpa_supplicant/src/wps/wps_enrollee.c
@@ -672,6 +672,19 @@ static int wps_process_cred_e(struct wps_data *wps, const u8 *cred,
 		goto _out;
 	}
 
+	//Check RF Bands
+	if(attr->rf_bands!=NULL)
+	{
+		u8 rf_bands = *attr->rf_bands;
+		if((rf_bands & 0x01)==0) // 0x01(Bit0) means 2.4GHz
+		{
+			wpa_printf(MSG_INFO, "WPS: Reject Credential "
+				   "due to unsupported RF Bands.");
+			ret = -2;
+			goto _out;
+		}
+	}
+
 	if (os_memcmp(wps->cred.mac_addr, wps->wps->dev.mac_addr, ETH_ALEN) !=
 	    0) {
 		wpa_printf(MSG_DEBUG,  "WPS: MAC Address in the Credential ("
```

kapilkedawat, 2022-09-13:
Unfortunately rf_bands attribute is also not mandatory for credential data type in WPS specification.  
[image]

kneko715, 2022-09-13:
Thats a shame. 
But, i think its useful for application to filter out the unnecessary credentials by  the attribute of rf_bands or ap_channel.

sakamoto330, 2022-09-15:
Thanks! @kneko715 

https://github.com/espressif/esp-idf/issues/9726#issuecomment-1245227252
Actually, we can reject the 5GHz AP with this fix.

Dear, @kapilkedawat 
I also agree with kneko715.
Could you consider committing this fix?

kapilkedawat, 2024-03-04:
Since its not a mandatory attribute, we believe it may cause unnecessary interoperability issue with some AP which can fill this randomly. Therefore we won't be making this change and keep it similar to upstream wpa_supplicant.

closed 2024-03-07 by Alvin1Zhang