espressif/esp-idf #9727: Decoding core dump fails on Windows

ddomnik, 2022-09-06:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

01d014c42def8d0c19e1ce55c07de6761e092ffa

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

PowerShell

### What is the expected behavior?

After enabling the sdkconfig option to print the decoded core dump to UART
[image]

I receive the following error:
```
ELF file SHA256: 953a138ca58cee78

Initiating core dump!
I (1785) esp_core_dump_uart: Press Enter to print core dump to UART...
I (1792) esp_core_dump_uart: Print core dump to uart...
Core dump started (further output muted)
Received  35 kB...
Core dump finished!
espcoredump.py v1.2

Traceback (most recent call last):
  File "C:\Espressif\frameworks\esp-idf-master\tools/idf_monitor.py", line 374, in <module>
    main()
  File "C:\Espressif\frameworks\esp-idf-master\tools/idf_monitor.py", line 365, in main
    monitor.main_loop()
  File "C:\Espressif\frameworks\esp-idf-master\tools/idf_monitor.py", line 162, in main_loop
    self._main_loop()
  File "C:\Espressif\frameworks\esp-idf-master\tools/idf_monitor.py", line 267, in _main_loop
    super()._main_loop()
  File "C:\Espressif\frameworks\esp-idf-master\tools/idf_monitor.py", line 207, in _main_loop
    self.serial_handler.handle_serial_input(data, self.console_parser, self.coredump,
  File "C:\Espressif\frameworks\esp-idf-master\tools\idf_monitor_base\serial_handler.py", line 102, in handle_serial_input
    with coredump.check(line):
  File "contextlib.py", line 113, in __enter__
  File "C:\Espressif\frameworks\esp-idf-master\tools\idf_monitor_base\coredump.py", line 128, in check
    self._check_coredump_trigger_before_print(line)
  File "C:\Espressif\frameworks\esp-idf-master\tools\idf_monitor_base\coredump.py", line 106, in _check_coredump_trigger_before_print
    self._process_coredump()
  File "C:\Espressif\frameworks\esp-idf-master\tools\idf_monitor_base\coredump.py", line 79, in _process_coredump
    coredump.info_corefile()
  File "C:\Espressif\python_env\idf5.0_py3.8_env\lib\site-packages\esp_coredump\coredump.py", line 358, in info_corefile
    self.print_threads_info(task_info)
  File "C:\Espressif\python_env\idf5.0_py3.8_env\lib\site-packages\esp_coredump\coredump.py", line 208, in print_threads_info
    threads, _ = self.gdb_esp.get_thread_info()
  File "C:\Espressif\python_env\idf5.0_py3.8_env\lib\site-packages\esp_coredump\corefile\gdb.py", line 99, in get_thread_info
    result = self._gdbmi_run_cmd_get_one_response('-thread-info', 'done', 'result')['payload']
  File "C:\Espressif\python_env\idf5.0_py3.8_env\lib\site-packages\esp_coredump\corefile\gdb.py", line 65, in _gdbmi_run_cmd_get_one_response
    return self._gdbmi_run_cmd_get_responses(cmd, resp_message, resp_type, multiple=False)[0]
  File "C:\Espressif\python_env\idf5.0_py3.8_env\lib\site-packages\esp_coredump\corefile\gdb.py", line 58, in _gdbmi_run_cmd_get_responses
    raise ESPCoreDumpError("Couldn't find response with message '{}', type '{}' in responses '{}'".format(
esp_coredump.corefile.ESPCoreDumpError: Couldn't find response with message 'done', type 'result' in responses '[]'
```

### What is the actual behavior?

The decoded core dump should be displayed.

### Steps to reproduce.

1. Set sdkconfig as above.
2. Create crash
3. See Error

### More Information.

Tried it with the latest version on Windows - also an error.
Tried it on Ubuntu (Raspberry Pi 4) with the same commit version as above - no error.

This error may be related to my other ticket here: https://github.com/espressif/esp-idf/issues/9458
But as it works on Linux I think it is indeed a bug.

igrr, 2022-09-06:
I suspect this has the same root cause as the last issue mentioned in https://github.com/espressif/esp-idf/issues/9379, which was solved by @Lapshin in https://github.com/espressif/esp-idf/commit/f44196c46bd37c0d386b32f49514fcee442a47f5. The fix was backported to release/v5.0 in https://github.com/espressif/esp-idf/commit/06cdc0ee4f6a254f4c682e4652e345506d194e3e. Could you please update your branch and give it another try?

ztefanjo, 2023-06-02:
@igrr 
We seem to get the same error on more or less all CoreDumps we analyze since we introduced ESP-IDF v4.4.4. We are still being able to analyze the dumps (from v4.4.4 SW) using ESP-IDF v4.4.3. Is the fix that you mention in your comment also backported to v5.0? If so, it does not work. For us, it appears that this error was introduced (or made more likely to happen) in v4.4.4.

Output from the tool is:
```
python command to file
Traceback (most recent call last):
  File "C:\Data\Git\mower\Ext\esp-idf\components\espcoredump\espcoredump.py", line 350, in <module>
    temp_core_files = info_corefile()
  File "C:\Data\Git\mower\Ext\esp-idf\components\espcoredump\espcoredump.py", line 200, in info_corefile
    threads, _ = gdb.get_thread_info()
  File "C:\Data\Git\mower\Ext\esp-idf\components\espcoredump\corefile\gdb.py", line 113, in get_thread_info
    result = self._gdbmi_run_cmd_get_one_response('-thread-info', 'done', 'result')['payload']
  File "C:\Data\Git\mower\Ext\esp-idf\components\espcoredump\corefile\gdb.py", line 79, in _gdbmi_run_cmd_get_one_response
    return self._gdbmi_run_cmd_get_responses(cmd, resp_message, resp_type, multiple=False)[0]
  File "C:\Data\Git\mower\Ext\esp-idf\components\espcoredump\corefile\gdb.py", line 72, in _gdbmi_run_cmd_get_responses
    raise ESPCoreDumpError("Couldn't find response with message '{}', type '{}' in responses '{}'".format(
corefile.ESPCoreDumpError: Couldn't find response with message 'done', type 'result' in responses '[]'
```

felixcollins, 2023-07-17:
I've got this issue too. I tried updating to head of branch 5.0  - 15 July 2023 - 8df440175af54b5a8bf7228faf3ba48fc0b5ca22

It did not fix the issue. I'll try with Head of Master soon.

This is a real PITA. Basic tooling like this should be a high priority to fix.  I've been suffering with this crash since starting a project with a 5.0 release.

`
Traceback (most recent call last):
  File "C:\Users\felix\esp5\esp-idf\tools\idf_monitor.py", line 378, in <module>
    main()
  File "C:\Users\felix\esp5\esp-idf\tools\idf_monitor.py", line 369, in main
    monitor.main_loop()
  File "C:\Users\felix\esp5\esp-idf\tools\idf_monitor.py", line 163, in main_loop
    self._main_loop()
  File "C:\Users\felix\esp5\esp-idf\tools\idf_monitor.py", line 269, in _main_loop
    super()._main_loop()
  File "C:\Users\felix\esp5\esp-idf\tools\idf_monitor.py", line 208, in _main_loop
    self.serial_handler.handle_serial_input(data, self.console_parser, self.coredump,
  File "C:\Users\felix\esp5\esp-idf\tools\idf_monitor_base\serial_handler.py", line 102, in handle_serial_input
    with coredump.check(line):
  File "contextlib.py", line 113, in __enter__
  File "C:\Users\felix\esp5\esp-idf\tools\idf_monitor_base\coredump.py", line 128, in check
    self._check_coredump_trigger_before_print(line)
  File "C:\Users\felix\esp5\esp-idf\tools\idf_monitor_base\coredump.py", line 106, in _check_coredump_trigger_before_print
    self._process_coredump()
  File "C:\Users\felix\esp5\esp-idf\tools\idf_monitor_base\coredump.py", line 79, in _process_coredump
    coredump.info_corefile()
  File "c:\Users\felix\esp5\esp-tools\python_env\idf5.0_py3.8_env\lib\site-packages\esp_coredump\coredump.py", line 372, in info_corefile
    self.print_threads_info(task_info)
  File "c:\Users\felix\esp5\esp-tools\python_env\idf5.0_py3.8_env\lib\site-packages\esp_coredump\coredump.py", line 218, in print_threads_info
    threads, _ = self.gdb_esp.get_thread_info()
  File "c:\Users\felix\esp5\esp-tools\python_env\idf5.0_py3.8_env\lib\site-packages\esp_coredump\corefile\gdb.py", line 100, in get_thread_info
    result = self._gdbmi_run_cmd_get_one_response('-thread-info', 'done', 'result', response_delay_sec=2)['payload']
  File "c:\Users\felix\esp5\esp-tools\python_env\idf5.0_py3.8_env\lib\site-packages\esp_coredump\corefile\gdb.py", line 65, in _gdbmi_run_cmd_get_one_response
    return self._gdbmi_run_cmd_get_responses(cmd, resp_message, resp_type, response_delay_sec=response_delay_sec,
  File "c:\Users\felix\esp5\esp-tools\python_env\idf5.0_py3.8_env\lib\site-packages\esp_coredump\corefile\gdb.py", line 58, in _gdbmi_run_cmd_get_responses
    raise ESPCoreDumpError("Couldn't find response with message '{}', type '{}' in responses '{}'".format(
esp_coredump.corefile.ESPCoreDumpError: Couldn't find response with message 'done', type 'result' in responses '[]'
`

ztefanjo, 2024-03-06:
@igrr, any update on this one? We are now using v5.1.2 and still have big problems to parse coredumps with the v5.1.2 parser. So we actually still use v4.4.3 to parse our dumps (but need to apply some patches for it to work, e.g. remove all "noreturn" on functions frequently used during panic).

Examples of errors:
```
python command to file
Traceback (most recent call last):
  File "C:\Data\Git\mower\Ext\esp-idf\components\espcoredump\espcoredump.py", line 77, in <module>
    main()
  File "C:\Data\Git\mower\Ext\esp-idf\components\espcoredump\espcoredump.py", line 62, in main
    temp_core_files = espcoredump.info_corefile()
  File "C:\Data\Git\mower\Ext\python_env\idf5.1_py3.9_env\lib\site-packages\esp_coredump\coredump.py", line 566, in info_corefile
    self.print_threads_info(task_info)
  File "C:\Data\Git\mower\Ext\python_env\idf5.1_py3.9_env\lib\site-packages\esp_coredump\coredump.py", line 334, in print_threads_info
    threads, _ = self.gdb_esp.get_thread_info(attempt * DEFAULT_GDB_TIMEOUT_SEC)
  File "C:\Data\Git\mower\Ext\python_env\idf5.1_py3.9_env\lib\site-packages\esp_coredump\corefile\gdb.py", line 122, in get_thread_info
    current_thread_id = result['current-thread-id']
KeyError: 'current-thread-id'
```

```
  File "C:\Python\Python39\lib\threading.py", line 973, in _bootstrap_inner
Exception in thread Thread-2:
Traceback (most recent call last):
  File "C:\Python\Python39\lib\threading.py", line 973, in _bootstrap_inner
    self.run()
  File "C:\Python\Python39\lib\threading.py", line 910, in run
    self.run()
    self._target(*self._args, **self._kwargs)
  File "C:\Python\Python39\lib\threading.py", line 910, in run
  File "C:\Python\Python39\lib\subprocess.py", line 1479, in _readerthread
    self._target(*self._args, **self._kwargs)
  File "C:\Python\Python39\lib\subprocess.py", line 1479, in _readerthread
    buffer.append(fh.read())
OSError: [Errno 22] Invalid argument
    buffer.append(fh.read())
OSError: [Errno 22] Invalid argument
```
The fact that v4.4.3 is able to parse the dump speaks for the the dump itself is correct, but the parser is not being able to handle it.

Our sdkconfig relating to this is:
```
CONFIG_ESP_COREDUMP_ENABLE_TO_FLASH=y
# CONFIG_ESP_COREDUMP_ENABLE_TO_UART is not set
# CONFIG_ESP_COREDUMP_ENABLE_TO_NONE is not set
# CONFIG_ESP_COREDUMP_DATA_FORMAT_BIN is not set
CONFIG_ESP_COREDUMP_DATA_FORMAT_ELF=y
CONFIG_ESP_COREDUMP_CHECKSUM_CRC32=y
CONFIG_ESP_COREDUMP_CHECK_BOOT=y
CONFIG_ESP_COREDUMP_ENABLE=y
CONFIG_ESP_COREDUMP_LOGS=y
CONFIG_ESP_COREDUMP_MAX_TASKS_NUM=20
CONFIG_ESP_COREDUMP_USE_STACK_SIZE=y
CONFIG_ESP_COREDUMP_STACK_SIZE=1280
```

igrr, 2024-03-06:
@ztefanjo Thanks for the information! Do you think you could share an example coredump binary along with the ELF file which reproduces this issue? (For example, over email if you can't upload them here.)

@erhankur PTAL if we need to backport any fixes to v5.1.x?

ztefanjo, 2024-03-06:
@erhankur : Strange, when reinstalling 5.1.2, I still get Python 3.9. Are we perhaps doing something wrong, or is the Python version stepped to 3.11 first in esp-idf v5.1.3?

erhankur, 2024-03-06:
@ztefanjo For Windows, the best way to install the tools is by using an online installer.  https://dl.espressif.com/dl/esp-idf/ 
You can keep your existing idf but re-install the other things around.

Zmmfly, 2024-04-07:
Same problem with python310 and idf 5.2.1

erhankur, 2024-04-08:
@Zmmfly Can you select the don't decode option and send the base64 output with your app elf file? (From here or by email)

I would like to make sure coredump data is recorded without an error and the issue is specific to Windows. 
I couldn't reproduce it yet also on Windows, though. Maybe you can prepare a minimal reproducible app.

And, did you try to transfer with a lower baud rate like 115200?