espressif/esp-idf #9731: ESP32-S3: Strapping pins undocumented pull-up resistances

yannishuber, 2022-09-06:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

### Concerned Hardware

ESP32-S3-DevKitC-1 v1.0 (D4N32R8V)

### Problem Description

The strapping pins GPIO0 and GPIO45 are pulled up by an external resistance.

In the [datasheet](https://www.espressif.com/sites/default/files/documentation/esp32-s3_datasheet_en.pdf) (Section 2.5), GPIO 0 and GPIO 45 are documented to be both during and after reset in the configuration E1, WPU1 and IE1, WPD1 respectively. However, when measuring GPIO 45 after reset we get 3.3V. This can be explained by the R1 resistor documented in the WROOM2 [datasheet](https://www.espressif.com/sites/default/files/documentation/esp32-s3-wroom-2_datasheet_en.pdf) (Page 23) which seems to be there even though it is labeled as "NC". When measuring the resistance between the 3.3V pin and GPIO 45 we can confirm that the 10KΩ resistor is there.

[image]

Because of that, the internal pull-down resistor is not enough to pull the line low when the GPIO is configured as input with pull-down. Even with the pull-down enabled, the line only drops to about 2.5V. We observed the same issue with GPIO 0. GPIO 0 is supposed to be pulled-up after reset so there is no problem there, however, if using the pin as input with pull-down, we only drop the voltage to about 2.7V.

The other strapping pins (GPIO 3 and GPIO 46) do not have the same issue.

### Expected Behavior

Configuring pins GPIO 45 and GPIO 0 as inputs with pull-down resistors should measure 0V on both lines.

### Actual Behavior

GPIO 45 and GPIO 0 have the wrong levels on input (~2.5V and ~2.7V)

[image]

murphymomo, 2022-09-08:
Hi @yannishuber 

For GPIO45: Would you please check which version of the ESP32-S3-WROOM-2 does your development board integrate? This pullup is populated in previous versions. If it is older versions, then your analysis is correct.

For GPIO0: Please kindly check the [ESP32-S3-DevKitC-1 schematic](https://dl.espressif.com/dl/schematics/SCH_ESP32-S3-DevKitC-1_V1.1_20220413.pdf) in [ESP32-S3-DevKitC-1 User guide](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/hw-reference/esp32s3/user-guide-devkitc-1.html). You can find that GPIO0 is connected to a transistor and this will give a strong pull up (see the screenshot below). That’s why the voltage only dropped to about 2.7V.
[image]

yannishuber, 2022-09-08:
Hello @esp-momo, thank you for your reply. How can I find out the version of the WROOM module I am using? The only information I found is the print on the chip that says ESP32-S3-WROOM2.

Thank you for pointing out the strong pull-up on GPIO 0, that makes more sense now.

In our use case, we use GPIO45 as an input/output pin connected to a bidirectional level translator which detects the direction using the level difference. So when GPIO45 is used as an input pin, the strong pull up disrupts the level translator. We can solve the issue by using another pin, however, I think the datasheet should be updated to describe this behavior instead of saying that the pin is input enabled and pulled-low after reset, which is clearly not the case.

esp-momo, 2022-09-13:
Hi @yannishuber

Regarding the module version, I am sorry that I got some misinformation earlier. After a double check, I found out that the current version of ESP32-S3-WROOM-2 modules on market do have this pull-up resistor on GPIO45. And we are just planning to release a new version of ESP32-S3-WROOM-2 module without this resistor on GPIO45.

Sorry for the trouble. We will update the module schematic for [ESP32-S3-WROOM-2 datasheet](https://www.espressif.com/sites/default/files/documentation/esp32-s3-wroom-2_datasheet_en.pdf) by removing the NC label for the resistor on GPIO45.

yannishuber, 2022-09-13:
All right, @esp-momo thank you for looking into this!

closed 2023-01-13 by Alvin1Zhang

tonystuart, 2024-07-03:
Hi,

I see that the [ESP32-S3-WROOM-2 datasheet](https://www.espressif.com/sites/default/files/documentation/esp32-s3-wroom-2_datasheet_en.pdf) link that is referenced in the response from @esp-momo  still shows the 10K R1 pull-up resistor on GPIO45 in gray, with the text NC.

Is this resistor only present in the 8V and 16V versions of the module that use 1.8V for VDD_SPI? If so, is it possible to add a note to the schematic? Also, do these versions use 1.8V for a high level on GPIO47 and GPIO48?

Finally, can you confirm that other versions of the module without the V suffix (e.g. ESP32-S3-WROOM-1-N16R8) do not have the pull-up resistor on GPIO45 and use 3.3V for a high level on GPIO47 and GPIO48?

Thanks!

esp-momo, 2024-07-04:
Hi @tonystuart,

For ESP32-S3-WROOM-1/1U, the pull-up resistor of GPIO45 is not populated in all module versions.

For ESP32-S3-WROOM-2, please refer to [BOM Change for ESP32-S3-WROOM-2](https://www.espressif.com/sites/default/files/pcn_downloads/PCN20220802%20ESP32-S3-WROOM-2%20%E6%A8%A1%E7%BB%84%20BOM%20%E5%8F%98%E6%9B%B4.pdf). The VDD_SPI voltage of the ESP32-S3R8V chip on the ESP32-S3-WROOM-2 has been set to 1.8V by eFuse VDD_SPI_TIEH and VDD_SPI_FORCE, and is no longer controlled by GPIO45. Therefore, the external pull up resistor R1 on the module is changed to be not mounted from Oct 2022. Therefore, the schematics in the datasheet is not changed.

VDD_SPI can be controlled by eFuse without the affect of GPIO45. Please refer to [ESP32-S3-WROOM-2 Datasheet](https://www.espressif.com/sites/default/files/documentation/esp32-s3-wroom-2_datasheet_en.pdf) or [ESP32-S3-WROOM-1/1U Datasheet](https://www.espressif.com/sites/default/files/documentation/esp32-s3-wroom-1_wroom-1u_datasheet_en.pdf) > Section 3.3.2 _VDD_SPI Voltage Control_:
[image]

For another, GPIO47 and GPIO48 work in VDD_SPI voltage. Since VDD_SPI of ESP32-S3R8V and ESP32-S3R16V has been set to 1.8 V by eFuse, for modules embedded with these two chip variants, i.e., the modules with the prefix of “R8V” or “R16V”, GPIO47 and GPIO48 also work in 1.8 V.

Hope it helps!

tonystuart, 2024-07-09:
Thanks @esp-momo -- very helpful!