espressif/esp-idf #9733: Add documentation for debugging with lldb

TheButlah, 2022-09-06:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

There are docs for [debugging with gdb](https://github.com/espressif/esp-idf/blob/572e79530c/docs/en/api-guides/jtag-debugging/using-debugger.rst) but I am on MacOS with the new apple m1 (aarch64) chip. GDB is not available on this platform, so I will need to use lldb.

Could the documentation be updated (or at least this github issue) to provide instructions for debugging with lldb?

igrr, 2022-09-06:
@TheButlah Could you please describe the issue you are facing on aarch64 macOS with GDB in more detail, please? With the `master` branch of ESP-IDF, GDB for both Xtensa and RISC-V should work fine on aarch64 macOS.
If the steps described in the documentation don't work for you, please provide more information and we will help you solve the issue.

TheButlah, 2022-09-06:
Basically, I tried to use `brew install gdb`. My logic was mainline gdb supports RISC-V so I wouldn't need to use the espressif fork.

My assumption was that the espressif fork would also not work on aarch64 macos, for the same reasons that regular gdb isn't distributed for the platform.

Maybe that was an incorrect assumption? Does espressif provide a gdb for aarch64?

Even still, it would be good to provide instructions for lldb users, even if its just a brief code snippet of lldb commands.

Lapshin, 2022-09-06:
> Maybe that was an incorrect assumption? Does espressif provide a gdb for aarch64 macos?

@TheButlah , you can get GDB for the popular operation systems via links in the documentation, https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/tools/idf-tools.html#riscv32-esp-elf-gdb

HeiderSati, 2024-10-17:
I ran install.sh (installation completed), and also did . ./export.sh (Environment Vars set), however, I cannot find gdb anywhere on the installation nor is that available on the path anywhere. 
Also, when I try something like "brew install gdb" (or any GDB) the message I get is that my MacOSX (Apple Silicon M3 Pro) is not supported for GDB. 
LLDB is the only option as it seems. 

I do get GDB (somehow) used (or installed?!) when I run CLION (C/C++ development tool) , but when that tries to use GDB to connect into OpenOCD to debug my ESP32-S3 chip it chokes out with an error of 'g' packet too long, expected 452 but received 512 bytes, no more information provided. 

When I use LLDB to "target create blink.elf", I get an error that the architecture is not one of the one it knows about (I.e. ARM, X86, etc). 

When I also try to use the GUI to load the ELF, same error. 

Therefore, if LLDB "can" work then would appreciate if any information could be shared to know what to do, or what additional libs to install to debug my ESP32 chip. 

Thanks

igrr, 2024-10-17:
@HeiderSati After install.sh and export.sh, there should be `xtensa-esp32s3-elf-gdb` in PATH. To troubleshoot, could you please:

- check the output of `which xtensa-esp32s3-elf-gdb` after `. ./export.sh`
- if nothing is found, capture the full output of `. ./export.sh` in a new terminal and share it here.

Thank you.

HeiderSati, 2024-10-17:
Oh My Goodness!!!... Thank you so much!... yes, it's there and I just launched it, I owe you a drink, Cheers :)