espressif/esp-idf #9734: 使用freertos  中xTaskCreate 堆大小1024 会无限重启 (Using the xTaskCreate heap size of 1024 in Freertos will restart indefinitely)

Nuozhihui, 2022-09-06:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### Development Kit.

ESP32-S3-WROOM-1

### Power Supply used.

USB

### What is the expected behavior?

Create a task in Freertos without restarting it

### What is the actual behavior?

创建一个任务在freertos
Create a task in Freertos

### Steps to reproduce.

代码
`#include <stdio.h>
#include "sdkconfig.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_chip_info.h"
#include "esp_flash.h"

void myTask(void * pvParameters){   //用空指针就可以传递任何类型，传进来后再转化
	while(1){//任务函数里必须是死循环
        printf("HI\n");
		//实际任务
		vTaskDelay(1000/portTICK_PERIOD_MS);//任务函数里必须有延时,也可用pdMS_TO_TICKS(1000)代替
	}
	vTaskDelete(NULL);//在本函数中删除，不需要传递句柄
}

void app_main(void){
	//创建任务句柄
    TaskHandle_t myHandle = NULL;
    //任务指针，任务标签，堆栈大小，传入参数，优先级，任务句柄
    xTaskCreate(myTask,"myTask",1024,NULL,1,&myHandle);//传出任务句柄

   // vTaskDelay(8000/portTICK_PERIOD_MS);//任务函数里必须有延时,也可用pdMS_TO_TICKS(1000)代替

    // //通过任务句柄删除
    if(myHandle!=NULL){
        vTaskDelete(myHandle);//在任务函数外删除任务
    }
}
`

无限重启，设置堆大小为2048时，不出现此问题
This problem does not occur when the heap size is set to 2048 after infinite restart

### Debug Logs.

```plain
I (24) boot: ESP-IDF ae256e9ca2-dirty 2nd stage bootloader
I (24) boot: compile time 00:48:48
I (24) boot: chip revision: V001
I (27) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (34) boot.esp32s3: Boot SPI Speed : 80MHz
I (39) boot.esp32s3: SPI Mode       : DIO
I (43) boot.esp32s3: SPI Flash Size : 2MB
I (48) boot: Enabling RNG early entropy source...
I (53) boot: Partition Table:
I (57) boot: ## Label            Usage          Type ST Offset   Length
I (64) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (72) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (79) boot:  2 factory          factory app      00 00 00010000 00100000
I (87) boot: End of partition table
I (91) boot_comm: chip revision: 1, min. application chip revision: 0
I (98) esp_image: segment 0: paddr=00010020 vaddr=3c020020 size=08494h ( 33940) map
I (113) esp_image: segment 1: paddr=000184bc vaddr=3fc90f00 size=02d3ch ( 11580) load
I (117) esp_image: segment 2: paddr=0001b200 vaddr=40374000 size=04e18h ( 19992) load
I (128) esp_image: segment 3: paddr=00020020 vaddr=42000020 size=191ech (102892) map
I (150) esp_image: segment 4: paddr=00039214 vaddr=40378e18 size=08060h ( 32864) load
I (158) esp_image: segment 5: paddr=0004127c vaddr=50000000 size=00010h (    16) load
I (164) boot: Loaded app from partition at offset 0x10000
I (164) boot: Disabling RNG early entropy source...
I (179) cpu_start: Pro cpu up.
I (179) cpu_start: Starting app cpu, entry point is 0x40375194
0x40375194: call_start_cpu1 at C:/Espressif/frameworks/esp-idf-master/components/esp_system/port/cpu_start.c:142

I (0) cpu_start: App cpu up.
I (194) cpu_start: Pro cpu start user code
I (194) cpu_start: cpu freq: 160000000 Hz
I (194) cpu_start: Application information:
I (197) cpu_start: Project name:     hello_world
I (202) cpu_start: App version:      1
I (206) cpu_start: Compile time:     Sep  7 2022 00:48:41
I (212) cpu_start: ELF file SHA256:  f51071143655e03f...
I (218) cpu_start: ESP-IDF:          ae256e9ca2-dirty
I (224) heap_init: Initializing. RAM available for dynamic allocation:
I (231) heap_init: At 3FC94650 len 000550C0 (340 KiB): D/IRAM
I (238) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DRAM
I (244) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (251) heap_init: At 600FE010 len 00001FF0 (7 KiB): RTCRAM
I (258) spi_flash: detected chip: generic
I (262) spi_flash: flash io: dio
W (266) spi_flash: Detected size(8192k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (279) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
HI
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0xb (SPI_FAST_FLASH_BOOT)
Saved PC:0x40375635
0x40375635: esp_restart_noos_dig at C:/Espressif/frameworks/esp-idf-master/comSPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3810,len:0x1698
load:0x403c9700,len:0xbd0
ponents/esp_system/esp_system.c:46 (discriminator 1)

load:0x403cc700,len:0x2c6c
entry 0x403c9900
```

ESP-Marius, 2022-09-07:
Hi,

`printf` (and related log functions) requires a lot of stack, so 1024 is not sufficient.

Nuozhihui, 2022-09-07:
栓Q(gratitude)

closed 2022-09-07 by Nuozhihui