espressif/esp-idf #9737: esp32 modbus problem

zhangxujun16, 2022-09-07:
**Environment**
Development Kit: [ESP32-Wrover-Kit]
Kit version (for WroverKit/PicoKit/DevKitC): [**v4.4**]
Module or chip used: [ESP32-WROVER-B]
IDF version: [**v4.4**]

Modbus master works well for several hours at first, but suddenly an error information happan,  _E (1767200) MB_PORT_COMMON: vMBMasterRunResRelease(194): Resource release failure. E (1767200) MB_CONTROLLER_MASTER: mbc_master_get_parameter(83): Master get parameter failure, error=(0x108) (ESP_ERR_INVALID_RESPONS  ,_  it lasts telling this error info, and never recover, unless reset  esp32 CPU .  If reset esp32 CPU , it will also work well for several hours but  then such error happens again.  

There is a similar problem linker : https://github.com/espressif/esp-idf/issues/5072 , but his verision is v4 .   I think the bug is improved and patch merge into V4.4.   But in fact the bug maybe still exist. 

Thanks!

zhangxujun16, 2022-09-07:
Thank you for your help.    How to fix this problem ?  And just do like this :  Add  vMBMasterRunResRelease( ) 

eMBMasterReqErrCode eMBMasterWaitRequestFinish( void ) {
         .....
        } else if (MB_PORT_CHECK_EVENT(xRecvedEvent, EV_MASTER_ERROR_EXECUTE_FUNCTION)) {
            eErrStatus = MB_MRE_EXE_FUN;
            **vMBMasterRunResRelease();**
        ....
Is it right ?

alisitsyn, 2022-09-07:
@ zhangxujun16 ,

Thank you for the issue. Unfortunately the `release/v4.4` branch does not have all required patches and bug still exists. Most likely the backports will not be merged because the `freemodbus` component has been removed from esp-idf starting from version v5.0. The v4.4 version of freemodbus component will be linked to new `esp-modbus` component. Please consider using the [esp-modbus component](https://github.com/espressif/esp-modbus) which can work with all esp-idf versions starting from `release/v4.1`.

zhangxujun16, 2022-09-07:
@alisitsyn
Thank you very much for help.  Do you mean that   [esp-modbus component](https://github.com/espressif/esp-modbus) has fix this problem ?  I need to use  [esp-modbus component](https://github.com/espressif/esp-modbus)  to replace the freemodbus of esp-idf .  Thanks!

alisitsyn, 2022-09-08:
@zhangxujun16 ,

Several things can cause this behavior that you described. The esp-modbus still do not have all existing fixes because they may cause unpredictable behavior for other ports when implemented on the stack side. Please see the [similar issue](https://github.com/espressif/esp-modbus/issues/11) which causes the same result with shifted data.

I can help to fix the issues on your side. Let me know if you have any issues implementing this.

zhangxujun16, 2022-09-08:
@alisitsyn
I only add vMBMasterRunResRelease( ) in V4.4 like bellow and not  use [esp-modbus component](https://github.com/espressif/esp-modbus) :  
eMBMasterReqErrCode eMBMasterWaitRequestFinish( void ) {
.....
} else if (MB_PORT_CHECK_EVENT(xRecvedEvent, EV_MASTER_ERROR_EXECUTE_FUNCTION)) {
eErrStatus = MB_MRE_EXE_FUN;
**vMBMasterRunResRelease( );**
....

And I have tested for 24 hours,  It seems work well .   Occasionally an error happens like _E (1767200) MB_PORT_COMMON: vMBMasterRunResRelease(194): Resource release failure. E (1767200) MB_CONTROLLER_MASTER: mbc_master_get_parameter(83): Master get parameter failure, error=(0x108) (ESP_ERR_INVALID_RESPONS_  , but it can recover in next operations.   Occaionally an error is acceptable for us . The most important point is that the master can recover immediately if the slavers have no problems.  I think this fix is OK for us.  In furture I will test with [esp-modbus component](https://github.com/espressif/esp-modbus) .  Thanks !

alisitsyn, 2022-09-08:
@zhangxujun16 ,

Thank you for information. However it helps in certain conditions and issue can come again later when they changed.
Good luck!

alisitsyn, 2024-01-24:
The issue is closed. Feel free to reopen it if you still have issues.

closed 2024-01-24 by alisitsyn