espressif/esp-idf #9738: Unexpected error messages

ammaree, 2022-09-07:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1-dev-644-g867745a05c

### Operating System used.

macOS

### How did you build your project?

Eclipse IDE

### Development Kit.

ESP32-WROVER-KIT v4.1

### Power Supply used.

External 5V

### What is the expected behavior?

Not get any unexplainable errors

### What is the actual behavior?

We consistently get 2 unexpected warnings:
0.535: #0 main vfs esp_vfs_register_fd_range is successful for range <54; 64) and VFS ID 4
4.399: #0 wifi wifi <ba-add>idx:0 (ifx:0, ba:69:f4:b5:56:75), tid:0, ssn:3, winSize:64

### Steps to reproduce.

Part of large private application and show up early in boot process.

VFS message appears as if it is something that should only come up in DEBUG level logging.

WIFI message appears soon after connection to AP.

### Debug Logs.

```plain
''0.369: #0 boot quad_psram This chip is ESP32-D0WD
0.370: #0 boot esp_psram Found 8MB PSRAM device
0.370: #0 boot esp_psram Speed: 80MHz
0.370: #0 boot esp_psram PSRAM initialized, cache is in low/high (2-core) mode.
0.377: #0 boot esp_psram Virtual address not enough for PSRAM, map as much as we can. 4MB is mapped
0.386: #0 boot cpu_start Pro cpu up.
0.389: #0 boot cpu_start Starting app cpu, entry point is 0x40081934
0.000: #1 boot cpu_start App cpu up.
0.410: #0 boot cpu_start Pro cpu start user code
0.410: #0 boot cpu_start cpu freq: 160000000 Hz
0.410: #0 boot cpu_start Application information:
0.414: #0 boot cpu_start Project name:     4bb87127839b4e9c8483eeef2948f8e
0.420: #0 boot cpu_start App version:      0.6.0.11
0.425: #0 boot cpu_start Compile time:     Sep  7 2022 08:56:16
0.431: #0 boot cpu_start ELF file SHA256:  395261f81dc81a7f...
0.436: #0 boot cpu_start ESP-IDF:          v5.1-dev-644-g867745a05c-dirty
0.443: #0 boot heap_init Initializing. RAM available for dynamic allocation:
0.450: #0 boot heap_init At 3FFAE6E0 len 00001920 (6 KiB): DRAM
0.455: #0 boot heap_init At 3FFC53F8 len 0001AC08 (107 KiB): DRAM
0.461: #0 boot heap_init At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
0.467: #0 boot heap_init At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
0.473: #0 boot heap_init At 40098B88 len 00007478 (29 KiB): IRAM
0.479: #0 boot esp_psram Adding pool of 4086K of PSRAM memory to heap allocator
0.000: #1 boot cpu_start Starting scheduler on APP CPU.
0.491: #0 boot esp_psram Reserving pool of 32K of internal memory for DMA/internal allocations
0.486: #0 preX spi_flash detected chip: generic
0.486: #0 preX spi_flash flash io: qio
0.486: #0 preX cpu_start Starting scheduler on PRO CPU.
0.535: #0 main vfs esp_vfs_register_fd_range is successful for range <54; 64) and VFS ID 4
0.692: #0 main wifi wifi driver task: 3ffd7b74, prio:23, stack:3584, core=0
0.694: #0 wifi system_api Base MAC address is not set
0.694: #0 wifi system_api read default base MAC address from EFUSE
0.716: #0 wifi wifi wifi firmware version: 9105b9c
0.717: #0 wifi wifi wifi certification version: v7.0
0.718: #0 wifi wifi config NVS flash: enabled
0.718: #0 wifi wifi config nano formating: disabled
0.719: #0 wifi wifi Init data frame dynamic rx buffer num: 32
0.720: #0 wifi wifi Init management frame dynamic rx buffer num: 32
0.721: #0 wifi wifi Init management short buffer num: 32
0.722: #0 wifi wifi Init static tx buffer num: 16
0.724: #0 wifi wifi Init tx cache buffer num: 32
0.725: #0 wifi wifi Init static rx buffer size: 1600
0.726: #0 wifi wifi Init static rx buffer num: 16
0.727: #0 wifi wifi Init dynamic rx buffer num: 32
0.728: #1 main wifi_init rx ba win: 16
0.729: #1 main wifi_init tcpip mbox: 32
0.730: #1 main wifi_init udp mbox: 6
0.730: #1 main wifi_init tcp mbox: 6
0.731: #1 main wifi_init tcp tx win: 5744
0.732: #1 main wifi_init tcp rx win: 5744
0.732: #1 main wifi_init tcp mss: 1440
0.733: #1 main wifi_init WiFi/LWIP prefer SPIRAM
0.734: #0 wifi wifi Set ps type: 0

0.762: #0 wifi phy_init phy_version 4670,719f9f6,Feb 18 2021,17:07:07
0.926: #0 wifi phy_init saving new calibration data because of checksum failure, mode(0)
1.063: #0 wifi wifi mode : sta (94:e6:86:0f:97:94)
1.064: #0 wifi wifi enable tsf
1.065: #1 main EvtFlag  TIMER L1
1.066: #1 main SysFlag  BufOK LFS APP
3.476: #0 wifi wifi new:<9,0>, old:<1,0>, ap:<255,255>, sta:<9,0>, prof:1
4.359: #0 wifi wifi state: init -> auth (b0)
4.362: #0 wifi wifi state: auth -> assoc (0)
4.368: #0 wifi wifi state: assoc -> run (10)
4.376: #0 wifi wifi connected with irmacos, aid = 1, channel 9, BW20, bssid = ba:69:f4:b5:56:75
4.377: #0 wifi wifi security: WPA2-PSK, phy: bgn, rssi: -36
4.392: #0 wifi wifi pm start, type: 0

4.393: #0 wifi wifi AP's beacon interval = 102400 us, DTIM period = 1
4.399: #0 wifi wifi <ba-add>idx:0 (ifx:0, ba:69:f4:b5:56:75), tid:0, ssn:3, winSize:64
4.471: #1 main EvtFlag  TIMER L2sta L1
5.192: #0 sys_evt esp_netif_handlers sta ip: 192.168.0.15, mask: 255.255.255.0, gw: 192.168.0.1
5.223: #0 SNTP xNtpGetTime 0.pool.ntp.org(13.244.55.14)  2022-09-07T07:02:04.804998Z  tOFF=344,069,055uS  tRTD=20,943uS
5.273: #1 main EvtFlag  TIMER HTTPS TNETS SNTP L3sta L2sta L1
5.275: #1 main halCONFIG_UpdateBootInfo P#2 v0.6.0.11 built 2022-09-07T06:56:16Z
6.168: #1 actuate gpio GPIO[0]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
6.170: #0 actuate gpio GPIO[2]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
6.172: #1 actuate gpio GPIO[4]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
6.175: #1 Events gpio GPIO[0]| InputEn: 1| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
6.177: #1 main EvtFlag  i2cOK HTTPS TNETS SNTP L3sta L2sta L1
6.178: #1 main SysFlag  BufOK LFS FWok FWchk APP
6.345: #1 main EvtFlag  i2cOK HTTPS TNETS MQTT SNTP L3sta L2sta L1
7.247: #1 main EvtFlag  i2cOK TIMER HTTPS TNETS MQTT SNTP L3sta L2sta L1
7.855: #1 main EvtFlag  RULES RGSTR i2cOK TIMER HTTPS TNETS MQTT SNTP L3sta L2sta L1
7.958: #1 main SysFlag  U0 BufOK LFS FWok FWchk APP
```

MaxwellAlan, 2022-09-13:
Hi @ammaree 

> 4.399: #0 wifi wifi idx:0 (ifx:0, ba:69:f4:b5:56:75), tid:0, ssn:3, winSize:64

This warrning log means that Sta setup BlockACK with AP, the tid is 0, start sequence num(ssn) is 3, window size is 64. You can treat this as an info log :)

AxelLin, 2022-09-13:
> Hi @ammaree
> 
> > 4.399: #0 wifi wifi idx:0 (ifx:0, ba:69:f4:b5:56:75), tid:0, ssn:3, winSize:64
> 
> You can treat this as an info log :)

Then please fix the log level to avoid misunderstanding. (Change W to I )

ammaree, 2022-09-13:
@MaxwellAlan 
I agree with @AxelLin, if it is not a warning, then can we please treat the cause (change W -> I) and not the symptom (trying to ignore the message)

And the other message is the same situation, reporting success as a warning or error does not make sense.

AxelLin, 2022-10-06:
> Hi @ammaree
> 
> > 4.399: #0 wifi wifi idx:0 (ifx:0, ba:69:f4:b5:56:75), tid:0, ssn:3, winSize:64
> 
> This warrning log means that Sta setup BlockACK with AP, the tid is 0, start sequence num(ssn) is 3, window size is 64. You can treat this as an info log :)

@MaxwellAlan

FYI, similar warning has been reported in many places, e.g. https://www.esp32.com/viewtopic.php?f=13&t=29794
If it's really not a warning message, why not change it to correct log level?

zhangyanjiaoesp, 2022-10-18:
> > Hi @ammaree
> > > 4.399: #0 wifi wifi idx:0 (ifx:0, ba:69:f4:b5:56:75), tid:0, ssn:3, winSize:64
> > 
> > 
> > You can treat this as an info log :)
> 
> Then please fix the log level to avoid misunderstanding. (Change W to I )

will fix the log level.

AxelLin, 2022-10-19:
@zhangyanjiaoesp 
If you are working on this, maybe you want to check if below is a typo or I misunderstand the message:
  I (78467) wifi:bcn_timout,ap_probe_send_start
The "bcn_timout" looks like a typo, maybe it should be "bcn_timeout"?

zhangyanjiaoesp, 2022-10-19:
> @zhangyanjiaoesp If you are working on this, maybe you want to check if below is a typo or I misunderstand the message: I (78467) wifi:bcn_timout,ap_probe_send_start The "bcn_timout" looks like a typo, maybe it should be "bcn_timeout"?

will fix it

Alvin1Zhang, 2023-01-12:
Thanks for reporting and sharing the updates, sorry for slow turnaround, fix is available at https://github.com/espressif/esp-idf/commit/e6c57c0d14ab134be77fc5f7f477395b81cdeacb. Feel free to reopen.

closed 2023-01-12 by Alvin1Zhang