espressif/esp-idf #9739: BLE Mesh and WIFI STA mode coexistence connectivity issue

Shane555, 2022-09-07:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

## Environment
- Development Kit:      [ESP32-C3-DevKitM-1]
- Kit version (for DevKitC): |v3|
- IDF version: v5.1.0
- Build System:         [idf.py]
- Compiler version xtensa-esp32-elf-gcc (crosstool-NG esp-2022r1-RC1) 11.2.0
- Operating System:     [Ubuntu 20.04 LTS]
- Using an IDE: [VSCode]
- Power Supply:         [USB]

## Problem Description
I am trying to see if it is possible to setup a BLE mesh and wifi coexistence on an ESP32-c3-M1 dev kit. According to documentations, it is feasible to run both network protocols on a single RF antenna module. To start things off, i tried running [BLE/Wifi Coexistence](https://github.com/espressif/esp-idf/tree/572e79530c467bb40ba6617ba021666c9871e19b/examples/bluetooth/esp_ble_mesh/ble_mesh_wifi_coexist) and when i tried using the board as a station with TCP server to communicate with my TCP client(PC) i am getting lots of disconnectivity issues and packets dropped as shown
```
esp> I (32065) iperf: accept: 192.168.176.65,42394

        Interval Bandwidth
W (34222) wifi:m f null

   0-   3 sec       1.24 Mbits/sec
   3-   6 sec       0.00 Mbits/sec
W (40602) wifi:m f probe req l=0

   6-   9 sec       0.00 Mbits/sec
W (41103) wifi:m f probe req l=0

W (41604) wifi:m f probe req l=0

W (42105) wifi:m f probe req l=0

W (42606) wifi:m f probe req l=0

W (43804) iperf: tcp server recv error, error code: 11, reason: No more processes
I (43807) iperf: TCP Socket server is closed.
I (43808) iperf: iperf exit
```
```
esp> I (32065) iperf: accept: 192.168.176.65,42394

        Interval Bandwidth
W (34222) wifi:m f null

   0-   3 sec       1.24 Mbits/sec
   3-   6 sec       0.00 Mbits/sec
W (40602) wifi:m f probe req l=0

   6-   9 sec       0.00 Mbits/sec
W (41103) wifi:m f probe req l=0

W (41604) wifi:m f probe req l=0

W (42105) wifi:m f probe req l=0

W (42606) wifi:m f probe req l=0

W (43804) iperf: tcp server recv error, error code: 11, reason: No more processes
I (43807) iperf: TCP Socket server is closed.
I (43808) iperf: iperf exit
```

Meanwhile, if i run just this [iperf example](https://github.com/espressif/esp-idf/tree/572e79530c467bb40ba6617ba021666c9871e19b/examples/wifi/iperf) without BLE, i am getting strong connectivity on wifi as shown
```
iperf> I (28935) iperf: accept: 192.168.176.65,54706

        Interval Bandwidth
   0-   3 sec       2.18 Mbits/sec
   3-   6 sec       1.38 Mbits/sec
   6-   9 sec       1.95 Mbits/sec
   9-  12 sec       3.43 Mbits/sec
  12-  15 sec       2.75 Mbits/sec
  15-  18 sec       1.77 Mbits/sec
  18-  21 sec       4.26 Mbits/sec
  21-  24 sec       3.66 Mbits/sec
  24-  27 sec       3.08 Mbits/sec
  27-  30 sec       3.22 Mbits/sec
  30-  33 sec       2.94 Mbits/sec
  33-  36 sec       1.09 Mbits/sec
  36-  39 sec       3.78 Mbits/sec
  39-  42 sec       2.41 Mbits/sec
  42-  45 sec       3.23 Mbits/sec
  45-  48 sec       1.82 Mbits/sec
  48-  51 sec       2.36 Mbits/sec
  51-  54 sec       2.40 Mbits/sec
  54-  57 sec       0.81 Mbits/sec
  57-  60 sec       4.13 Mbits/sec
```

What is the issue here? Am i not configuring something correctly or is it simply the hardware's limitations? If so why did the documentation claim to be able to allow coexistence? It seems as if during the BLE scan, it is occupying alot of the TX/RX and it does not leave room for the wifi. I am unable to debug this any further and i would hope that someone can shed light on this issue. Thank you.

wmy-espressif, 2022-09-23:
Hi @Shane555 , I am sorry that I don't reproduce the issue locally using ESP32-C3. But I can think of one reason for the disconnection of Wi-Fi link in your test: lack of memory.

For higher throughput, the sdkconfig.defaults.xx files of iperf example sets Wi-Fi many Tx/Rx buffers and large LWIP TCP/UDP Tx window, which will consume a lot of memory. But if you enable BLE mesh at the same time, the memory may not be enough for the project, and if Wi-Fi cannot allocate memory for Rx buffer, beacon timeout can occur and the link can disconnect.

I suggest you to try reducing the Wi-Fi buffer numbers to save more memory. You can use these configurations as in the attached file: 
[sdkconfig.defaults.zip](https://github.com/espressif/esp-idf/files/9632762/sdkconfig.defaults.zip)

Shane555, 2022-09-23:
@wmy-espressif could I try using wrover module? They have external ram inbuilt right? Would that work? I also heard it is Bluetooth 4.2 and not 5.0

wmy-espressif, 2022-09-23:
Hi @Shane555 if you need large memory you can try modules with PSRAM, I don't know if Bluetooth 5.0 is necessary for you, if yes you can take a look at the ESP32-S3 module series: https://www.espressif.com/en/products/modules

Shane555, 2022-10-06:
Hi @wmy-espressif for wrover board, are there any specific configuration settings to ensure wifi and ble mesh coexistence work seamlessly? In particular are there any configurations that you used which verified it indeed works? ESP claims that coexistence works but so far i havent been able to get it working even on the wrover. This time round the RSSI signal from ble cannot be detected when iperf is sending data via wifi. 
[sdkconfig.zip](https://github.com/espressif/esp-idf/files/9721231/sdkconfig.zip)

wmy-espressif, 2022-10-09:
Hi @Shane555 , here is an introduction and tutorial for espressif RF coexistence design: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/coexist.html, there are some menuconfig settings and available APIs suggested to use that can possibly help improve the coexistence performance. Can you please first check this document?

ESP32, ESP32-C3 and ESP32-S3 all have only one RF path which has to be shared by Wi-Fi and Bluetooth. We cannot expect a same level of performance as we are using BLE or Wi-Fi alone. And some parameters chosen may count. 

For the issue you reported, I don't find risky items in your sdkconfig file that can block coexistence.  Will you please provide more details on your test condition?
   1. Are there any other Bluetooth activities besides BLE advertising?
   2. What is the BLE advertising interval? What is the BLE scan parameter on the peer device?
   3. Wi-Fi role, are you using STA mode or Soft-AP mode?

Shane555, 2022-10-10:
@wmy-espressif 
1) Not that i know of any.
2) I am not particular sure about this as i was using an android phone to scan for the rssi of the esp device.
3) STA mode

But comparing the rssi signals, i could not detect anything when coexistence is on and iperf is running. Whereas the signal is stable at a fixed distance when i was using solely ble mesh

Shane555, 2022-10-11:
Hi, thanks for the help. I managed to get it provisioning although the rssi signal is weak. It does work now.

closed 2022-10-11 by Shane555