espressif/esp-idf #9740: Touch Pad read not working properly on esp32s3

nikhil-robinson, 2022-09-07:
I have been testing the touchpad read example from the ESP_IDF examples. I am trying to read touchpad 10,11,12,13 but the values are getting stuck on  these set of numbers no change.

IDF version 4.4.2
using vscode extension

[image]

FloHomi, 2022-12-22:
I discovered the same issue IDF5.0

5ami, 2023-01-05:
It might get stuck if the capacitance you are trying to read is to big. If you have bigger capacitors try to reduce number of charge and discharge cycles

closed 2023-06-28 by nikhil-robinson

brainstorm, 2024-07-28:
Be mindful of transient issues and re-init (or delay the init in the worst case) if the initial values are clearly out of range, see: https://github.com/micropython/micropython/issues/13178#issuecomment-2254331069