espressif/esp-idf #9742: how i can disable provisioning process after finish provisioner UUID

xuanchibkd14, 2022-09-07:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

hi, i'm testing provisioner node, it worked, but i don't know how to turn off the provisioner after successful provisioning, if using disable function  `esp_ble_mesh_provisioner_prov_disable(ESP_BLE_MESH_PROV_ADV | ESP_BLE_MESH_PROV_GATT)`
the device can't work, but if not disable i can't set up a new mesh network because it will be provisioned by old provisioner.
tks for help.

xuanchibkd14, 2022-09-08:
I found the solution for this, just disable `recv_unprov_adv_pkt()` in `ble_mesh_provisioning_cb`

JiaLWang, 2022-09-18:
Hi @xuanchibkd14

For provisioner example, you can change the [match uuid](https://github.com/espressif/esp-idf/blob/master/examples/bluetooth/esp_ble_mesh/ble_mesh_provisioner/main/main.c#L587).
And for the node that want to enter the mesh network, the [dev_uuid](https://github.com/espressif/esp-idf/blob/master/examples/bluetooth/esp_ble_mesh/ble_mesh_node/onoff_server/main/main.c#L33) need also change and same with the match uuid.

So you can using different match uuid to create different mesh netwoek.

Hope it can help you.

xuanchibkd14, 2022-09-19:
> Hi @xuanchibkd14
>
> For provisioner example, you can change the [match uuid](https://github.com/espressif/esp-idf/blob/master/examples/bluetooth/esp_ble_mesh/ble_mesh_provisioner/main/main.c#L587). [...]

tks for help

closed 2022-09-19 by xuanchibkd14