espressif/esp-idf #9744: Out of date CJSON

bvanhoose14, 2022-09-07:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

all

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### What is the expected behavior?

The cJSON submodule is out of date. Since then, they have patched a memory leak that can be problematic. While I can manually update it myself, we have multiple people using it. If someone misses it or forgets to do it on a fresh install it can cause major problems. Would you guys consider updating cJSON in one of your team's next releases? I'm sure there are other people that don't even realize they have this memory leak as well. Thanks.

### What is the actual behavior?

n/a

### Steps to reproduce.

1. Step
2. Step
3. Step
...

AxelLin, 2022-09-07:
Current esp-idf is using cJSON-v1.7.15 which is latest release version on upstream.
If you need bugfix after cJSON-v1.7.15, I think you need to ask upstream to tag a new release.

igrr, 2022-09-08:
One thing to note, in this comment one of the contributors has mentioned that they consider that specific memory leak as an edge case, which shouldn't affect the majority of applications: https://github.com/DaveGamble/cJSON/commit/f50dafc7d0bfd4f45449ab5665bfea831a82f2eb#commitcomment-55507388.

Perhaps if that is not the case and the memory leak might actually happen, an issue should be opened at https://github.com/DaveGamble/cJSON/issues to ask the maintainers to tag a new release.

AxelLin, 2022-10-02:
> ### What is the expected behavior?
> The cJSON submodule is out of date. Since then, they have patched a memory leak that can be problematic. While I can manually update it myself, we have multiple [...]

Hi @bvanhoose14 
Any update? It's not clear to me if you are talking about the memory leak in error path.
If this is still an issue, please illustrate what is exactly the **major** issue you hit.

mahavirj, 2022-10-06:
@bvanhoose14 We will prefer to update to next stable release of cJSON. Please help to file an upstream issue asking for the new release of this library. Will close this ticket interim of that.

closed 2022-10-06 by mahavirj

AxelLin, 2023-07-08:
> @bvanhoose14 We will prefer to update to next stable release of cJSON. Please help to file an upstream issue asking for the new release of this library. Will close this ticket interim of that.

@mahavirj 
FYI, cJSON 1.7.16 is released which include memory leak fix for merge_patch, https://github.com/DaveGamble/cJSON/releases/tag/v1.7.16
So it's probable time to upgrade cJSON in esp-idf.

mahavirj, 2023-07-12:
@AxelLin Thanks for notifying. We will handle the upgrade.

AxelLin, 2024-01-06:
@mahavirj 
FYI, cJSON 1.7.17 is released which includes CVE fixes.
https://github.com/DaveGamble/cJSON/releases/tag/v1.7.17

AxelLin, 2024-05-14:
@mahavirj 
FYI, cJSON 1.7.18 is released which includes CVE fixes.
https://github.com/DaveGamble/cJSON/releases/tag/v1.7.18

AxelLin, 2025-09-09:
@mahavirj 
FYI, cJSON 1.7.19 is released which includes CVE fixes.
https://github.com/DaveGamble/cJSON/releases/tag/v1.7.19

mahavirj, 2025-09-10:
@AxelLin We had updated the submodule pointer to address couple of CVEs. We will update to this formal release now.