espressif/esp-idf #9748: OTA after SmartConfig not working. example_connect: Waiting for IP(s)

gomez9656, 2022-09-07:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

ESP32-C3-DevKitM1-v1.0

### Power Supply used.

USB

### What is the expected behavior?

Hello Espressif team :)

My project uses OTA(advanced_https_ota), SmartConfig, and BLE-MESH. 

An overview of the project is that a user can control several lights using BLE-MESH, and at any given time, he/she can start an OTA process. 

The expected behavior when the user starts SmartConfig is:
 1. start SmartConfig
 2. receive WiFi credentials using the EspTouch app
 3. start OTA 
 4. OTA example download binary file and restart device with new binary

### What is the actual behavior?

The actual behavior is that after SmartConfig finished "smartconfig_example: smartconfig over" the OTA functions start their work but they don't go further "example_connect: Waiting for IP(s)"

### Steps to reproduce.

1. Add BLE-MESH, OTA, and SmartConfig code into the same project
2. integrate the SDKConfig file(attached here)
3. For simplicity and testing: start SmartConfig first, once it is finished, start OTA
4. this is all being done while the server with the binary file is up and running 

 
[sdkconfig.txt](https://github.com/espressif/esp-idf/files/9513823/sdkconfig.txt)

### Debug Logs.

```plain
I (30) boot: ESP-IDF v4.4-dirty 2nd stage bootloader
I (30) boot: compile time 13:23:07
I (30) boot: chip revision: 3
I (32) boot.esp32c3: SPI Speed      : 80MHz
I (37) boot.esp32c3: SPI Mode       : DIO
I (42) boot.esp32c3: SPI Flash Size : 4MB
I (46) boot: Enabling RNG early entropy source...
I (52) boot: Partition Table:
I (55) boot: ## Label            Usage          Type ST Offset   Length
[... 153 lines cut ...]

    uint8_t ssid[32] = "TIGO-8E48";
    uint8_t  pass64] = "4D9697504814";
    wifi_config_t wifi_config = {
        .sta = {
            .scan_method = EXAMPLE_WIFI_SCAN_METHOD,
            .sort_method = EXAMPLE_WIFI_CONNECT_AP_SORT_METHOD,
            .threshold.rssi = CONFIG_EXAMPLE_WIFI_SCAN_RSSI_THRESHOLD,
            .threshold.authmode = EXAMPLE_WIFI_SCAN_AUTH_MODE_THRESHOLD,
        },
    };
    memcpy(wifi_config.sta.ssid, ssid, sizeof(ssid));
    memcpy(wifi_config.sta.password, pass, sizeof(pass));`

I think the problem is that SmartConfig connects to the WiFi and when OTA starts it also connects to the WiFI, so problems arise in this part, and the device restarts. I added this code which solved that part but doesn't go further "  example_connect: Waiting for IP(s)"

     esp_event_loop_delete_default();
     esp_netif_destroy(sta_netif);
     esp_wifi_disconnect();
     esp_wifi_stop();
     esp_wifi_deinit();`

Overall, I think the way to solve it is to correctly deinit and reinit WiFi after SmartConfig is over, but I'm not sure if this is the correct way to do it.

Thanks a lot in advance Espressif team :)

gomez9656, 2022-09-13:
Any help would be really appreciated :)

mhdong, 2024-02-01:
Hi @gomez9656 you are right, SmartConfig connects to the WiFi and when OTA starts it also connects to the WiFI. The router assumes that the previous connection is still present and will reject the current connection. So disconnect first, then try connecting again can solve the issue.

closed 2024-02-19 by Alvin1Zhang