espressif/esp-idf #9750: [Websockets] ws api is incomplete - there's no way to refuse a websocket request?

chipweinberger, 2022-09-08:
### Describe the solution you'd like.

**ESP-IDF 4.4.2
ESP32-S3**

To accept a websocket, servers must reply with an upgrade request:

```
HTTP/1.1 101 Switching Protocols
Upgrade: websocket
Connection: Upgrade
Sec-WebSocket-Accept: HSmrc0sMlYUkAGmm5OPpG2HaGWk=
Sec-WebSocket-Protocol: chat
```

This "success" response happens automatically as far as I can determine. 
https://github.com/espressif/esp-idf/blob/master/components/esp_http_server/src/httpd_ws.c

However, there is no way to refuse a connection in my testing. returning ESP_FAIL, or setting the code to `HTTP 500` does not prevent the `HTTP/1.1 101 Switching Protocols` code response from being sent. 

### Additional context.

[david-cermak](https://github.com/david-cermak)

chipweinberger, 2022-09-08:
[david-cermak](https://github.com/david-cermak)

chipweinberger, 2022-09-08:
In my testing:

This code still sends the HTTP 101 upgrade request, but results in chrome complaining about invalid frames. 
```
if (req->method == HTTP_GET) {
  httpd_resp_set_status(req, "500 Internal Server Error");
  httpd_resp_set_type(req, "application/json");
  httpd_resp_sendstr(req, json);
  return ESP_FAIL;
}
```

This code also results in a HTTP 101 upgrade request, but causes the connection to immediately close.
```
if (req->method == HTTP_GET) {
  return ESP_FAIL;
}
```

chipweinberger, 2022-09-08:
In terms of API, I think it would be simplest to register 2 handlers. 

`HTTP_GET`
`HTTP_WS_FRAME`

The GET handler would be optional, and would let the user decide how to respond themselves.

ESP-YJM, 2022-09-20:
@chipweinberger When the code flow enter the `httpd_uri_t` callback function(wsHandler you mention above), the websocket handshake is finished and already send `HTTP/1.1 101 Switching Protocols` to peer. So it is late to send error code to peer. The response send is in here https://github.com/espressif/esp-idf/blob/master/components/esp_http_server/src/httpd_uri.c#L316. And the callback function is in here https://github.com/espressif/esp-idf/blob/master/components/esp_http_server/src/httpd_uri.c#L329. If you want to refuse the websocket request, you can unregister the websocket `httpd_uri_t` and will return `404 NOT found`. I think it is a easy way to refuse the websocket requests.

chipweinberger, 2022-09-20:
Thanks for taking time to describe a solution @ESP-YJM 

I did not think of that approach. It is not perfect, but reasonable.

Thanks.

closed 2022-09-20 by chipweinberger