espressif/esp-idf #9751: cannot get the animation

XiaoLiuAI, 2022-09-08:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1-dev-286-gbcb34ca7ae

### Operating System used.

macOS

### How did you build your project?

VS Code IDE

### What is the expected behavior?

See the animation described in README. 

### What is the actual behavior?

After flush and press the reset button, I got LCD back light on, but no animation is presented.  

### Steps to reproduce.

1. add `add_definitions("-DLV_LVGL_H_INCLUDE_SIMPLE")` into main/CMakeLists.txt
2. add `#include "rom/gpio.h"` into main/main.c
3. copy managed_components into components, and change `%d` into `%ld` in some file due to the compiler warning.
4. run `idf.py PORT build flush monitor`

I used the LilyGo T-Display S3(esp-s3 16M + 1.9inch LCD).

### Build or installation Logs.

```plain
-- Found Git: /usr/bin/git (found version "2.32.0 (Apple Git-132)") 
-- Component directory /Users/xiaoliu/ExSpace/esp/esp32s3_lilygo_8bit_parallel_display_lvgl/components/lvgl does not contain a CMakeLists.txt file. No component will be added
-- The C compiler identification is GNU 11.2.0
-- The CXX compiler identification is GNU 11.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /Users/xiaoliu/.espressif/tools/xtensa-esp32s3-elf/esp-2022r1-11.2.0/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /Users/xiaoliu/.espressif/tools/xtensa-esp32s3-elf/esp-2022r1-11.2.0/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /Users/xiaoliu/.espressif/tools/xtensa-esp32s3-elf/esp-2022r1-11.2.0/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32s3
Processing 1 dependencies:
[1/1] idf (5.1.0)
-- Project sdkconfig file /Users/xiaoliu/ExSpace/esp/esp32s3_lilygo_8bit_parallel_display_lvgl/sdkconfig
Loading defaults file /Users/xiaoliu/ExSpace/esp/esp32s3_lilygo_8bit_parallel_display_lvgl/sdkconfig.defaults...
-- Looking for sys/types.h
-- Looking for sys/types.h - found
-- Looking for stdint.h
-- Looking for stdint.h - found
-- Looking for stddef.h
-- Looking for stddef.h - found
-- Check size of time_t
-- Check size of time_t - done
-- Found Python3: /Users/xiaoliu/.espressif/python_env/idf5.1_py3.10_env/bin/python (found version "3.10.4") found components: Interpreter 
-- Performing Test C_COMPILER_SUPPORTS_WFORMAT_SIGNEDNESS
-- Performing Test C_COMPILER_SUPPORTS_WFORMAT_SIGNEDNESS - Success
[... 184 lines cut ...]
      |                                                                                                                                   ^
/Users/xiaoliu/ExSpace/esp/esp32s3_lilygo_8bit_parallel_display_lvgl/components/lvgl__lvgl/src/core/lv_indev.c:346:91: warning: suggest braces around empty body in an 'if' statement [-Wempty-body]
  346 |     if(data->point.y < 0) LV_LOG_WARN("Y is %d which is smaller than zero", data->point.y);
      |                                                                                           ^
/Users/xiaoliu/ExSpace/esp/esp32s3_lilygo_8bit_parallel_display_lvgl/components/lvgl__lvgl/src/core/lv_indev.c:347:131: warning: suggest braces around empty body in an 'if' statement [-Wempty-body]
  347 |     if(data->point.y >= lv_disp_get_ver_res(i->driver->disp)) LV_LOG_WARN("Y is %d which is greater than hor. res", data->point.y);
      |                                                                                                                                   ^
[995/1007] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/main.c.obj
/Users/xiaoliu/ExSpace/esp/esp32s3_lilygo_8bit_parallel_display_lvgl/main/main.c: In function 'app_main':
/Users/xiaoliu/ExSpace/esp/esp32s3_lilygo_8bit_parallel_display_lvgl/main/main.c:220:15: warning: unused variable 'scr' [-Wunused-variable]
  220 |     lv_obj_t *scr = lv_disp_get_scr_act(disp);
      |               ^~~
At top level:
/Users/xiaoliu/ExSpace/esp/esp32s3_lilygo_8bit_parallel_display_lvgl/main/main.c:85:13: warning: 'main_creatSysteTasks' defined but not used [-Wunused-function]
   85 | static void main_creatSysteTasks(void)
      |             ^~~~~~~~~~~~~~~~~~~~
[1006/1007] Generating binary image from built executable
esptool.py v4.2.1
Creating esp32s3 image...
Merged 2 ELF sections
Successfully created esp32s3 image.
Generated /Users/xiaoliu/ExSpace/esp/esp32s3_lilygo_8bit_parallel_display_lvgl/build/lcd_lvgl.bin
[1007/1007] cd /Users/xiaoliu/ExSpace/esp/esp32s3_lilygo_8bit_parallel_display...aoliu/ExSpace/esp/esp32s3_lilygo_8bit_parallel_display_lvgl/build/lcd_lvgl.bin
lcd_lvgl.bin binary size 0x60800 bytes. Smallest app partition is 0x177000 bytes. 0x116800 bytes (74%) free.
```

tore-espressif, 2022-09-08:
@XiaoLiuAI could you share your code? From the issue description I have no idea what example or other app you are running

XiaoLiuAI, 2022-09-08:
@tore-espressif I used this repository https://github.com/UsefulElectronics/esp32s3_lilygo_8bit_parallel_display_lvgl.git , which claimed to drive the board and screen directly on the lvgl and esp-idf (the official example is based on arduino-esp & lvgl & TFT-eSPI, but I prefer to use esp-idf directly without the arduino wrapper)

tore-espressif, 2022-09-08:
We don't maintain that code. You can try filing an issue there https://github.com/UsefulElectronics/esp32s3_lilygo_8bit_parallel_display_lvgl/issues

XiaoLiuAI, 2022-09-08:
@tore-espressif 
Thanks. I am sorry, I thought I have posted this issue there, but in the wrong repository.

closed 2022-09-08 by Alvin1Zhang