espressif/esp-idf #9752: ESP32s3 Bluetooth example

farbodpm, 2022-09-08:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4

### Operating System used.

Linux

### How did you build your project?

Other (please specify in More Information)

### Development Kit.

Custome Board

### Power Supply used.

USB

### What is the expected behavior?

I expect when advertising event come I can find my esp bluetooth name in other phones.

### What is the actual behavior?

I am trying to run farr_server example. The code run fine in esp32 so that my core is okay. but when I change my board which is a custom esp32s3 board I can not find it.

### Steps to reproduce.

1. attach power
2. send adv configs
3. wait for advertise start event
4. check the bluetooth name in my devices.

### Debug Logs.

```plain
I (839) boot: ESP-IDF v4.4-dev-3540-g4e03a9c34c 2nd stage bootloader
I (839) boot: compile time 09:24:35
I (839) boot: chip revision: 0
I (862) boot.esp32s3: Boot SPI Speed : 80MHz
I (862) boot.esp32s3: SPI Mode       : DIO
I (862) boot.esp32s3: SPI Flash Size : 8MB
I (865) boot: Enabling RNG early entropy source...
W (870) bootloader_random: RNG for ESP32-S3 not currently supported
I (877) boot: Partition Table:
I (881) boot: ## Label            Usage          Type ST Offset   Length
I (888) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (896) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (904) boot:  2 factory          factory app      00 00 00010000 007f0000
I (911) boot: End of partition table
I (915) esp_image: segment 0: paddr=00010020 vaddr=3c0d0020 size=38ca4h (232612) map
I (966) esp_image: segment 1: paddr=00048ccc vaddr=3fc93820 size=03974h ( 14708) load
I (969) esp_image: segment 2: paddr=0004c648 vaddr=40374000 size=039d0h ( 14800) load
I (975) esp_image: segment 3: paddr=00050020 vaddr=42000020 size=c0228h (786984) map
I (1121) esp_image: segment 4: paddr=00110250 vaddr=403779d0 size=0be4ch ( 48716) load
I (1132) esp_image: segment 5: paddr=0011c0a4 vaddr=50000000 size=00010h (    16) load
I (1139) boot: Loaded app from partition at offset 0x10000
I (1139) boot: Disabling RNG early entropy source...
W (1141) bootloader_random: RNG for ESP32-S3 not currently supported
I (1160) cpu_start: Pro cpu up.
I (1160) cpu_start: Starting app cpu, entry point is 0x4037650c
0x4037650c - call_start_cpu1
    at /home/esp/esp-idf-ble/.embuild/espressif/esp-idf/release-v4.4/components/esp_system/port/cpu_start.c:148
I (0) cpu_start: App cpu up.
I (1174) cpu_start: Pro cpu start user code
I (1174) cpu_start: cpu freq: 160000000
I (1174) cpu_start: Application information:
I (1177) cpu_start: Project name:     libespidf
I (1182) cpu_start: App version:      1
I (1187) cpu_start: Compile time:     Sep  7 2022 10:13:31
I (1193) cpu_start: ELF file SHA256:  0000000000000000...
I (1199) cpu_start: ESP-IDF:          6c44001
I (1204) heap_init: Initializing. RAM available for dynamic allocation:
I (1211) heap_init: At 3FC9C060 len 0004D6B0 (309 KiB): D/IRAM
I (1218) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DRAM
I (1225) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (1231) heap_init: At 600FE000 len 00002000 (8 KiB): RTCRAM
I (1238) spi_flash: detected chip: winbond
I (1242) spi_flash: flash io: dio
I (1247) sleep: Configure to isolate all GPIO pins in sleep state
I (1253) sleep: Enable automatic switching of GPIO sleep configuration
I (1260) coexist: coexist rom version e7ae62f
I (1265) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (1286) esp_idf_svc::eventloop: Dropped
I (1286) esp_idf_svc::netif: Dropped
I (1306) BT_INIT: BT controller compile version [3a49744]
I (1306) phy_init: phy_version 503,13653eb,Jun  1 2022,17:47:08
I (1346) system_api: Base MAC address is not set
I (1346) system_api: read default base MAC address from EFUSE
I (1346) BT_INIT: Bluetooth MAC: f4:12:fa:44:a7:d2

I (1346) esp_idf_ble: init bluetooth
I (1376) esp_idf_ble: register_gatt_service_application enter
I (1376) esp_idf_ble: Called gatt service event handler with gatts_if: 3, event { Register { status: 0, app_id: 1 } }
I (1376) esp_idf_ble: register_gatt_service_application exit
I (1386) gatt_server: application registered
I (1386) gatt_server: GattService to be created: GattService { is_primary: true, id: Uuid16(255), instance_id: 1, handle: 4 }
I (1406) esp_idf_ble: Called gatt service event handler with gatts_if: 3, event { Create }
I (1416) gatt_server: SVC Handle: 40
I (1416) esp_idf_ble: Called gatt service event handler with gatts_if: 3, event { StartComplete }
I (1426) esp_idf_ble: Called gatt service event handler with gatts_if: 3, event { AddCharacteristicComplete }
I (1436) gatt_server: Characteristic Attribute handle: 42
I (1446) esp_idf_ble: len: 12, data: 0x3fca6d3c
I (1446) gatt_server: Characteristic values: [72, 101, 108, 108, 111, 32, 87, 111, 114, 108, 100, 0]
I (1456) esp_idf_ble: Called gatt service event handler with gatts_if: 3, event { AddDescriptorComplete }
I (1466) gatt_server: Descriptor Attribute handle: 43
I (1476) esp_idf_ble: configure_advertising enter
I (1476) esp_idf_ble: svc_uuid: { [251, 52, 155, 95, 128, 0, 0, 128, 0, 16, 0, 0, 255, 0, 0, 0] }
I (1486) esp_idf_ble: 0:fb
I (1496) esp_idf_ble: 1:34
I (1496) esp_idf_ble: 2:9b
I (1496) esp_idf_ble: 3:5f
I (1506) esp_idf_ble: Configuring advertising with { esp_ble_adv_data_t { set_scan_rsp: false, include_name: true, include_txpower: false, min_interval: 6, max_interval: 16, appearance: 0, manufacturer_len: 0, p_manufacturer_data: 0x0, service_data_len: 0, p_service_data: 0x0, service_uuid_len: 16, p_service_uuid: 0x3fcf4b64, flag: 6 } }
I (1536) esp_idf_ble: Called gap event handler with event { AdvertisingDatasetComplete }
I (1546) esp_idf_ble: configure_advertising exit
I (1546) gatt_server: advertising configured
I (1556) esp_idf_ble: configure_advertising enter
I (1556) esp_idf_ble: svc_uuid: { [251, 52, 155, 95, 128, 0, 0, 128, 0, 16, 0, 0, 255, 0, 0, 0] }
I (1566) esp_idf_ble: 0:fb
I (1566) esp_idf_ble: 1:34
I (1576) esp_idf_ble: 2:9b
I (1576) esp_idf_ble: 3:5f
I (1576) esp_idf_ble: Configuring advertising with { esp_ble_adv_data_t { set_scan_rsp: true, include_name: false, include_txpower: true, min_interval: 0, max_interval: 0, appearance: 0, manufacturer_len: 0, p_manufacturer_data: 0x0, service_data_len: 0, p_service_data: 0x0, service_uuid_len: 16, p_service_uuid: 0x3fcf4b64, flag: 0 } }
I (1616) esp_idf_ble: Called gap event handler with event { ScanResponseDatasetComplete }
I (1616) esp_idf_ble: configure_advertising exit
I (1626) esp_idf_ble: start_advertise enter
I (1636) esp_idf_ble: Called gap event handler with event { AdvertisingStartComplete }
I (1636) esp_idf_ble: start_advertise exit
I (1646) gatt_server: advertising started
```

### More Information.

I have tested my code on esp32 and it works fine.

dongmenhai, 2022-09-15:
@farbodpm 
Hi, which example is farr_server? You can run [gatt_server_service_table example](https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/bluedroid/ble/gatt_server_service_table) on a custom esp32s3 board and use the mobile app to scan to see if you can scan the broadcast sent by esp32s3?
Is the hardware antenna good? The distance between the scanning device and the broadcasting device can be pulled in to see if the results have changed

farbodpm, 2022-09-16:
I checked the example and no progress. I put both devices near but no broad casting.

xulongzhuang, 2022-12-08:
> I checked the example and no progress. I put both devices near but no broad casting.

@farbodpm  Is there any progress on this issue? 
You can run the [gatt_server ](https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/bluedroid/ble/gatt_server)or [gatt_server_service_table ](https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/bluedroid/ble/gatt_server_service_table)example on esp32s3 and upload the running log.
If you still cannot scan the broadcast named "ESP_GATTS_DEMO" using the mobile app, it may be affected by hardware or antenna

zhp0406, 2024-02-26:
@farbodpm 
has the issue been resolved?

closed 2024-03-10 by Alvin1Zhang