espressif/esp-idf #9753: BLE Mesh: Dynamic Configuration of node property

Shane555, 2022-09-08:
### Is your feature request related to a problem?

Currently, the node properties like LPN, friend, relay cannot be set at run time. We have to go into menuconfig and enable or disable these functions. I would like the capability to do so at run time e.g. change this relay node to a low power node while the program is running. I noticed there is a function `esp_err_t esp_ble_mesh_config_client_set_state(esp_ble_mesh_client_common_param_t *params, esp_ble_mesh_cfg_client_set_state_t *set_state)` that could use a configuration client model to modify the configuration server parameters, which seems like it might be possible to do so at run time. However, `esp_ble_mesh_cfg_client_set_state_t` only contains `esp_ble_mesh_cfg_relay_set_t` and `esp_ble_mesh_cfg_friend_set_t` but does not have something for the LPN equivalent.

 Another thing is i am not sure if simply setting say the esp_ble_mesh_cfg_friend_set_t to TRUE would allow the node to be a friend node instantly. Do clarify regarding this issue   

### Describe the solution you'd like.

I would like to be able to flash the firmware once and have the node be able to do friendship, LPN or relay as and when i would want to without reflashing.

Shane555, 2022-09-16:
Regarding the APIs provided for a configuration client model node to send a message to configure another configuration server model node, it seems like there are things missing.  For the client side to send a message i will need to use `sp_err_t esp_ble_mesh_config_client_set_state(esp_ble_mesh_client_common_param_t *params,
                                               esp_ble_mesh_cfg_client_set_state_t *set_state)`

Shane555, 2022-09-16:
I am not sure if you can simply modify `esp_ble_mesh_cfg_server_state_change_t` to account for the friend and relay component but i believe it should work without a hiccup 
```cpp
typedef union {
    /**
     * The recv_op in ctx can be used to decide which state is changed.
     */
    esp_ble_mesh_state_change_cfg_mod_pub_set_t         mod_pub_set;        /*!< Config Model Publication Set */
    esp_ble_mesh_state_change_cfg_model_sub_add_t       mod_sub_add;        /*!< Config Model Subscription Add */
    esp_ble_mesh_state_change_cfg_model_sub_delete_t    mod_sub_delete;     /*!< Config Model Subscription Delete */
    esp_ble_mesh_state_change_cfg_netkey_add_t          netkey_add;         /*!< Config NetKey Add */
    esp_ble_mesh_state_change_cfg_netkey_update_t       netkey_update;      /*!< Config NetKey Update */
    esp_ble_mesh_state_change_cfg_netkey_delete_t       netkey_delete;      /*!< Config NetKey Delete */
    esp_ble_mesh_state_change_cfg_appkey_add_t          appkey_add;         /*!< Config AppKey Add */
    esp_ble_mesh_state_change_cfg_appkey_update_t       appkey_update;      /*!< Config AppKey Update */
    esp_ble_mesh_state_change_cfg_appkey_delete_t       appkey_delete;      /*!< Config AppKey Delete */
    esp_ble_mesh_state_change_cfg_model_app_bind_t      mod_app_bind;       /*!< Config Model App Bind */
    esp_ble_mesh_state_change_cfg_model_app_unbind_t    mod_app_unbind;     /*!< Config Model App Unbind */
    esp_ble_mesh_state_change_cfg_kr_phase_set_t        kr_phase_set;       /*!< Config Key Refresh Phase Set */

/* ADDED THIS PART IN TO ALLOW FOR MODIFICATION OF RELAY/FRIEND*/
   esp_ble_mesh_state_change_cfg_friend_set_t     friend_state_set
   esp_ble_mesh_state_change_cfg_relay_set_t    relay_state_set 
} esp_ble_mesh_cfg_server_state_change_t;
```

However, do note that there is still no LPN(Low power node) modification whatsoever which i believe should be allowed in the program for dynamic configuration when the systems are live

JiaLWang, 2022-09-18:
Hi @Shane555 

Sorry for the late reply. 
Enable the relay or friend option in the menuconfig. It just means that the device supports the feature. Whether this function is enabled depends on the default configuration of [configuration server model](https://github.com/espressif/esp-idf/blob/master/examples/bluetooth/esp_ble_mesh/ble_mesh_node/onoff_server/main/main.c#L35). And if you want to change the state at run time, you can use configuration client to send message by esp_ble_mesh_config_client_set_state() function to configuration server.
About the struct of esp_ble_mesh_cfg_server_state_change_t in server side, compared with client_set_msg , there are indeed many fewer events. These events are only used to notify the application layer that the underlying cfg_srv state has changed.
We will add the corresponding state change event later.

But for the lpn feature, more available for nodes with low power consumption requirements. You can enable the option in menuconfig, and once the node enters the mesh network, it will try to find a Friend node and establish a friendship. In mesh spec, doesn't support disable low power feature when the node support it. So there is no lpn_set in the struct.

Hope it can help you.

Shane555, 2022-09-19:
@JiaLWang I will save you the trouble and modify the struct in server side to add new events

closed 2022-11-28 by Alvin1Zhang