espressif/esp-idf #9755: ESP Modem crash when GNSS is started

RilabsAutomotive, 2022-09-08:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0-dev-3290-g01d014c42d

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

Custom board with ESP32S3

### Power Supply used.

Battery

### What is the expected behavior?

I expected that GNSS is started, modem SIM7070G is started and runs normally

### What is the actual behavior?

A crash occurs whenever the GNSS is started and trying to get fix status.

### Steps to reproduce.

1. Power Off 
2. Power On
3. Set baudrate
4. Set configuration
5. Set CMUX mode
6. Attach to network
7. Wait for IP
8. Start GNSS
9. Connect to MQTT Backend over TLS (may be does not affect)

### Debug Logs.

```plain
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x2f (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd03e0,len:0x32c0
load:0x403b6000,len:0xbb8
load:0x403b9000,len:0x4c0c
entry 0x403b6260
W (30) boot.esp32s3: eFuse virtual mode is enabled. If Secure boot or Flash encryption is enabled then it does not provide any security. FOR TESTING ONLY!
I (46) boot: ESP-IDF v5.0-dev-3290-g01d014c42d-dirty 2nd stage bootloader
I (47) boot: compile time 19:16:19
I (47) boot: chip revision: 0
I (50) boot.esp32s3: Boot SPI Speed : 80MHz
I (55) boot.esp32s3: SPI Mode       : DIO
I (60) boot.esp32s3: SPI Flash Size : 8MB
I (65) boot: Enabling RNG early entropy source...
[... 260 lines cut ...]
0x42022705: std::function<bool (unsigned char*, unsigned int)>::operator()(unsigned char*, unsigned int) const at c:\espressif\tools\xtensa-esp32s3-elf\esp-2021r2-patch3-8.4.0\xtensa-esp32s3-elf\xtensa-esp32s3-elf\include\c++\8.4.0\bits/std_function.h:687

0x42022852: esp_modem::UartTerminal::task() at C:/Projects/ksc-2n/src/SERVICES/EspModem_Driver/src/esp_modem_uart.cpp:139

0x420229c1: esp_modem::UartTerminal::s_task(void*) at C:/Projects/ksc-2n/src/SERVICES/EspModem_Driver/src/esp_modem_uart.cpp:83

0x40382045: vPortTaskWrapper at C:/Espressif/frameworks/esp-idf-master/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:141

ELF file SHA256: 67b17f8232d892e0

Rebooting...
�ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x2f (SPI_FAST_FLASH_BOOT)
Saved PC:0x42028bf5
0x42028bf5: esp_modem::dce_commands::get_gnss_information_sim70xx(esp_modem::CommandableIf*, gps_s&) at C:/Projects/ksc-2n/src/SERVICES/EspModem_Driver/src/esp_modem_command_library.cpp:760

SPIWP:0xee
mode:DIO, clock div:1
```

### More Information.

GNSS is started via set_gnss_power_mode()
And then the fix status is checked every 100 ms via get_gnss_information_sim70xx()

david-cermak, 2022-09-09:
I have a similar bugfix in progress in https://github.com/espressif/esp-protocols/pull/125 fixing a race condition in the DTE command. Could you please try to cherry-pick the fixing commit (https://github.com/espressif/esp-protocols/pull/125/commits/f673106df7fdbdc6073d530e680539fb83288528) to see if it helps.

Although I'm not sure it would, since according to the backtrace it looks like the internal DTE's signal (SignalGroup) is set to `nullptr`. Might it be  possible that your DTE goes out of scope somewhere and gets destroyed at some point?

RilabsAutomotive, 2022-09-12:
I think that the signal set of "PPP_STARTED" causes the issue.
I update the code in set ang clear functions like this 
void SignalGroup::set(uint32_t bits)
{
  if(event_group != NULL)
  {
    xEventGroupSetBits(event_group, bits);
  }
  else
  {
    ESP_LOGE("ESPDRV", "Signal %d set failed. event_group is NULL", bits);
  }
}
and got the following log entry
E (65111) ESPDRV: Signal 1 set failed. event_group is NULL

sylvioalves, 2022-09-12:
Does the GNSS mode works in PPP mode? I think not, I guess it is missing changing from PPP to Command mode before retrieving GNSS info. Can you check whether this procedure is being followed?

RilabsAutomotive, 2022-09-12:
No, we are using the CMUX mode in order to have the DATA mode(PPP) and the COMMAND mode (AT).

sylvioalves, 2022-09-12:
> No, we are using the CMUX mode in order to have the DATA mode(PPP) and the COMMAND mode (AT). Why do you say that GNSS works only in COMMAND mode.. is this documented somewhere that we've missed ?

I was thinking about BG96. In the past, I had to keep switching between command and PPP mode in order to get GNSS positioning and keep network working. So I wasn't aware about the CMUX option. it seems SIM70x has it available, right?

david-cermak, 2022-10-21:
Same as here. @RilabsAutomotive Could you please check if https://github.com/espressif/esp-protocols/commit/a02bf05eed7c96f703958a3a3f52c6234e386ffb fixes your issue?

RilabsAutomotive, 2023-01-24:
The issue actually came from a limitation of the SIM7070G from SIMCOM. It cannot currently handle mobile data connection and provide GPS coordinates at the same time. The data line drops when a GPS location is requested. The issue is reported to SIMCOM and they are currently investigating it.

closed 2023-03-24 by david-cermak