espressif/esp-idf #9756: ESP32S3: When TWAI start transmit data, SPI W5500 Ethernet slowdown and stop responding

MacWyznawca, 2022-09-08:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

ESP-IDF: v5.0-beta1-dirty
CPU:  ESP32S3
Board: Custom

After TWAI start transmitting data between ESP32S3 with SPI Ethernet w5500 adapter and ESP32, Ethernet slowdown, and stop responding (TWAI works OK).
After disable by software TWAI (twai_stop(); twai_driver_uninstall();) in remote board, Ethernet start working in seconds.

Dazza0, 2022-09-13:
@MacWyznawca Could be something to do with task scheduling and/or interrupts.

Could you check:

- The priorities of the tasks calling the TWAI and Ethernet driver APIs
- On what core the TWAI/Ethernet interrupts are installed (usually on what cores the driver `install()` functions are called on).

MacWyznawca, 2022-09-13:
I was not able to put the TWAI process on the APP-CPU. The attempt produced an error on interrupt allocation.
I left TWAI on CPU 0, and moved all other processes I had control over to CPU 1.

Unfortunately, this did not change anything.

I am now basically sure that not only the SPI/Ethernet process is slowing down dramatically, but all other processes as well.

closed 2022-11-12 by MacWyznawca