espressif/esp-idf #9757: Scheduler crash when returning from interrupt to blocked task

kristinpaget, 2022-09-08:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.2

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### Development Kit.

ESP32_DevKitC_v4

### Power Supply used.

USB

### What is the expected behavior?

When an interrupt returns and the highest-priority task is blocked I expect the idle task to be scheduled instead.

### What is the actual behavior?

Various kinds of crash, including:
```
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x40086495  PS      : 0x00060033  A0      : 0x80086d61  A1      : 0x3ffb6c30  
0x40086495: taskSelectHighestPriorityTaskSMP at /home/kpaget/Documents/Code/TypeCFuzzer/Tests/esp-idf-v4.4.2/components/freertos/tasks.c:3423

A2      : 0x00000000  A3      : 0x00000000  A4      : 0x00000000  A5      : 0x3ffb26ac  
A6      : 0x00000001  A7      : 0x3ffb26b4  A8      : 0x00000002  A9      : 0x3ffb68dc  
A10     : 0x3ffb28a0  A11     : 0x0000001b  A12     : 0x0000001b  A13     : 0x00000000  
A14     : 0x0000000f  A15     : 0x3ffb0b50  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000063  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff  

Backtrace:0x40086492:0x3ffb6c300x40086d5e:0x3ffb6c50 0x40085480:0x3ffb6c70 0x40085432:0xa5a5a5a5  |<-CORRUPTED
0x40086492: taskSelectHighestPriorityTaskSMP at /home/kpaget/Documents/Code/TypeCFuzzer/Tests/esp-idf-v4.4.2/components/freertos/tasks.c:3409 (discriminator 1)

0x40086d5e: vTaskSwitchContext at /home/kpaget/Documents/Code/TypeCFuzzer/Tests/esp-idf-v4.4.2/components/freertos/tasks.c:3519 (discriminator 3)

0x40085480: _frxt_dispatch at /home/kpaget/Documents/Code/TypeCFuzzer/Tests/esp-idf-v4.4.2/components/freertos/port/xtensa/portasm.S:436

0x40085432: _frxt_int_exit at /home/kpaget/Documents/Code/TypeCFuzzer/Tests/esp-idf-v4.4.2/components/freertos/port/xtensa/portasm.S:231

```
Note that I've seen this manifest in many different ways depending on the exact code used to trigger the condition.

### Steps to reproduce.

The following code will demonstrate the issue:
```
#include <stdio.h>
#include "driver/timer.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/queue.h"
#include <esp_log.h>

// Handle for the tick-counting task
[... 153 lines cut ...]
A14     : 0x0000000f  A15     : 0x3ffb0b50  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000063  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff  

Backtrace:0x40086492:0x3ffb6c300x40086d5e:0x3ffb6c50 0x40085480:0x3ffb6c70 0x40085432:0xa5a5a5a5  |<-CORRUPTED
0x40086492: taskSelectHighestPriorityTaskSMP at /home/kpaget/Documents/Code/TypeCFuzzer/Tests/esp-idf-v4.4.2/components/freertos/tasks.c:3409 (discriminator 1)

0x40086d5e: vTaskSwitchContext at /home/kpaget/Documents/Code/TypeCFuzzer/Tests/esp-idf-v4.4.2/components/freertos/tasks.c:3519 (discriminator 3)

0x40085480: _frxt_dispatch at /home/kpaget/Documents/Code/TypeCFuzzer/Tests/esp-idf-v4.4.2/components/freertos/port/xtensa/portasm.S:436

0x40085432: _frxt_int_exit at /home/kpaget/Documents/Code/TypeCFuzzer/Tests/esp-idf-v4.4.2/components/freertos/port/xtensa/portasm.S:231

The first few printf() lines show that the interrupt and tick counter are working as intended, but the moment that the high-priority task (CountTicks) is blocked by its call to printf() the scheduler crashes and the system resets.
```

### More Information.

This is a minimal reproduction of an issue that I first noticed in a much more complex codebase, using an NMI-level interrupt to decode an edge-timed protocol.

- When the NMI is triggered by an edge it enables a timer interrupt with a short delay
- Each time the NMI fires it resets the countdown on the timer
- When the timer interrupt fires it assumes that the end of the packet has been reached; it then disables the NMI and notifies a higher-level processing task to parse the received data.
- If the higher-level task is blocked (such as by a call to printf() or ESP_LOGx()) then the scheduler locks up; the NMI continues to fire but eventually the TWDT resets the system with a failure address on the "rfi 7" instruction in the NMI handler since it never manages to schedule any other task (including the idle task) after the NMI.

I have a few different reproductions of this failure which manifest as different error reports, but all triggered by the same condition (returning from an interrupt to a high-priority task that is blocked). I can share some but not all of these codebases, so when you have a potential fix please let me know and I will test against all the manifestations of this issue to make sure it applies to all of them.

nopnop2002, 2022-09-28:
>xTaskNotifyFromISR(TickCounter, 0, eSetValueWithOverwrite, &TaskAwoken);

Bad parameter.

```
 BaseType_t xTaskNotifyIndexedFromISR( TaskHandle_t xTaskToNotify, 
                                       UBaseType_t uxIndexToNotify, 
                                       uint32_t ulValue, 
                                       eNotifyAction eAction, 
                                       BaseType_t *pxHigherPriorityTaskWoken );
```

nopnop2002, 2022-09-28:
ESP-IDF v4.4.1-405-g6c5fb29c2c worked with just this change.

```
    //int Ret = xTaskCreate(CountTicks, "CountTicks", 1024, 0, 15, &TickCounter);
    int Ret = xTaskCreate(CountTicks, "CountTicks", 1024*2, 0, 15, &TickCounter);
```

nopnop2002, 2022-09-28:
@Dazza0 

ESP-IDF v4.4.1-405-g6c5fb29c2c worked with just this change.

```
    //int Ret = xTaskCreate(CountTicks, "CountTicks", 1024, 0, 15, &TickCounter);
    int Ret = xTaskCreate(CountTicks, "CountTicks", 1024*2, 0, 15, &TickCounter);
```

Dazza0, 2022-09-28:
@kristinpaget I've been able to recreate the issue locally. This looks like a stack overflow issue. Could you please try increasing the stack size from `1024` to something large (like `4096`)?