espressif/esp-idf #9758: After TWAI start transmitting data all other FreeRTOS task slowdown, Ethernet W5500 SPI stop responding

MacWyznawca, 2022-09-09:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0-beta1-dirty

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### Development Kit.

ESP32S3 Custom boards

### Power Supply used.

External 3.3V

### What is the expected behavior?

Run task (SPI Ethernet, RMT WS2812 LED, netif, IP connection, ESP-NOW with normal speed, during TWAI (ESP32S3) is conected with last one TWAI node (ESP32).
TWAI speed 100 or 125 Kbps, 1 meter of CAN bus correctly terminated.

### What is the actual behavior?

All task (especially SPI W5500 Ethernet, netif, ESP-NOW) slow down very much after last one TWAI node initialized TWAI interface (even without transmitting data).

TWAI speed 100 or 125 Kbps, 1 meter of CAN bus correctly terminated.

### Steps to reproduce.

Compile some project with task ESP-NOW, IP connection and other (ESP32S3 board with TWAI interface).
Connect last one active TWAI node (ESP32 or other).

MacWyznawca, 2022-09-09:
Same code (without any changes accept change target) have same problems with ESP32 (SPI Ethernet W5500)

MacWyznawca, 2022-09-09:
Which last IDF (v5.1-dev-644-g867745a05c-dirty) same problem

abhi-ratnman, 2022-09-15:
@team any update on the issue?

Dazza0, 2022-09-28:
@MacWyznawca Could you provide more details such as:

- Which core the TWAI task is running on
- Which core the TWAI interrupt is installed on (i.e., which core `twai_driver_install()` was called)
- What the are core affinity and priorities of all the other tasks in your application?
- What is the status of the TWAI driver? (Any alerts, error counters etc).

closed 2022-11-12 by MacWyznawca