espressif/esp-idf #9760: How do I convert a PlatformIO build environment to ESP-IDF?

RoscoeTheDog, 2022-09-09:
I am using a ESP32-S2 variant. It seems like the support for this board is always behind the latest framework updates which is frustrating. Right now it seems as though the basic example for the smartconfig feature no longer sends acknowledgements to the smartphone app. I am trying to rebuild the project purely in ESP-IDF using VScode and the esp-idf extension to see if I can get it working, but it uses a different build system than PlatformIO does and I can't build it.

I understand that CMake wants us to specify what is required dependencies for each component. This seems like a huge huge amount of trouble for me in my instance where I have like 10 headers and source files for each different peripheral on the board I am making use of previously in my platformIO project.

The simplist and sloppiest way I could think of getting everything migrated over was to throw the headers in the main ESP-IDF project folder and tell CMake to look at the same root path for headers like this:

```
idf_component_register(SRCS "main.c"
                    INCLUDE_DIRS ".")
```
main.c
```

/*
	Include C++ headers here.

	If you use C++ libraries, they should be included in this CPP scope OR explicitely wrapped as extern "C++" {} so the compiler knows what conventions to use
	and how to link them together.
*/

/*
	Tell the compiler to wrap main for C convention since most of the IDF framework in standard C but we may want to use C++ occasionally.
*/

#ifdef __cplusplus
extern "C" {
#endif
	// Include C-headers here.
	#include <stdio.h>
	#include <stdlib.h>
	#include <string.h>
	#include <esp_timer.h>
	#include <freeRTOS/FreeRTOS.h>

	#include <configGpio.h>
	#include <configLedc.h>
	#include <configTimers.h>
	#include <configSteppers.h>
	#include <configWifi.h>
	#include <httpRequests.h>
	#include <esp_err.h>
	#include <string.h>
	#include <esp_pm.h>
	#include <esp_sleep.h>
	#include <esp_freertos_hooks.h>
	#include <freertos/task.h>
	#include <esp_sntp.h>
	#include <sys/time.h>
	#include <time.h>
	#include <globals.h>

	#include <cJSON.h>
	#include <cJSON_Utils.h>

    void app_main(void);

#ifdef __cplusplus
}
#endif

void app_main(void) {
	// configure the JSON library to use FREERTOS thread safe malloc / free methods
    cJSON_Hooks hooks;
    hooks.malloc_fn = pvPortMalloc;
    hooks.free_fn = vPortFree;
    cJSON_InitHooks(&hooks);

	// esp_pm_config_esp32s2_t cfg;
	// esp_wifi_set_ps(WIFI_PS_MIN_MODEM);
	// cfg.light_sleep_enable = true;
[... 76 lines cut ...]
c:/users/admin/.espressif/tools/xtensa-esp32s2-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld.exe: esp-idf/main/libmain.a(main.c.obj):(.literal.app_main+0x24): undefined reference to `initialize_ledc_config_0'
c:/users/admin/.espressif/tools/xtensa-esp32s2-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld.exe: esp-idf/main/libmain.a(main.c.obj):(.literal.app_main+0x28): undefined reference to `vInitCurtainMotorConfig_0'
c:/users/admin/.espressif/tools/xtensa-esp32s2-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld.exe: esp-idf/main/libmain.a(main.c.obj):(.literal.app_main+0x2c): undefined reference to `initializeTimerConfig'
c:/users/admin/.espressif/tools/xtensa-esp32s2-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld.exe: esp-idf/main/libmain.a(main.c.obj):(.literal.app_main+0x30): undefined reference to `nvsRestoreSystemState'
c:/users/admin/.espressif/tools/xtensa-esp32s2-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld.exe: esp-idf/main/libmain.a(main.c.obj):(.literal.app_main+0x34): undefined reference to `setStatusLEDGreen'
c:/users/admin/.espressif/tools/xtensa-esp32s2-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld.exe: esp-idf/main/libmain.a(main.c.obj):(.literal.app_main+0x38): undefined reference to `setStatusLEDYellow'
c:/users/admin/.espressif/tools/xtensa-esp32s2-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld.exe: esp-idf/main/libmain.a(main.c.obj):(.literal.app_main+0x3c): undefined reference to `initializeTasks'
c:/users/admin/.espressif/tools/xtensa-esp32s2-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld.exe: esp-idf/main/libmain.a(main.c.obj):(.literal.app_main+0x40): undefined reference to `initializeWifi'
c:/users/admin/.espressif/tools/xtensa-esp32s2-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld.exe: esp-idf/main/libmain.a(main.c.obj): in function `app_main':
C:\Users\admin\Documents\Github\ESP32-S2-Saola-Project\ESP32_S2_Saola1_ESPIDF\build/../main/main.c:72: undefined reference to `vInitGpioConfig'
c:/users/admin/.espressif/tools/xtensa-esp32s2-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld.exe: C:\Users\admin\Documents\Github\ESP32-S2-Saola-Project\ESP32_S2_Saola1_ESPIDF\build/../main/main.c:82: undefined reference to `setStatusLEDRed'
c:/users/admin/.espressif/tools/xtensa-esp32s2-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld.exe: C:\Users\admin\Documents\Github\ESP32-S2-Saola-Project\ESP32_S2_Saola1_ESPIDF\build/../main/main.c:84: undefined reference to `initialize_ledc_config_0'
c:/users/admin/.espressif/tools/xtensa-esp32s2-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld.exe: C:\Users\admin\Documents\Github\ESP32-S2-Saola-Project\ESP32_S2_Saola1_ESPIDF\build/../main/main.c:86: undefined reference to `vInitCurtainMotorConfig_0'
c:/users/admin/.espressif/tools/xtensa-esp32s2-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld.exe: C:\Users\admin\Documents\Github\ESP32-S2-Saola-Project\ESP32_S2_Saola1_ESPIDF\build/../main/main.c:87: undefined reference to `initializeTimerConfig'
c:/users/admin/.espressif/tools/xtensa-esp32s2-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld.exe: C:\Users\admin\Documents\Github\ESP32-S2-Saola-Project\ESP32_S2_Saola1_ESPIDF\build/../main/main.c:92: undefined reference to `nvsRestoreSystemState'
c:/users/admin/.espressif/tools/xtensa-esp32s2-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld.exe: C:\Users\admin\Documents\Github\ESP32-S2-Saola-Project\ESP32_S2_Saola1_ESPIDF\build/../main/main.c:93: undefined reference to `setStatusLEDGreen'
c:/users/admin/.espressif/tools/xtensa-esp32s2-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld.exe: C:\Users\admin\Documents\Github\ESP32-S2-Saola-Project\ESP32_S2_Saola1_ESPIDF\build/../main/main.c:97: undefined reference to `setStatusLEDRed'
c:/users/admin/.espressif/tools/xtensa-esp32s2-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld.exe: C:\Users\admin\Documents\Github\ESP32-S2-Saola-Project\ESP32_S2_Saola1_ESPIDF\build/../main/main.c:104: undefined reference to `setStatusLEDYellow'
c:/users/admin/.espressif/tools/xtensa-esp32s2-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld.exe: C:\Users\admin\Documents\Github\ESP32-S2-Saola-Project\ESP32_S2_Saola1_ESPIDF\build/../main/main.c:104: undefined reference to `initializeTasks'
c:/users/admin/.espressif/tools/xtensa-esp32s2-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld.exe: C:\Users\admin\Documents\Github\ESP32-S2-Saola-Project\ESP32_S2_Saola1_ESPIDF\build/../main/main.c:117: undefined reference to `initializeWifi'
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.

 *  The terminal process "C:\Windows\System32\cmd.exe /d /c ninja " terminated with exit code: 1.
 ```

igrr, 2022-09-10:
@RoscoeTheDog Could you please upload your project somewhere? As is, there is not enough information provided in this ticket for us to give you some help. It's not clear where the functions like `initializeWifi` are defined, whether they have C or C++ linkage, in which component they are located, and so on.

In general, you can also take a look at some of the example projects provided with ESP-IDF. They should demonstrate how to structure the project, splitting it into multiple components. https://github.com/espressif/esp-idf/tree/master/examples/system/console/advanced is one such example.

brainstorm, 2022-10-12:
This issue is timely and convenient since it scratches a similar itch I have with my project (having it working with both platformio and plain esp-idf at the same time):

https://github.com/brainstorm/esp-modbus-power-meter/

The platformio side works fine so far, but I'm trying to migrate to ESP-IDF 5.x and have it compile with `idf.py` (esp-idf plain tooling), but there are some dissimilarities including but not limited to header management:

https://github.com/brainstorm/esp-modbus-power-meter/commit/fb52ef9da4897232c5d0db4db2543c4aa3c72385

I'm trying to quickly fix this project up now that I have a small window of testing opportunity (this week). I'm also reporting to the following issues as I go, for context:

https://github.com/espressif/esp-insights/issues/12
https://github.com/espressif/esp-modbus/issues/3

I hope some espressif expert can take a peek at my repo and point out how to get my project compilable with both esp-idf and platformio easily (build system/CMake tweaks) so that I can followup with other issues on the application side?

closed 2023-04-03 by Alvin1Zhang