espressif/esp-idf #9761: change the example "examples\bluetooth\bluedroid\ble\gatt_server" by custom server UUID，some error happen

giu-wxyao, 2022-09-10:
target ：ESP32,
system：windows
version ：ESP-IDF-V4.4

in the example i use one gatts_profile_inst, and the server UUID is custom uuid as follow :
static uint8_t GATTS_SERVICE_UUID_TEST_A[16] = {0x19, 0x00, 0x08, 0x1, 0x11, 0x91, 0x00, 0x10, 0x90, 0x74, 0x27, 0x06, 0x06, 0x01, 0x11, 0x91};

the 1st question is how to understand the last parameter  as follow "esp_ble_gatts_create_service(gatts_if, &gl_profile_tab[PROFILE_A_APP_ID].service_id, GATTS_NUM_HANDLE_TEST_A)

the 2nd : donw load to esp32,  havent connected. but have the read event. i dont know why. follow is the information printed. to see the last 3 lines;

I (470) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (518) BTDM_INIT: BT controller compile version [6a07b06]
I (518) system_api: Base MAC address is not set
I (518) system_api: read default base MAC address from EFUSE
I (528) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07
I (928) GATTS_DEMO: REGISTER_APP_EVT, status 0, app_id 0

W (928) BT_BTM: BTM_BleWriteAdvData, Partial data write into ADV
I (928) GATTS_DEMO: notify TASK
W (928) BT_BTM: BTM_BleWriteScanRsp, Partial data write into ADV
I (938) GATTS_DEMO: CREATE_SERVICE_EVT, status 0,  service_handle 40

I (958) GATTS_DEMO: SERVICE_START_EVT, status 0, service_handle 40

I (958) GATTS_DEMO: GATT_READ_EVT, conn_id 0, trans_id 2621482, handle 37137

E (968) BT_GATT: GATTS_SendRsp Unknown  conn_id: 3

E (968) BT_APPL: Sending response failed

esp-cjh, 2022-09-14:
1 The last param of `esp_ble_gatts_create_service()` indicate the number of attributes of the service.
2 Through your log, I found that you didn't add the characteristic. Can you provide your source code?
3 I suggest you refer to gatt_security_server example to create service table.

giu-wxyao, 2022-09-23:
yes ，you are right, I didnt add the characteristic ,after i add the char, no error message.

closed 2022-09-23 by giu-wxyao