espressif/esp-idf #9762: When I use console example, use USB_SERIAL_JTAG, when I close putty, the esp32c3 will reboot.

WangShuoran, 2022-09-10:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v.4.4.2

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

ESP32C3

### Power Supply used.

USB

### What is the expected behavior?

The esp32c3 do not reboot.

### What is the actual behavior?

When I use example system/console/basic, I use esp32c3 with USB JTAG/serial debug unit to input content and show output.

Will I open Windows putty and open uart port and close. The esp32c3 will reshow any content that show in open.

I don't know this behavior why.

### Steps to reproduce.

code:
[basic_c3.zip](https://github.com/espressif/esp-idf/files/9539575/basic_c3.zip)

igrr, 2022-09-10:
I think the reason is that Putty changes the state of DTR/RTS control lines of the virtual serial port when connecting and disconnecting. These control lines are hardwired to reset the ESP32-C3 or put it into the download mode. You can find more details in the ESP32-C3 TRM, section 28.3.1 "CDC­ACM USB Interface Functional Description".

WangShuoran, 2022-09-12:
@igrr , Thank you give me any answer.
But I do not think is this question. But I will find DTR/RTS config in Putty.

I think DTR/RTS config influence ESP32C3 reboot will UART is CH340/PL2303 etc USB to UART mcu will connect DTR/RTS to EN/IO9(ESP32C3). When the PC change DTR/RTS, CH340/PL2303 will set DTR/RTS.

But in this, I use ESP32C3 USB JTAG/serial debug unit, not CH340.

igrr, 2022-09-12:
The USB_SERIAL_JTAG peripheral implements DTR/RTS logic inside the chip. When the host changes the logic values of DTR and RTS signals via USB, this is translated into "reset" or "reset into download mode" actions inside the chip. Direct connection to EN or IO9 is not necessary in this case. I think this is explained in the TRM chapter which i mentioned above.

WangShuoran, 2022-09-13:
You are right.

closed 2022-09-13 by WangShuoran