espressif/esp-idf #9764: setenv memory leak

JimmyO-GH, 2022-09-10:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0

### Operating System used.

Linux

### How did you build your project?

Command line with CMake

### Development Kit.

ESP32-DevKit

### Power Supply used.

External 5V

### What is the expected behavior?

constant free memory

### What is the actual behavior?

free memory is becoming less and less

### Steps to reproduce.

Little example with follow code

	for(i = 0; i < 100; i++) {
		setenv("ABC", "some text", 1);
		unsetenv("ABC");
		printf("Totals: free %lu \n", esp_get_free_heap_size());
	}

### Debug Logs.

```plain
Totals: free 188532 
Totals: free 188512 
Totals: free 188492 
Totals: free 188472 
Totals: free 188452 
Totals: free 188432 
Totals: free 188412 
Totals: free 188392 
Totals: free 188372 
Totals: free 188352 
Totals: free 188332 
Totals: free 188312 
Totals: free 188292 
Totals: free 188272 
Totals: free 188252 
Totals: free 188232 
Totals: free 188212 
Totals: free 188192 
Totals: free 188172
```

igrr, 2022-09-11:
Hi @JimmyO-GH,
This question has been answered earlier in https://github.com/espressif/esp-idf/issues/3046#issuecomment-499168477.
Please check the reply there and this explanation on the newlib mailing list: https://sourceware.org/legacy-ml/newlib/2018/msg00305.html.

JimmyO-GH, 2022-09-11:
thanks for this information. i will use putenv instead of setenv.

closed 2022-09-11 by JimmyO-GH