espressif/esp-idf #9765: ESP32 C3 S2 softap fails to start with no error condition

tablatronix, 2022-09-10:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Reposting from arduino
No adjustments in arduino can I get to fix this condition, must be issue in IDF

ESP32 C3 S2 softap fails to start with no error condition
https://github.com/espressif/arduino-esp32/issues/7232

I will try to reproduce with IDF

Arduino
```c++
// wifi_basic.ino

#include <arduino.h>
#include <WiFi.h>
#include <esp_wifi.h>  

void setup(){
	Serial.begin(115200);
	delay(2000);
	Serial.println("Startup....");
	Serial.setDebugOutput(true);

	WiFi.begin("hellowifi","noonehere");
	
    while (WiFi.status() != WL_CONNECTED && millis()<10000) {
        delay(500);
        Serial.print(".");
    }

	if(WiFi.status() == WL_CONNECTED){
		Serial.println("");
		Serial.println("WiFi connected.");
		Serial.println("IP address: ");
		Serial.println(WiFi.localIP());
	}
	else {
		Serial.println("WiFi NOT CONNECTED, starting ap");
		WiFi.enableSTA(false); // BREAKING <-------------------
		delay(2000);		
	    WiFi.softAP("espsoftap","12345678");
	}
}

void loop(){

}
```

```php
Startup....
[  2160][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 0 - WIFI_READY
[  2189][V][WiFiGeneric.cpp:338] _arduino_event_cb(): STA Started
[  2190][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 2 - STA_START
[  2192][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
[  2194][E][WiFiSTA.cpp:317] begin(): connect failed! 0x300a
................WiFi NOT CONNECTED, starting ap
[ 10195][V][WiFiGeneric.cpp:341] _arduino_event_cb(): STA Stopped
[ 10196][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 3 - STA_STOP
[ 12205][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 0 - WIFI_READY
[ 12215][V][WiFiGeneric.cpp:390] _arduino_event_cb(): AP Started
[ 12216][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 10 - AP_START
```
NO AP IS RUNNING, no tools can see SSID

chegewara, 2022-09-10:
WiFi.softAP(ssid, password);

tablatronix, 2022-09-16:
This seems to already be fixed in the IDF according to testing by others and slated for arduino 2.0.5 already
I will close this as this is most likely already fixed in IDF at this point

closed 2022-09-16 by tablatronix