espressif/esp-idf #9766: ESP32 Modem Sleep high latency and loss rates

readmodifywrite, 2022-09-10:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

ESP-IDF 4.4.1
ESP32 (not the S or C)

I'm seeing really erratic and high ping times along with high packet loss on some ESP32 devices when running in modem sleep.  Ping times look like this:

64 bytes from 10.0.0.99: icmp_seq=323 ttl=255 time=1079 ms
64 bytes from 10.0.0.99: icmp_seq=324 ttl=255 time=65.4 ms
64 bytes from 10.0.0.99: icmp_seq=325 ttl=255 time=64.5 ms
64 bytes from 10.0.0.99: icmp_seq=326 ttl=255 time=1538 ms
64 bytes from 10.0.0.99: icmp_seq=327 ttl=255 time=535 ms
64 bytes from 10.0.0.99: icmp_seq=328 ttl=255 time=285 ms
64 bytes from 10.0.0.99: icmp_seq=329 ttl=255 time=3.75 ms
64 bytes from 10.0.0.99: icmp_seq=330 ttl=255 time=709 ms
64 bytes from 10.0.0.99: icmp_seq=331 ttl=255 time=223 ms
64 bytes from 10.0.0.99: icmp_seq=332 ttl=255 time=2.47 ms
64 bytes from 10.0.0.99: icmp_seq=333 ttl=255 time=64.4 ms
64 bytes from 10.0.0.99: icmp_seq=334 ttl=255 time=801 ms
64 bytes from 10.0.0.99: icmp_seq=335 ttl=255 time=219 ms
64 bytes from 10.0.0.99: icmp_seq=336 ttl=255 time=132 ms
64 bytes from 10.0.0.99: icmp_seq=337 ttl=255 time=2.40 ms
64 bytes from 10.0.0.99: icmp_seq=338 ttl=255 time=689 ms
64 bytes from 10.0.0.99: icmp_seq=339 ttl=255 time=102 ms
64 bytes from 10.0.0.99: icmp_seq=340 ttl=255 time=21.2 ms
64 bytes from 10.0.0.99: icmp_seq=341 ttl=255 time=657 ms
64 bytes from 10.0.0.99: icmp_seq=342 ttl=255 time=2278 ms
64 bytes from 10.0.0.99: icmp_seq=343 ttl=255 time=1259 ms
64 bytes from 10.0.0.99: icmp_seq=344 ttl=255 time=236 ms
64 bytes from 10.0.0.99: icmp_seq=345 ttl=255 time=96.5 ms
64 bytes from 10.0.0.99: icmp_seq=346 ttl=255 time=116 ms
64 bytes from 10.0.0.99: icmp_seq=347 ttl=255 time=1226 ms
64 bytes from 10.0.0.99: icmp_seq=348 ttl=255 time=198 ms

I have seen this behavior on 2 APs from different vendors.  DTIM is set to 1.

If I disable modem sleep, the devices will consistently return ping times less than 10 milliseconds, so I'm reasonably convinced this is not a signal strength or power supply issue.

I have also pinged phones and laptops on the same network, none exhibit ping times this erratic and high.

Not all devices in my fleet exhibit modem sleep performance this bad, many of them are completely acceptable.

I'm using esp_wifi_set_ps(WIFI_PS_MIN_MODEM).

My application code runs on a separate core from the wifi/RTOS, and I've tested it in a bare bones configuration and still see the same behavior.

I'm not sure what else to try at this point, other than just giving up on modem sleep entirely for the ESP32.  I cannot predict which devices will be affected, and when they are, they often become almost unusable.

Espressif-liuuuu, 2022-09-14:
Hi @readmodifywrite , thanks for your report. Could you please give me a piece of WiFi capture? BTW did you use light-sleep mode in that case?

readmodifywrite, 2022-09-15:
Hi @Espressif-liuuuu ,

Automatic light sleep is enabled.

What kind of WiFi capture do you need?  Wireshark of the ICMP traffic?  Or a low level sniffer capture (that will take some time to set up...).

readmodifywrite, 2022-09-15:
Note that I also see the opposite problem sometimes, the ping time is way to low for a device that should only be waking on the DTIM interval:

PING 10.0.0.143 (10.0.0.143) 56(84) bytes of data.
64 bytes from 10.0.0.143: icmp_seq=1 ttl=255 time=8.75 ms
64 bytes from 10.0.0.143: icmp_seq=2 ttl=255 time=2.34 ms
64 bytes from 10.0.0.143: icmp_seq=3 ttl=255 time=3.23 ms
64 bytes from 10.0.0.143: icmp_seq=4 ttl=255 time=4.28 ms
64 bytes from 10.0.0.143: icmp_seq=5 ttl=255 time=3.17 ms
64 bytes from 10.0.0.143: icmp_seq=6 ttl=255 time=3.41 ms
64 bytes from 10.0.0.143: icmp_seq=7 ttl=255 time=5.18 ms
64 bytes from 10.0.0.143: icmp_seq=8 ttl=255 time=35.3 ms
64 bytes from 10.0.0.143: icmp_seq=9 ttl=255 time=6.00 ms
64 bytes from 10.0.0.143: icmp_seq=10 ttl=255 time=2.18 ms
64 bytes from 10.0.0.143: icmp_seq=11 ttl=255 time=4.38 ms
64 bytes from 10.0.0.143: icmp_seq=12 ttl=255 time=14.5 ms
64 bytes from 10.0.0.143: icmp_seq=13 ttl=255 time=6.69 ms
64 bytes from 10.0.0.143: icmp_seq=14 ttl=255 time=4.84 ms
64 bytes from 10.0.0.143: icmp_seq=15 ttl=255 time=11.5 ms
64 bytes from 10.0.0.143: icmp_seq=16 ttl=255 time=7.58 ms
64 bytes from 10.0.0.143: icmp_seq=17 ttl=255 time=10.6 ms
64 bytes from 10.0.0.143: icmp_seq=18 ttl=255 time=3.25 ms
64 bytes from 10.0.0.143: icmp_seq=19 ttl=255 time=2.19 ms
64 bytes from 10.0.0.143: icmp_seq=20 ttl=255 time=10.8 ms
64 bytes from 10.0.0.143: icmp_seq=21 ttl=255 time=2.40 ms
64 bytes from 10.0.0.143: icmp_seq=22 ttl=255 time=2.15 ms
64 bytes from 10.0.0.143: icmp_seq=23 ttl=255 time=15.2 ms
64 bytes from 10.0.0.143: icmp_seq=24 ttl=255 time=3.48 ms
64 bytes from 10.0.0.143: icmp_seq=25 ttl=255 time=2.33 ms
64 bytes from 10.0.0.143: icmp_seq=26 ttl=255 time=2.96 ms
64 bytes from 10.0.0.143: icmp_seq=27 ttl=255 time=9.36 ms
64 bytes from 10.0.0.143: icmp_seq=28 ttl=255 time=2.19 ms
64 bytes from 10.0.0.143: icmp_seq=29 ttl=255 time=2.93 ms
64 bytes from 10.0.0.143: icmp_seq=30 ttl=255 time=2.24 ms
64 bytes from 10.0.0.143: icmp_seq=31 ttl=255 time=1.81 ms
64 bytes from 10.0.0.143: icmp_seq=32 ttl=255 time=1.96 ms
64 bytes from 10.0.0.143: icmp_seq=33 ttl=255 time=2.20 ms
64 bytes from 10.0.0.143: icmp_seq=34 ttl=255 time=3.50 ms
64 bytes from 10.0.0.143: icmp_seq=35 ttl=255 time=1.69 ms
64 bytes from 10.0.0.143: icmp_seq=36 ttl=255 time=8.67 ms
64 bytes from 10.0.0.143: icmp_seq=37 ttl=255 time=2.10 ms
64 bytes from 10.0.0.143: icmp_seq=38 ttl=255 time=2.11 ms
64 bytes from 10.0.0.143: icmp_seq=39 ttl=255 time=16.6 ms
64 bytes from 10.0.0.143: icmp_seq=40 ttl=255 time=16.3 ms
64 bytes from 10.0.0.143: icmp_seq=41 ttl=255 time=4.47 ms
64 bytes from 10.0.0.143: icmp_seq=42 ttl=255 time=2.46 ms
64 bytes from 10.0.0.143: icmp_seq=43 ttl=255 time=1.77 ms
64 bytes from 10.0.0.143: icmp_seq=44 ttl=255 time=5.02 ms
64 bytes from 10.0.0.143: icmp_seq=45 ttl=255 time=5.41 ms
64 bytes from 10.0.0.143: icmp_seq=46 ttl=255 time=2.40 ms

readmodifywrite, 2022-09-15:
Here are some more pings, showing the device transition from low pings to highly variable ones (more realistic of modem sleep, though with some that are far too high to be plausible):

64 bytes from 10.0.0.143: icmp_seq=1825 ttl=255 time=2.34 ms
64 bytes from 10.0.0.143: icmp_seq=1826 ttl=255 time=2.06 ms
64 bytes from 10.0.0.143: icmp_seq=1827 ttl=255 time=2.26 ms
64 bytes from 10.0.0.143: icmp_seq=1828 ttl=255 time=2.30 ms
64 bytes from 10.0.0.143: icmp_seq=1829 ttl=255 time=4.26 ms
64 bytes from 10.0.0.143: icmp_seq=1830 ttl=255 time=11.7 ms
64 bytes from 10.0.0.143: icmp_seq=1831 ttl=255 time=2.39 ms
64 bytes from 10.0.0.143: icmp_seq=1832 ttl=255 time=6.85 ms
64 bytes from 10.0.0.143: icmp_seq=1833 ttl=255 time=2.84 ms
64 bytes from 10.0.0.143: icmp_seq=1834 ttl=255 time=6.87 ms
64 bytes from 10.0.0.143: icmp_seq=1835 ttl=255 time=6.03 ms
64 bytes from 10.0.0.143: icmp_seq=1836 ttl=255 time=5.48 ms
64 bytes from 10.0.0.143: icmp_seq=1837 ttl=255 time=2.16 ms
64 bytes from 10.0.0.143: icmp_seq=1838 ttl=255 time=6.88 ms
64 bytes from 10.0.0.143: icmp_seq=1839 ttl=255 time=6.25 ms
64 bytes from 10.0.0.143: icmp_seq=1840 ttl=255 time=2.32 ms
64 bytes from 10.0.0.143: icmp_seq=1841 ttl=255 time=5.01 ms
64 bytes from 10.0.0.143: icmp_seq=1842 ttl=255 time=5.98 ms
64 bytes from 10.0.0.143: icmp_seq=1843 ttl=255 time=2.32 ms
64 bytes from 10.0.0.143: icmp_seq=1844 ttl=255 time=2.61 ms
64 bytes from 10.0.0.143: icmp_seq=1845 ttl=255 time=2.78 ms
64 bytes from 10.0.0.143: icmp_seq=1846 ttl=255 time=2.58 ms
64 bytes from 10.0.0.143: icmp_seq=1847 ttl=255 time=3.32 ms
64 bytes from 10.0.0.143: icmp_seq=1848 ttl=255 time=55.1 ms
64 bytes from 10.0.0.143: icmp_seq=1849 ttl=255 time=2.22 ms
64 bytes from 10.0.0.143: icmp_seq=1850 ttl=255 time=3.20 ms
64 bytes from 10.0.0.143: icmp_seq=1851 ttl=255 time=845 ms
64 bytes from 10.0.0.143: icmp_seq=1852 ttl=255 time=45.1 ms
64 bytes from 10.0.0.143: icmp_seq=1853 ttl=255 time=67.7 ms
64 bytes from 10.0.0.143: icmp_seq=1854 ttl=255 time=143 ms
64 bytes from 10.0.0.143: icmp_seq=1855 ttl=255 time=113 ms
64 bytes from 10.0.0.143: icmp_seq=1856 ttl=255 time=32.5 ms
64 bytes from 10.0.0.143: icmp_seq=1857 ttl=255 time=10.8 ms
64 bytes from 10.0.0.143: icmp_seq=1858 ttl=255 time=78.3 ms
64 bytes from 10.0.0.143: icmp_seq=1859 ttl=255 time=2.32 ms
64 bytes from 10.0.0.143: icmp_seq=1860 ttl=255 time=21.2 ms
64 bytes from 10.0.0.143: icmp_seq=1861 ttl=255 time=44.7 ms
64 bytes from 10.0.0.143: icmp_seq=1862 ttl=255 time=68.5 ms
64 bytes from 10.0.0.143: icmp_seq=1863 ttl=255 time=88.3 ms
64 bytes from 10.0.0.143: icmp_seq=1864 ttl=255 time=337 ms
64 bytes from 10.0.0.143: icmp_seq=1865 ttl=255 time=58.6 ms
64 bytes from 10.0.0.143: icmp_seq=1866 ttl=255 time=158 ms
64 bytes from 10.0.0.143: icmp_seq=1867 ttl=255 time=180 ms
64 bytes from 10.0.0.143: icmp_seq=1868 ttl=255 time=208 ms
64 bytes from 10.0.0.143: icmp_seq=1869 ttl=255 time=124 ms
64 bytes from 10.0.0.143: icmp_seq=1870 ttl=255 time=44.4 ms
64 bytes from 10.0.0.143: icmp_seq=1871 ttl=255 time=8.06 ms
64 bytes from 10.0.0.143: icmp_seq=1872 ttl=255 time=91.0 ms
64 bytes from 10.0.0.143: icmp_seq=1873 ttl=255 time=115 ms
64 bytes from 10.0.0.143: icmp_seq=1874 ttl=255 time=51.1 ms
64 bytes from 10.0.0.143: icmp_seq=1875 ttl=255 time=6.52 ms
64 bytes from 10.0.0.143: icmp_seq=1876 ttl=255 time=182 ms

This device was sitting otherwise idle.

Espressif-liuuuu, 2022-09-16:
Thanks @readmodifywrite, low level sniffer capture would be appreciated, you could refer to https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-guides/wireshark-user-guide.html?highlight=wireshark. It would be better to set OPEN for your router.  
On the other hand, you could disable lightsleep and have a retry to see if the high latency happen again.

readmodifywrite, 2022-09-16:
Ok, I'll see what I can do.  Might be next week before I can get a full capture set up.

I'll try turning off light sleep as well.

Additionally, I noticed this line in the v4.4.2 changelog (I am running 4.4.1): "Fixed data loose when WiFi power-save is enabled with Wi-Fi API being called" - any idea what this is referring to?

In any case, I should probably update to 4.4.2 anyway so I will try that as well.

Espressif-liuuuu, 2022-09-16:
That change is related to another issue, the chip may not able to recieve data in a long time after a API being called. That may be not the same with this issue, yes you could try that as well.

Espressif-liuuuu, 2022-09-19:
Oh, thats did a weird one. Could you check the value of the abnormal current? The radio current of ESP32 is 100mA at least, lets make sure thats did radio current at first.  
I suspect that may caused by something about scheduling or dual-core DFS. Could you try your code on uni-core and test it again? We would look into the dual-core situation also.

readmodifywrite, 2022-09-19:
The scope traces are 50 mV/div and the current amplifier has a gain of 100V/V on a 10 mohm shunt, which makes it 1:1 volts to current.  So each 50 mV division on the vertical axis is 50 mA of current.

With wifi disabled, the board (WROVER-KIT) idles between 60 mA and 90 mA (there is some noise on the scope, and it is only 8 bits).  
Wifi enabled with no power save: 160 mA to 230 mA
Modem sleep (WIFI_PS_MIN_MODEM): around 60 mA on the low side and peaking at 210 mA,

Here's the trace in single core mode with a 20 ms task delay:
[image]

Which seems to confirm your suspicion this is an issue with the dual core scheduler.

My dual core app runs the Espressif system code (wifi, net, etc) on core 0 and my app runs pinned to core 1.  I also tried pinning my app to core 0 (leaving core 1 active, but unused) - the issue remains in that configuration as well.  Unpinning the app task also yields the same result.

readmodifywrite, 2022-10-02:
It looks like enabling CONFIG_FREERTOS_OPTIMIZED_SCHEDULER is also part of the puzzle.  If that is not enabled, then even in single core mode the modem sleep seems to be buggy.

readmodifywrite, 2022-10-02:
Finally, I am still seeing hardware specific issues with modem sleep, even with the workaround of running in single core mode with an optimized scheduler.  This seems to work running my application code on my JTAG dev kit, but deploying the same firmware to other units sometimes results in the same erratic ping behavior and high packet loss.  I think the ping bug may be a separate (but related?) issue that has hardware to hardware variances.

Note that on all affected hardware, disabling modem sleep always corrects the issue.  Ping times return to <10 ms, packet loss disappears, and the device performs quite well.  This is not a signal strength or WiFi network related issue.

readmodifywrite, 2022-10-02:
Final note for today:  I'm seeing this behavior across a fleet of 10 devices in my test lab.  It is consistent on all of them.  The dev kit seems fine, but any device deployed to custom board exhibits this problem.  The single core/optimized scheduler workaround seems to work on the dev kit but it doesn't work elsewhere (as far as the pings and packet loss goes).

So, to summarize, despite my best efforts, modem sleep remains hopelessly broken beyond usability.

Is there *anything* else I can check or try?  The always on power consumption of the ESP32 is absolutely terrible.  And while modem + auto light sleep isn't exactly anything to brag about (I assume you've seen the DA16200), it still makes a huge and worthwhile difference.  Even wall powered devices would benefit - wasted power is wasted regardless of the source.

readmodifywrite, 2022-10-02:
Final final note (for real this time!):

I rechecked pings on the dev kit, and it has the same issue.  Pings are all over the place.  700 to 1100 ms is common.  Turn modem sleep off and the vast majority are under 20 ms.

So, while there is clearly some kind of issue with the dual core scheduler, the ping times and poor network performance are something else and are not helped by switching the scheduler configuration.

I am guessing that if your application has 100 ms sleep intervals, then maybe modem sleep will work ok.  If you are waking up more frequently than that, modem sleep is not usable at this time.

Espressif-liuuuu, 2022-10-11:
Hi @readmodifywrite , sorry for late reply, thanks for your greater information.
I tried to reproduce the current issue today and unfortunately,  I failed. I pinned app task on core 1 while others` on core 0, at most of the time, the current would be shown as below
[image]
I enabled DFS in that example, so the current would be among 10~40mA if WiFi was IDLE. It turned to about 100mA every 100ms to receive beacon.

Rarely the high current would stayed during a beacon interval(100ms) since that beacon missed.
[image]

Im wondering if you already ran your application in your case, it seems that the current was much higher than mine. Also, that could be caused by different CPU frequency configuration. The RTOS frequency could be another factor.

Could you please provide me the reproducible project? And so does the sdkconfig. I`d like to reproduce that locally.

You could also take a try in example/wifi/power-save. I used default configuration besides `ESP32_WIFI_TASK_PINNED_TO_CORE_1=y` and `TICKLESS_IDLE=n`.

Espressif-liuuuu, 2022-10-11:
For the high latency issue, we are still eager to get the caputure. Also, please provide me with your sdkconfig and I will reproduce that locally:>

readmodifywrite, 2022-10-12:
I'll try and get you a reproducible project this week.  I have some travel coming up so it might be awhile otherwise.  Thanks!

readmodifywrite, 2022-10-12:
See attached power_save.c and sdkconfig.

This is set for dual core and 1000 Hz RTOS ticks with tickless idle enabled.

Most of the problems in this configuration seem to occur with the max frequency set to 240 MHz and min at 80 MHz.  Setting both to 80 MHz seems to do better regardless of light sleep or the task tick delay.

If the ESP32 misses a beacon, why would it leave the modem on for the entire 100 ms duration?  Wouldn't it make sense to power down and wait for the next beacon?

Note that the files have .txt appended, because apparently github cannot handle uploading a C source file to the issue tracker.  If anyone out there wants to make that make sense to me, feel free to chime in.

[sdkconfig.txt](https://github.com/espressif/esp-idf/files/9768681/sdkconfig.txt)
[power_save.c.txt](https://github.com/espressif/esp-idf/files/9768692/power_save.c.txt)

Espressif-liuuuu, 2022-10-20:
Yes thats right, this issue was all about the huge precise influence on system timer which came from DFS process between 80MHz and 240MHz. 

As you can see in TRM, the clock source of 240MHz-CPU is 480MHz-PLL_CLK while the one of 80MHz-CPU is 320MHz-PLL. 
[image]
The switch between these two clock sources would cost lots of time and also make greate impact on the precise of system timer. 
In fact, I triggered a ets_timer to wakeup each 100ms and toggle GPIO signal to see the real time. In my test, the real wakeup duration would be about 130ms around, if DFS from 160MHz to 240MHz. The TBTT timer is a system timer infact, such a huge delay is unaccept for it and the beacon miss is almost happen in each TBTT. Since the RF resources would be kept if beacon is lost, the current would stay high until next TBTT. Thats the reason why it shows about half time of high current.
 
The influence is always there, with more frequent the switch, more serious it could be. Thats also why it always shows at dual core and short vTaskDelay() situation.  

Now here we have 3 solutions 
>1. Change max frequency to 160MHz or change the min one to 10/20/40MHz. Its not recommended to use freq change between 80 to 240MHz on ESP32.
> 2. Use chip like ESP32C3/ESP32S3. The TBTT timer is not system timer on these chips anymore, so that the freq change process would not influence the precise.
> 3. Use v5,0 version. On that version, the RF resources kept time is configurable. Its upto you to decide how much time you want to receive a beacon at most. So the half high current wouldn`t happen even with the 80~240MHz switch. Notice that the system timer use would still be affected in this way.

readmodifywrite, 2022-10-24:
@Espressif-liuuuu 

Thanks for the update!

That makes sense.  Can we get a docs update to clarify the timing issues with switching the PLL frequency?

For now I will disable DFS - I don't really need it in my application.  I should also be able to better coordinate wakeup times for my application tasks to minimize the wake up rate.

I'll continue to monitor and let you know if I'm still seeing any issues.

Espressif-liuuuu, 2022-10-25:
@readmodifywrite  
Yes sure, we will see if any docs could be added in this part.
If there is still any issue in latency, please let me know and provide me with a piece of Wi-Fi capture

readmodifywrite, 2022-10-25:
@Espressif-liuuuu 

Thanks.  I suspect there is still an issue - I'm seeing some system timing drift (now that I know to look for that!) with modem sleep enabled and DFS off - but I need to eliminate more possible variables.  I'll get you some more info and some captures in the next week or so.

closed 2024-02-27 by Alvin1Zhang