espressif/esp-idf #9768: ###A issue about the sample frequency of esp32-s2

kurau1, 2022-09-11:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Module or chip used: ESP32-S2
Operating System: Windows
Software:Visual studio code

```
    adc_digi_configuration_t dig_cfg = {
        .conv_limit_en = ADC_CONV_LIMIT_EN,
        .conv_limit_num = 200,
        .sample_freq_hz = 1000*16,
        .conv_mode = ADC_CONV_MODE,
        .format = **ADC_OUTPUT_TYPE,**
    };

typedef struct {
    bool conv_limit_en;                      ///< To limit ADC conversion times. Conversion stops after finishing `conv_limit_num` times conversion
    uint32_t conv_limit_num;                 ///< Set the upper limit of the number of ADC conversion triggers. Range: 1 ~ 255.
    uint32_t pattern_num;                    ///< Number of ADC channels that will be used
    adc_digi_pattern_config_t *adc_pattern;    ///< List of configs for each ADC channel that will be used
    uint32_t sample_freq_hz;  /*!< The expected ADC sampling frequency in Hz. Range: 611Hz ~ 83333Hz
                                   Fs = Fd / interval / 2
                                   Fs: sampling frequency;
                                   Fd: digital controller frequency, no larger than 5M for better performance
                                   interval: interval between 2 measurement trigger signal, the smallest interval should not be smaller than the ADC measurement period, the largest interval should not be larger than 4095 */
    adc_digi_convert_mode_t conv_mode;      ///< ADC DMA conversion mode, see `adc_digi_convert_mode_t`.
    adc_digi_output_format_t format;        ///< ADC DMA conversion output format, see `adc_digi_output_format_t`.
} adc_digi_configuration_t;
```

in this structure, .conv_limit_num and .sample_freq_hz make me confused.
the sample frequency of dma_read can up tp 2MHz, but in this structure, can i just get 83333hz as my maximum sample frequency? how can i get 2MHz?

i test the sample frequency is equal to the value of sample_freq_hz. how about conv_limit_num, what's its use?

```
uint32_t ret_num = 0;
void count_func(void *pvParam)
{
    while (1)
    {
        printf("count = %d,ret_num = %d\n", count_num,ret_num);
        count_num = 0;
        vTaskDelay(1000 / portTICK_PERIOD_MS);
    }
}

void adc_capture(void *pvParam)
{
    
    uint8_t result[TIMES] = {0};
    memset(result, 0xcc, TIMES);
    while (1)
    {
        adc_digi_read_bytes(result, TIMES, &ret_num, ADC_MAX_DELAY);
        count_num = count_num + (ret_num/2);  //the result is 2 bytes, so dived by 2.
    }
}
```

below is my debug result:
- count = 15936,ret_num = 128
- count = 16000,ret_num = 128
- count = 16000,ret_num = 128
- count = 16000,ret_num = 128

so i have 2 quetions:
1. how can i get 2MHz sample frequency?
2. what's the use of conv_limit_num? can it affect sample frequency?
what I'm talking maybe are very confusing i think, my english is not skilled. forgive me.

kurau1, 2022-09-12:
i find the way to improve adc sample frequency!
```
static void adc_hal_digi_sample_freq_config(adc_hal_context_t *hal, uint32_t freq)
{
#if !CONFIG_IDF_TARGET_ESP32
    //uint32_t interval = APB_CLK_FREQ / (ADC_LL_CLKM_DIV_NUM_DEFAULT + ADC_LL_CLKM_DIV_A_DEFAULT / ADC_LL_CLKM_DIV_B_DEFAULT + 1) / 2 / freq;
    uint32_t interval = APB_CLK_FREQ / (ADC_LL_CLKM_DIV_NUM_DEFAULT + ADC_LL_CLKM_DIV_A_DEFAULT / ADC_LL_CLKM_DIV_B_DEFAULT + 1) / 200 / freq;
    // set sample interval
    adc_ll_digi_set_trigger_interval(interval);
    // Here we set the clock divider factor to make the digital clock to 5M Hz
    adc_ll_digi_controller_clk_div(ADC_LL_CLKM_DIV_NUM_DEFAULT, ADC_LL_CLKM_DIV_B_DEFAULT, ADC_LL_CLKM_DIV_A_DEFAULT);
    adc_ll_digi_clk_sel(0); // use APB
#else
    i2s_ll_rx_clk_set_src(hal->dev, I2S_CLK_D2CLK); /*!< Clock from PLL_D2_CLK(160M)*/
    uint32_t bck = I2S_BASE_CLK / (ADC_LL_CLKM_DIV_NUM_DEFAULT + ADC_LL_CLKM_DIV_B_DEFAULT / ADC_LL_CLKM_DIV_A_DEFAULT) / 2 / freq;
    i2s_ll_mclk_div_t clk = {
        .mclk_div = ADC_LL_CLKM_DIV_NUM_DEFAULT,
        .a = ADC_LL_CLKM_DIV_A_DEFAULT,
        .b = ADC_LL_CLKM_DIV_B_DEFAULT,
    };
    i2s_ll_rx_set_clk(hal->dev, &clk);
    i2s_ll_rx_set_bck_div_num(hal->dev, bck);
#endif
}
```
this function is in `adc_hal.c`

i change default number `2 `  to `200` in   `uint32_t interval = APB_CLK_FREQ / (ADC_LL_CLKM_DIV_NUM_DEFAULT + ADC_LL_CLKM_DIV_A_DEFAULT / ADC_LL_CLKM_DIV_B_DEFAULT + 1) / 200 / freq;` can change the sample frequency and can break the 83333Hz limitation 。

below are bebug result：
```
count = 1183552,ret_num = 128,adc_value = 0
count = 1183552,ret_num = 128,adc_value = 0
count = 1183616,ret_num = 128,adc_value = 0
count = 1183552,ret_num = 128,adc_value = 0
count = 1183552,ret_num = 128,adc_value = 0
```
i set the `.sample_freq_hz = 10000` and `200/2 = 100`, so i get `10000*100 = 1000000Hz` . but there are some error when high frequency.

Icarus113, 2022-09-13:
Hi @kurau1 , thanks for your report. WRT your questions:
>  how can i get 2MHz sample frequency?

From the HW side, you can only get 2MHz on ESP32. Up to now, on other chips, you can only get 83333 Hz freq at max. WRT  your  way to modify the frequency divider, it's correct. But HW may get wrong, when in a frequency higher than 83333 Hz.

> what's the use of conv_limit_num? can it affect sample frequency?

It's a legacy configuration public to users. It's deprecated on v5.0 and later. It works combining with `conv_limit_en`. On chips other than ESP32, if `conv_limit_en` is set 1, HW will stop when `conv_limit_num` number of bytes are converted (into digital bits). If set to 0, then just a normal continuous loop doing conversion. On ESP32, you shouldn't set `conv_limit_en` to 0 (which means it must always be set to 1). 
But again, these 2 configurations are deprecated. I'd suggest migrating to [this header on v5.0 and later](https://github.com/espressif/esp-idf/blob/master/components/esp_adc/include/esp_adc/adc_continuous.h)

kurau1, 2022-09-14:
HI!@Icarus113, thanks for your answers. these answers are very clear. 
for your fisrt answer, my project needs high sample frequency, so i have to break the limitation. i don't get any wrong now, just now haha. so i wanna have a try. 
for your second answer. say it again, it's so clear! i understand your meanings. But now, i am thinking adc would always capture the value if `conv_limit_en ` is set 0? WRT the low power of project.  I wanna adc stop when i don't need to get the value of signal. This time is about 1s, After 1s, i restart adc and get enough data then send them by wifi_STA. And then stop it again. It's a cycle. i find 2 API and try to use `adc_digi_stop();` and `adc_digi_start();` to stop adc and restart adc. But i will get wrong.：
```
assert failed: xQueueGenericSend queue.c:832 (pxQueue->pcHead != ((void *)0) || pxQueue->u.xSemaphore.xMutexHolder == ((void *)0) || pxQueue->u.xSemaphore.xMutexHolder == xTaskGetCurrentTaskHandle())

Backtrace:0x40023902:0x3ffdf1900x4002a87d:0x3ffdf1b0 0x40031469:0x3ffdf1d0 0x4002b07f:0x3ffdf2f0 0x40024ccd:0x3ffdf330 0x40024d7d:0x3ffdf360 0x400888ba:0x3ffdf380 0x40086e06:0x3ffdf3a0 0x4002d305:0x3ffdf3c0
0x40023902: panic_abort at F:/ESP-IDF/Espressif/frameworks/esp-idf-v4.4.2/components/esp_system/panic.c:402

0x4002a87d: esp_system_abort at F:/ESP-IDF/Espressif/frameworks/esp-idf-v4.4.2/components/esp_system/esp_system.c:128

0x40031469: __assert_func at F:/ESP-IDF/Espressif/frameworks/esp-idf-v4.4.2/components/newlib/assert.c:85

0x4002b07f: xQueueGenericSend at F:/ESP-IDF/Espressif/frameworks/esp-idf-v4.4.2/components/freertos/queue.c:830 (discriminator 8)

0x40024ccd: lock_release_generic at F:/ESP-IDF/Espressif/frameworks/esp-idf-v4.4.2/components/newlib/locks.c:201

0x40024d7d: _lock_release at F:/ESP-IDF/Espressif/frameworks/esp-idf-v4.4.2/components/newlib/locks.c:207

0x400888ba: adc_digi_stop at F:/ESP-IDF/Espressif/frameworks/esp-idf-v4.4.2/components/driver/adc.c:452

0x40086e06: adc_capture at F:/Project_Visual/Pro4_dma_Read_2/dma_read/main/my_ADC_DMA/my_ADC_DMA.c:38

0x4002d305: vPortTaskWrapper at F:/ESP-IDF/Espressif/frameworks/esp-idf-v4.4.2/components/freertos/port/xtensa/port.c:131

ELF file SHA256: 0bcce9f354e16584

Rebooting...
ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0x3 (RTC_SW_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x4002346d
0x4002346d: esp_restart_noos_dig at F:/ESP-IDF/Espressif/frameworks/esp-idf-v4.4.2/components/esp_system/esp_system.c:44 (discriminator 1)

SPIWP:0xee
mode:DIO, clock div:1
load:0x3ffe6100,len:0x1770
load:0x4004c000,len:0xa80
load:0x40050000,len:0x2e58
SHA-256 comparison failed:
Calculated: 69792dccdfd640125e17fadc06f51b1dc38c7bdc9aa2d84e700e7a4935a6766f
Expected: 10542a7316688db6771f59bf24d8128f87a66de98b00c4c108d70075b421b5e4
Attempting to boot anyway...

entry 0x4004c210
```

Do i need reinitialize the adc？ How can i accomplish stop and restart? I can't find more API
If you know it. if you have time. I'm looking forward for your reply.
Have a good day.

kurau1, 2022-09-19:
Don't change the default number` 2`, the max frequency is 83333Hz. Although the test sample frequency will increase, but in fact, the neighboring adc samples are the same. What i did make interval is smaller than the ADC measurement period. So, i choose ESP32 now to higher sample frequency.

Icarus113, 2022-09-21:
Hi @kurau1 , 
> Although the test sample frequency will increase, but in fact, the neighboring adc samples are the same

Yes, this could be a situation, when you violate the max freq limitation.

WRT your previous crash, it's possibly because you call the ADC APIs in an ISR context.

WRT your requirement of stopping ADC continuous work after a certain time:
If you use IDF master version, you'll see this [callback](https://github.com/espressif/esp-idf/blob/master/components/esp_adc/include/esp_adc/adc_continuous.h#L99). Please also read the comment to know the details. In this callback, you can notify your task to stop the ADC.

Due to shared hardware issue, ADC related APIs are not supposed to be used in an ISR context. We're considering adding few APIs to make it possible.

kurau1, 2022-09-21:
HI,@trustmiao 
what i use is idf V4.4. so the frequency scope is different.
in V4.4, i can't also break the limitation of 83333(i forget its minimum limitation, but it has) just like u can't break the minimum limitation. That's my thinking. HW will have some problems if u broke the rule.
WRT your problem of frequency. I can't undestand your program. i am a rookie. if u don't mind, i give my suggstion :Try to count like what i do. adding ret_num and printfing it after 1S delay then clear it. I gave the program in my first massage.

trustmiao, 2022-09-21:
1. it's so big difference in max min hz in idf 4.4 and 5 , this is not reasonable.
2. those part of code is basically get result from adc_continuous_read, count every returned data, when it adds to 1000 datas, plot the time to check fps. From observation, it's 16k ,it's not 20k hz as i set.

kurau1, 2022-09-21:
Hello,@Icarus113 
Sorry, i miss your suggestion about using IDF v5.0. I will try tomorrow and look for callback.
i can't undersand, i use `adc_digi_start()` and `adc_digi_stop()` in a task function. Is task function an ISR?
task function is like this:
```
void adc_capture(void *pvParam)
{
    uint8_t result[TIMES] = {0};
    memset(result, 0xcc, TIMES);
    while (1)
    {
        adc_digi_read_bytes(result, TIMES, &ret_num, ADC_MAX_DELAY);
        count_num = count_num + (ret_num/2);  //the result is 2 bytes, so dived by 2.
        if(true)    //this 'if' is a example, my code is in my lib PC, forgive me.
          {
          adc_digi_start();
          vTaskDelay(1000 / portTICK_PERIOD_MS);
          adc_digi_stop()
          }

    }
}
```
using like above is like using API in ISR? i can't understand.

kurau1, 2022-09-21:
> 1. it's so big difference in max min hz in idf 4.4 and 5 , this is not reasonable.
> 2. those part of code is basically get result from adc_continuous_read, count every returned data, when it adds to 1000 datas, plot the time to check fps. From observation, it's 16k ,it's not 20k hz as i set.

i don't know if i can break the limitation of 83333Hz in IDF v5.0. HAHA. i wish i can.

kurau1, 2022-09-24:
> My issue is closely related, therefore i decided to put up my discovery here. if it is not appropriate , i may make a new issue instead.
> 
> In my case, esp32 idf5, vscode. the minimun sample_freq_hz i could set is 20k, as the range is defined as 20k-2kk in "soc/soc_caps.h"
> 
> ```
> #define SOC_ADC_SAMPLE_FREQ_THRES_HIGH          (2*1000*1000)
> #define SOC_ADC_SAMPLE_FREQ_THRES_LOW           (20*1000)
> ```
> 
> which is different from @kurau1 's 83333hz. and the bottom line is also different, if i set sample_freq_hz = 15k, i will give error: `E (339) adc_continuous: adc_continuous_config(543): ADC sampling frequency out of range`
> 
> Another interesting thing is , though I dfined sample_freq_hz = 20k, conv_frame_size = 100, .max_store_buf_size = 4096, i get actually got 16khz
> 
> ```
> ulTaskNotifyTake(pdTRUE, portMAX_DELAY);
>         ret = adc_continuous_read(handle, result, EXAMPLE_READ_LEN, &ret_num, 0);
>         count+=ret_num/SOC_ADC_DIGI_RESULT_BYTES;
>         if(count>1000){
>         long t=esp_timer_get_time();//micro second
>         int fps=1000000/(t-lastt)*1000;
>         printf("1 time: %d ret: %d ret_num: %d fps: %d divider is %d \n",(int)t,ret,(int)ret_num,fps,(int)SOC_ADC_DIGI_RESULT_BYTES);
>         lastt=t;
>         count-=1000;
>         }
> ```
> 
> ```
> 1 time: 5063768 ret: 0 ret_num: 100 fps: 16000 divider is 2 
> 1 time: 5124791 ret: 0 ret_num: 100 fps: 16000 divider is 2 
> 1 time: 5185815 ret: 0 ret_num: 164 fps: 16000 divider is 2 
> 1 time: 5246838 ret: 0 ret_num: 100 fps: 16000 divider is 2 
> 1 time: 5307864 ret: 0 ret_num: 32 fps: 16000 divider is 2 
> 1 time: 5368885 ret: 0 ret_num: 100 fps: 16000 divider is 2 
> 1 time: 5429909 ret: 0 ret_num: 100 fps: 16000 divider is 2 
> 1 time: 5490932 ret: 0 ret_num: 100 fps: 16000 divider is 2 
> 1 time: 5551955 ret: 0 ret_num: 100 fps: 16000 divider is 2 
> ```
> 
> is it because @kurau1 and I are using different version of esp32? what made the frequency smaller than sample_freq_hz i set. thanks

Hello, @trustmiao 
Could u tell my how to upgrade ESP-IDF V5? I am in China, i can't uninstall old ESP-IDF and ESP-TOOL then install new ESP-IDF and ESP-TOOL by `esp-idf-tools-setup-online-2.16.exe`(because it's so hard to link GITHUB during install progress). I have spent 2 days in this issue. It's a torture. 
Could u tell me your install progress or if have some guides?

Thanks, have a good day!

kurau1, 2022-09-30:
> 1. it's so big difference in max min hz in idf 4.4 and 5 , this is not reasonable.
> 2. those part of code is basically get result from adc_continuous_read, count every returned data, when it adds to 1000 datas, plot the time to check fps. From observation, it's 16k ,it's not 20k hz as i set.

it's so big difference in max min hz inesp32 and esp32s2，not in idf 4.4 and 5. In IDF5, in addition to esp32, the sampling rate range of other chips is still 611~83333hz. the sampling rate range of esp32 is 20k~2Mhz

closed 2022-11-25 by kurau1