espressif/esp-idf #9769: BT_HIDD: Data size over 3!

XiaoLiuAI, 2022-09-12:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.2

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### Development Kit.

ESP32-WROOM-32E

### Power Supply used.

USB

### What is the expected behavior?

As described in the README, type key and move mouse via bluetooth.

### What is the actual behavior?

`E (11258263) BT_HIDD: Data size over 3!`

### Steps to reproduce.

1. copy example ESP_HID_DEVICE from esp-idf/examples
2. idf.py build
3. idf.py PORT flash monitor
4. type any key as described in the help message

### Debug Logs.

```plain
I (11250123) BT_HIDD: Connected to 3c:22:fb:05:0a:e9
I (11250123) HID_DEV_BT: CONNECT OK
I (11250133) HID_DEV_BT: Setting to non-connectable, non-discoverable
########################################################################
BT hid mouse demo usage:
You can input these value to simulate mouse: 'q', 'w', 'e', 'a', 's', 'd', 'h'
q -- click the left key
w -- move up
e -- click the right key
a -- move left
s -- move down
d -- move right
h -- show the help
########################################################################

E (11250173) BT_HIDD: Data size over 3!
########################################################################
BT hid mouse demo usage:
You can input these value to simulate mouse: 'q', 'w', 'e', 'a', 's', 'd', 'h'
q -- click the left key
w -- move up
e -- click the right key
a -- move left
s -- move down
d -- move right
h -- show the help
########################################################################

W (11255313) BT_HCI: hci cmd send: sniff: hdl 0x80, intv(10 18)
W (11255333) BT_HCI: hcif mode change: hdl 0x80, mode 2, intv 18, status 0x0
I (11255333) ESP_HID_GAP: BT GAP MODE_CHG_EVT mode:2
########################################################################
BT hid mouse demo usage:
You can input these value to simulate mouse: 'q', 'w', 'e', 'a', 's', 'd', 'h'
q -- click the left key
w -- move up
e -- click the right key
a -- move left
s -- move down
d -- move right
h -- show the help
########################################################################

E (11258263) BT_HIDD: Data size over 3!
```

boblane1, 2022-09-13:
Hi @XiaoLiuAI，

Thanks for your report! Please check if you have changed the descriptor `mouseReportMap` in your example. If you want to change the descriptor, please remember to change the sending procedure as well.

XiaoLiuAI, 2022-09-14:
> Hi @XiaoLiuAI，
> 
> Thanks for your report! Please check if you have changed the descriptor `mouseReportMap` in your example. If you want to change the descriptor, please remember to change the sending procedure as well.

I didn't change any thing, just as described, copy and build, flash, monitor.

boblane1, 2022-09-14:
Have you ever received logs like `PROTOCOL MODE[0]: BOOT`? If the HID mouse device is in BOOT mode, the data size shall be 3. If not, would you mind trying to apply the follow patch, which add some debug logs.

[0001-test.zip](https://github.com/espressif/esp-idf/files/9564483/0001-test.zip)

XiaoLiuAI, 2022-09-23:
@boblane1 You are correct, it is in BOOT mode. What I do to change this behavior?

boblane1, 2022-09-26:
@XiaoLiuAI The right to change the mode of the HID device belongs to the HID host. You can check out if the HID host can initiate the `SET_PROTOCOL` command. If not, you can discard the wheel data in the function of `send_mouse` in BOOT mode, according to the definition in `Device Class Definition for Human Interface Devices (HID) version 1.11`.

[image]

closed 2023-04-25 by Alvin1Zhang