espressif/esp-idf #9770: API for user-defined advertisement interval

herculanodavi, 2022-09-12:
### Is your feature request related to a problem?

I am designing a low-power device that uses protocomm and BLE light sleep, but I have seen that the advertisement interval is not as big as I would like to, and that I cannot change it since it's hardcoded and no API is available to change it:

In `transports/protocomm_ble.c`
```c
static esp_ble_adv_params_t adv_params = {
    .adv_int_min         = 0x100,
    .adv_int_max         = 0x100,
    .adv_type            = ADV_TYPE_IND,
    .own_addr_type       = BLE_ADDR_TYPE_PUBLIC,
    .channel_map         = ADV_CHNL_ALL,
    .adv_filter_policy   = ADV_FILTER_ALLOW_SCAN_ANY_CON_ANY,
};
```

### Describe the solution you'd like.

I'd like a protocomm_ble function to set advertisement interval. Maybe the struct `protocomm_ble_config` could have the `esp_ble_adv_params_t` struct, then the original `adv_params` struct would contain all the defaults.

```
/**
 * @brief   Config parameters for protocomm BLE service
 */
typedef struct protocomm_ble_config {
    /**
     * BLE device name being broadcast at the time of provisioning
     */
    char         device_name[MAX_BLE_DEVNAME_LEN];

    ...

    /**
     * Advertisement parameters (PROPOSAL)
     */
    esp_ble_adv_params_t adv_params;

    ...

} protocomm_ble_config_t;

### Describe alternatives you've considered.

I tried to use NimBLE api directly, but I wasn't able to find a solution to change the interval after the advertisement had begun.

rahult-github, 2022-09-13:
Hi @herculanodavi ,

 Protocomm layer is basically created to give secured path for upper protocols like provisioning / local control transport. However, the current config structure fields was predominantly created to adhere to existing use cases. 
 
 In order to implement the requested changes , can you kindly share describe the use case which will assist us to make changes ?
 Is the end application directly using protocomm APIs like protocomm_ble_start ( which are currently designed to be used for provisioning ? ) 
 Can the end application not call nimble stack APIs directly ? ( something which existing sample applications in IDF exhibit ? )  
Is advertising the only functionality that the use case needs or any other BLE functionality too ? How does the application do other operations ( say connection or GATT based operations ? ) 

As mentioned in original issue:

> I tried to use NimBLE api directly, but I wasn't able to find a solution to change the interval after the advertisement had begun.

Since i don't have the code that you tried, so just mentioning the APIs (In NimBLE) that could otherwise be used to achieve the above.

1. Stop Adv ( if adv is in progress )   [ ble_gap_adv_stop ]
2. Set new adv data params             [ Populate the ble_hs_adv_fields structure and then invoke ble_gap_adv_set_fields ]
3. Enable adv  ( This should advertise with new parameters)  [ ble_gap_adv_start]

Existing bleprph / blecent examples in IDF showcases usage of NimBLE APIs. Can you please check if those helps ? 

Thanks,
 Rahul

rahult-github, 2023-04-07:
@herculanodavi   Thanks for reporting, would you please help share if any further updates for the issue?

rahult-github, 2023-04-25:
Closing this issue. Feel free to reopen in case of any further comments.

closed 2023-04-25 by rahult-github