espressif/esp-idf #9772: SDCard Performance Issue

AAJAY5, 2022-09-12:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

## I have run SDCard Benchmark for `Arduino and ESPIDF` framework for the same hardware setup. And ESPIDF `SDSPI` benchmark seems very poor.

```c
/*
########### COMMON CONFIGURATIONS ###########
# Board: ESP32 DOIT DEVKIT V1
# SDCard Adapter: [Link][https://www.electronicscomp.com/micro-sd-card-module-india?gclid=CjwKCAjwsfuYBhAZEiwA5a6CDOc3-TJetLBaSR2x-z-lwNNHwusR9L57QjiJzJT6Wh1_PPE480NnuBoC4qEQAvD_BwE]
# CPU Speed: 240MHz
# Mode: SDSPI
# Freq: 4MHz
*/

#define TIME_ARRAY_SIZE     100
#define WRITE_BUFFER_SIZE   (16 * 1024)
#define PRINT_DIFF          0

uint64_t time_array[TIME_ARRAY_SIZE];
char write_buffer[WRITE_BUFFER_SIZE]

/*
# Framework: ESPIDF
############# Benchmark Results #########
write buffer size = 1024
maximum = 174 millsec.
minimum = 51 millsec.
write buffer size = 2048
maximum = 196 millsec.
minimum = 83 millsec.
write buffer size = 4096
maximum = 339 millsec.
minimum = 126 millsec.
write buffer size = 8192
maximum = 323 millsec.
minimum = 211 millsec.
*/
void sdBenchi(size_t writeSize) {
    // Check if destination file exists before writing
    struct stat st;
    if (stat("/sdcard/hello.txt", &st) == 0) {
        // Delete it if it exists
        unlink("/sdcard/hello.txt");
    }

    FILE *f = fopen("/sdcard/hello.txt", "w");
    if (f == NULL) {
        ESP_LOGE(TAG, "Failed to open file for writing");
        return;
    }

    // Get time in microseconds since boot.
    uint64_t start = esp_timer_get_time();
    for (int counter = 0; counter < TIME_ARRAY_SIZE; counter++) {
        fwrite(write_buffer, 1, writeSize, f);
        time_array[counter] = esp_timer_get_time();
    }
    fclose(f);
    ESP_LOGD(TAG, "File written");

    uint64_t sum = 0;
    uint64_t maximum = 0;
    uint64_t minimum = UINT64_MAX;
    for (int i = 0; i < TIME_ARRAY_SIZE; i++) {
        uint64_t end = time_array[i];
        uint64_t diff = end - start;
#if PRINT_DIFF
        printf("%d: start=%llu, end=%llu, diff=%llu\n", i, start, end, diff);
#endif
        maximum = (diff > maximum) ? diff : maximum;
        minimum = (diff < minimum) ? diff : minimum;
        sum += diff;
        start = end;
    }
    // uint64_t average = sum / TIME_ARRAY_SIZE;
    printf("write buffer size = %d\n", writeSize);
    printf("maximum = %lld millsec.\n", maximum / 1000);
    // printf("average = %lld millsec.\n", average/1000);
    printf("minimum = %lld millsec.\n", minimum / 1000);
}

/*
# Framework: Arduino
############# Benchmark Results #########
write buffer size = 1024
maximum = 12 millsec.
minimum = 3 millsec.
write buffer size = 2048
maximum = 14 millsec.
minimum = 5 millsec.
write buffer size = 4096
maximum = 26 millsec.
minimum = 10 millsec.
write buffer size = 8192
maximum = 27 millsec.
minimum = 19 millsec.
*/
void sdBenchi(size_t writeSize) {
    // Check if destination file exists before writing
    SD.remove("/hello.txt");

    // Get time in microseconds since boot.
    uint64_t start = esp_timer_get_time();
    File f = SD.open("/hello.txt", "w", true);
    for (int counter = 0; counter < TIME_ARRAY_SIZE; counter++) {
        if (f == NULL) {
            return;
        }
        f.write((const uint8_t *)write_buffer, writeSize);
        time_array[counter] = esp_timer_get_time();
    }
    f.close();

    uint64_t sum = 0;
    uint64_t maximum = 0;
    uint64_t minimum = UINT64_MAX;
    for (int i = 0; i < TIME_ARRAY_SIZE; i++) {
        uint64_t end = time_array[i];
        uint64_t diff = end - start;
#if PRINT_DIFF
        printf("%d: start=%llu, end=%llu, diff=%llu\n", i, start, end, diff);
#endif
        maximum = (diff > maximum) ? diff : maximum;
        minimum = (diff < minimum) ? diff : minimum;
        sum += diff;
        start = end;
    }
    // uint64_t average = sum / TIME_ARRAY_SIZE;
    printf("write buffer size = %d\n", writeSize);
    printf("maximum = %lld millsec.\n", maximum / 1000);
    // printf("average = %lld millsec.\n", average/1000);
    printf("minimum = %lld millsec.\n", minimum / 1000);
}

// Testing 
sdBenchi(1024*1);
sdBenchi(1024*2);
sdBenchi(1024*4);
sdBenchi(1024*8);
```

igrr, 2022-09-12:
@AAJAY5 Could you please include the code where you are initializing `sdspi` in IDF and `SD` in Arduino, as well?

AAJAY5, 2022-09-12:
> @AAJAY5 Could you please include the code where you are initializing `sdspi` in IDF and `SD` in Arduino, as well?

## ESPIDF
- Initialized and mounted using ESPIDF examples (SDSPI).

```C
#define TAG         "sdcard"
#define mount_point "/sdcard"
sdmmc_card_t *card;

esp_err_t sd_card_mount(void) {
    esp_err_t ret;

    // Options for mounting the filesystem.
    // If format_if_mount_failed is set to true, SD card will be partitioned and
    // formatted in case when mounting fails.
    esp_vfs_fat_sdmmc_mount_config_t mount_config = {
#if 1//def CONFIG_EXAMPLE_FORMAT_IF_MOUNT_FAILED
        .format_if_mount_failed = true,
#else
        .format_if_mount_failed = false,
#endif  // EXAMPLE_FORMAT_IF_MOUNT_FAILED
        .max_files = 5,
        .allocation_unit_size = 64 * 1024};
    
    
    ESP_LOGI(TAG, "Initializing SD card");

    // Use settings defined above to initialize SD card and mount FAT filesystem.
    // Note: esp_vfs_fat_sdmmc/sdspi_mount is all-in-one convenience functions.
    // Please check its source code and implement error recovery when developing
    // production applications.
    ESP_LOGI(TAG, "Using SPI peripheral");

    sdmmc_host_t host = SDSPI_HOST_DEFAULT();
    host.slot = SPI3_HOST;
    host.max_freq_khz = 4000;

    spi_bus_config_t bus_cfg = {
        .mosi_io_num = PIN_NUM_MOSI,
        .miso_io_num = PIN_NUM_MISO,
        .sclk_io_num = PIN_NUM_CLK,
        .quadwp_io_num = -1,
        .quadhd_io_num = -1,
        .max_transfer_sz = 0,
    };
    ret = spi_bus_initialize(host.slot, &bus_cfg, SDSPI_DEFAULT_DMA);
    if (ret != ESP_OK) {
        ESP_LOGE(TAG, "Failed to initialize bus.");
        return ret;
    }

    // This initializes the slot without card detect (CD) and write protect (WP) signals.
    // Modify slot_config.gpio_cd and slot_config.gpio_wp if your board has these signals.
    sdspi_device_config_t slot_config = SDSPI_DEVICE_CONFIG_DEFAULT();
    slot_config.gpio_cs = PIN_NUM_CS;
    slot_config.host_id = host.slot;

    ESP_LOGI(TAG, "Mounting filesystem");
    ret = esp_vfs_fat_sdspi_mount(mount_point, &host, &slot_config, &mount_config, &card);

    if (ret != ESP_OK) {
        if (ret == ESP_FAIL) {
            ESP_LOGE(TAG,
                     "Failed to mount filesystem. "
                     "If you want the card to be formatted, set the CONFIG_EXAMPLE_FORMAT_IF_MOUNT_FAILED menuconfig option.");
        } else {
            ESP_LOGE(TAG,
                     "Failed to initialize the card (%s). "
                     "Make sure SD card lines have pull-up resistors in place.",
                     esp_err_to_name(ret));
        }
        return ret;
    }
    ESP_LOGI(TAG, "Filesystem mounted");
    // Card has been initialized, print its properties
    sdmmc_card_print_info(stdout, card);
    return ret;
}

esp_err_t sd_card_unmount(void) {
    esp_err_t ret;
    // All done, unmount partition and disable SPI peripheral
    ret = esp_vfs_fat_sdcard_unmount(mount_point, card);
    ESP_LOGI(TAG, "Card unmounted");

    // deinitialize the bus after all devices are removed
    // spi_bus_free(host.slot);
    return ret;
}
```

## Arduino
```C
if (!SD.begin(21)) {
        Serial.println("Card Mount Failed");
}
```

igrr, 2022-09-12:
I see, thanks for adding the extra details.

Currently SDSPI driver in ESP-IDF uses one of three frequencies after the card probing is done:
* "probing" frequency — 400 kHz
* "default speed" frequency — 20 MHz. (It should be 25 MHz according to the SD specification, but we are using 20 since we can't produce 25 MHz exactly.)
* "high speed" frequency — 40 MHz. (Should be 50 MHz according to the spec, using 40 for the same reason.)

For any other value specified in sdmmc_host_t::max_freq_khz, we use the next lowest frequency. For example, if specifying max_freq_khz=30000 (30 MHz), "default speed" frequency (20 MHz) will be used. If 4 MHz is specified, 400 kHz will be used instead.

There is an open issue (originally created in the Arduino project, https://github.com/espressif/arduino-esp32/issues/6225) to allow arbitrary frequencies to be specified in `max_freq_khz`. This feature is now in progress, it should be released in IDF v5.1.

In the meantime, you can try using same frequency for benchmarking both libraries, e.g. 20 MHz. I also suggest checking the communication with a logic analyzer to make sure that the same frequency is actually being used for data transfer in both cases.

AAJAY5, 2022-09-12:
# max_freq_khz = 4000
```c
Name: 00000
Type: SDHC/SDXC
Speed: 4 MHz
Size: 15238MB
write buffer size = 1024
maximum = 50 millsec.
minimum = 23 millsec.
write buffer size = 2048
maximum = 71 millsec.
minimum = 44 millsec.
write buffer size = 4096
maximum = 114 millsec.
minimum = 87 millsec.
write buffer size = 8192
maximum = 247 millsec.
minimum = 172 millsec.
```

# max_freq_khz = 10000
```c
Name: 00000
Type: SDHC/SDXC
Speed: 10 MHz
Size: 15238MB
write buffer size = 1024
maximum = 50 millsec.
minimum = 23 millsec.
write buffer size = 2048
maximum = 77 millsec.
minimum = 44 millsec.
write buffer size = 4096
maximum = 114 millsec.
minimum = 87 millsec.
write buffer size = 8192
maximum = 199 millsec.
minimum = 172 millsec.
```

# max_freq_khz = 20000
```C
// E (809) sdcard: Failed to initialize the card (ESP_ERR_INVALID_CRC). Make sure SD card lines have pull-up resistors in place.
```

igrr, 2022-09-12:
Ah yes, the unfortunate downside of 20 MHz is that you might not be able to get it working on a breakout board.

Currently 4 MHz and 10 MHz settings would both result in 400 kHz frequency being used, as I explained above.

At the moment you can either try to make the connections between the ESP and the breakout board shorter or switch from the breakout board to a development board with an on-board SD card slot. That should help you get 20 MHz setting working.

AAJAY5, 2022-09-12:
> Ah yes, the unfortunate downside of 20 MHz is that you might not be able to get it working on a breakout board.
>
> Currently 4 MHz and 10 MHz settings would both result in 400 kHz frequency being [...]

Then why it's working well with Arduino Framework? My hardware setup is same for both test.

igrr, 2022-09-12:
The reason is that Arduino `SD` library doesn't use ESP-IDF `sdspi` driver under the hood, and it is actually using the 4 MHz frequency you have requested.

AAJAY5, 2022-09-12:
# Changed following file and getting good benchmark on ESPIDF framework with same hardware.
- File: `sdmmc_types.h`
- #define SDMMC_FREQ_PROBING    4000         /*!< SD/MMC probing speed */ Before: 400
[image]

igrr, 2022-09-12:
Yes, that would do as a workaround if you don't mind modifying IDF source code!

nopnop2002, 2022-10-01:
@igrr 

>This feature is now in progress, it should be released in IDF v5.1.

IDE V5.1 has been released, does it include this feature?

nopnop2002, 2022-10-01:
@igrr 

I took a look here.
I don't know how to enable this feature.

https://github.com/espressif/esp-idf/tree/master/examples/storage/sd_card/sdmmc

nopnop2002, 2022-10-01:
@igrr 

Thank you.

pacucha42, 2022-10-10:
Hi @nopnop2002, could you please confirm the fix works for you? If so, I will close this issue as all related updates are already merged in the master branch

pacucha42, 2022-10-11:
Closing the issue - feel free to reopen or create a new one if needed. Thank you for your report.

closed 2022-10-11 by pacucha42