espressif/esp-idf #9773: Mesh: invalid child

piro-pp, 2022-09-12:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1-dev-644-g867745a05c

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### Development Kit.

ESP32_DevKitc_V4

### Power Supply used.

USB

### What is the expected behavior?

Mesh with multiple devices should be stable when single devices perform resets.

### What is the actual behavior?

Sometimes it works fine, but sometimes the reseted device connects to the root and log output seems to be fine. But the root prints
periodically  prints "mesh: 1316[recv]invalid child <mac address>" and mesh doesn't receive data.

### Steps to reproduce.

1. select, configure and build examples/mesh/internal_communication (but works with other mesh examples too)
2. run firmware on at least 3 boards
3. reset some board (NODE role)

### Debug Logs.

```plain
Log of the root node:
W (325496) mesh_main: [#RX:191/191][L:1] parent:e0:28:6d:7f:0d:f8, receive from e0:e2:e6:9b:7f:e4, size:1460, heap:157844, flag:0[err:0x0, proto:0, tos:0]
W (327906) mesh_main: [#RX:192/192][L:1] parent:e0:28:6d:7f:0d:f8, receive from e0:e2:e6:9b:7f:e4, size:1460, heap:157844, flag:0[err:0x0, proto:0, tos:0]
W (330956) mesh_main: [#RX:193/193][L:1] parent:e0:28:6d:7f:0d:f8, receive from e0:e2:e6:9b:7f:e4, size:1460, heap:157844, flag:0[err:0x0, proto:0, tos:0]
# one node is reseted
I (332196) mesh_main: <MESH_EVENT_CHILD_DISCONNECTED>aid:2, 8c:4b:14:1d:cc:f0
W (332196) mesh_main: <MESH_EVENT_ROUTING_TABLE_REMOVE>remove 1, new:3, layer:1
I (332196) wifi:new:<1,1>, old:<1,1>, ap:<1,1>, sta:<1,0>, prof:1
I (332206) wifi:Send SA Query req with transaction id 201a
I (332416) wifi:Send SA Query req with transaction id 6878
I (332616) wifi:Send SA Query req with transaction id cc6c
W (332646) mesh_main: [#RX:194/194][L:1] parent:e0:28:6d:7f:0d:f8, receive from e0:e2:e6:9b:7f:e4, size:1460, heap:157844, flag:0[err:0x0, proto:0, tos:0]
I (332826) wifi:Send SA Query req with transaction id 7da5
I (333026) wifi:Send SA Query req with transaction id ce05
I (333236) wifi:STA not responded to 5 SA Query attempts, Reset connection sending disassoc
I (333236) wifi:station: 8c:4b:14:1d:cc:f0 leave, AID = 2, bss_flags is 134243, bss:0x3ffd60e4
I (333236) wifi:new:<1,1>, old:<1,1>, ap:<1,1>, sta:<1,0>, prof:1
I (333246) mesh_main: <MESH_EVENT_CHILD_DISCONNECTED>aid:2, 8c:4b:14:1d:cc:f0
I (333256) mesh: <nvs>write assoc:2
I (333756) wifi:new:<1,1>, old:<1,1>, ap:<1,1>, sta:<1,0>, prof:1
I (333756) wifi:station: 8c:4b:14:1d:cc:f0 join, AID=2, bgn, 40U
I (333826) mesh: 1316[recv]invalid child 8c:4b:14:1d:cc:f0
I (333826) mesh: 1316[recv]invalid child 8c:4b:14:1d:cc:f0
W (334346) mesh_main: [#RX:195/195][L:1] parent:e0:28:6d:7f:0d:f8, receive from e0:e2:e6:9b:7f:e4, size:1460, heap:157844, flag:0[err:0x0, proto:0, tos:0]
I (334446) mesh: 1316[recv]invalid child 8c:4b:14:1d:cc:f0
I (334446) mesh: 1316[recv]invalid child 8c:4b:14:1d:cc:f0
I (335026) mesh: 1316[recv]invalid child 8c:4b:14:1d:cc:f0
I (335026) mesh: 1316[recv]invalid child 8c:4b:14:1d:cc:f0
I (335636) mesh: 1316[recv]invalid child 8c:4b:14:1d:cc:f0
I (335636) mesh: 1316[recv]invalid child 8c:4b:14:1d:cc:f0
W (335786) mesh_main: [#RX:196/196][L:1] parent:e0:28:6d:7f:0d:f8, recei
```

piro-pp, 2022-09-15:
The problem seems to be that the Event MESH_EVENT_CHILD_DISCONNECTED comes two times for the same child while the second one should be a MESH_EVENT_CHILD_CONNECTED (as the child is connected to the parent then).

287662354, 2022-09-16:
你好 我也是相同的问题  请问你解决了吗     我直接下载的官方的例程/ex.ample/mesh/ip_intenal_commuicate

piro-pp, 2022-09-19:
The problem may be connected to the fact that i have a network setup with multiple access points (non-mesh routers). This problem seems to have disappeared when switching to a network setup with only one access point.

shenjun7, 2022-09-21:
Thanks for reporting. This problem is caused by pmf support for softAP. we will fix it later.

wishinlife, 2022-11-08:
I have the same problem on release/5.0. I found two problems from logs:
1. On one child node power off and power on quickly or reset the child node, the root logs print two times MESH_EVENT_CHILD_DISCONNECTED;
2. on child startup, the MESH_EVENT_ROUTING_TABLE_REMOVE event was triggered, but the child node mac not removed from  wifi_sta_list_t.

`I (132897) wifi:new:<11,2>, old:<11,2>, ap:<11,2>, sta:<0,0>, prof:11
I (132897) wifi:station: 7c:df:a1:f2:fd:30 join, AID=1, bgn, 40D
I (132902) WIFI-MESH: <MESH_EVENT_PS_CHILD_DUTY>cidx:0, 7c:df:a1:f2:fd:30, duty:0
W (132922) WIFI-MESH: <MESH_EVENT_ROUTING_TABLE_ADD>add 1, new:2, layer:1
I (132922) WIFI-MESH: <MESH_EVENT_CHILD_CONNECTED>aid:1, 7c:df:a1:f2:fd:30
W (142027) wifi:Haven't to connect to a suitable AP now!
I (142027) WIFI-MESH: System information, channel: 11, layer: 1, self mac: 7c:df:a1:f5:51:7c, parent bssid: 00:00:00:00:00:00, parent rssi: -120, node num: 2, free heap: 2194560
I (142037) WIFI-MESH: Child mac: 7c:df:a1:f2:fd:30
W (152027) wifi:Haven't to connect to a suitable AP now!
I (152027) WIFI-MESH: System information, channel: 11, layer: 1, self mac: 7c:df:a1:f5:51:7c, parent bssid: 00:00:00:00:00:00, parent rssi: -120, node num: 2, free heap: 2194560
I (152037) WIFI-MESH: Child mac: 7c:df:a1:f2:fd:30
W (162027) wifi:Haven't to connect to a suitable AP now!
I (162027) WIFI-MESH: System information, channel: 11, layer: 1, self mac: 7c:df:a1:f5:51:7c, parent bssid: 00:00:00:00:00:00, parent rssi: -120, node num: 2, free heap: 2194560
I (162037) WIFI-MESH: Child mac: 7c:df:a1:f2:fd:30
I (167602) WIFI-MESH: <MESH_EVENT_CHILD_DISCONNECTED>aid:1, 7c:df:a1:f2:fd:30
I (167602) mesh: [scan]new scanning time:300ms, beacon interval:100ms
I (167607) wifi:new:<11,2>, old:<11,2>, ap:<11,2>, sta:<0,0>, prof:11
I (167612) wifi:Send SA Query req with transaction id 6ac0
**_W (167607) WIFI-MESH: <MESH_EVENT_ROUTING_TABLE_REMOVE>remove 1, new:1, layer:1_**
I (167822) wifi:Send SA Query req with transaction id 6095
I (168027) wifi:Send SA Query req with transaction id 7a15
I (168232) wifi:Send SA Query req with transaction id b752
I (168437) wifi:Send SA Query req with transaction id 745d
I (168642) wifi:STA not responded to 5 SA Query attempts, Reset connection sending disassoc
I (168642) wifi:station: 7c:df:a1:f2:fd:30 leave, AID = 1, bss_flags is 134243, bss:0x3c109690
I (168647) wifi:new:<11,0>, old:<11,2>, ap:<11,2>, sta:<0,0>, prof:11
I (168657) WIFI-MESH: <MESH_EVENT_CHILD_DISCONNECTED>aid:1, 7c:df:a1:f2:fd:30
I (169162) wifi:new:<11,2>, old:<11,0>, ap:<11,2>, sta:<0,0>, prof:11
I (169167) wifi:station: 7c:df:a1:f2:fd:30 join, AID=1, bgn, 40D
I (169227) mesh: 1316[recv]invalid child 7c:df:a1:f2:fd:30
I (169732) mesh: 1316[recv]invalid child 7c:df:a1:f2:fd:30
I (170372) mesh: 1813[XON]async, from child 7c:df:a1:f2:fd:30
I (170732) mesh: 1316[recv]invalid child 7c:df:a1:f2:fd:30
I (171732) mesh: 1316[recv]invalid child 7c:df:a1:f2:fd:30
W (172027) wifi:Haven't to connect to a suitable AP now!
**_I (172027) WIFI-MESH: System information, channel: 11, layer: 1, self mac: 7c:df:a1:f5:51:7c, parent bssid: 00:00:00:00:00:00, parent rssi: -120, node num: 1, free heap: 2194560
I (172037) WIFI-MESH: Child mac: 7c:df:a1:f2:fd:30_**
I (172232) mesh: 1316[recv]invalid child 7c:df:a1:f2:fd:30
`

wishinlife, 2022-11-08:
Use the same example, If you specify the root node manually, and reset the root node , it takes a long time for the child nodes to recover after the root node is reset. It takes about 5 minutes to print 250 error retry logs.

wishinlife, 2022-11-10:
@shenjun7 Hi，Is there a fix schedule for this bug? I also has this problem with release/v5.0.

shenjun7, 2022-11-15:
We have fixed this problem internally and will sync it to github soon．

wishinlife, 2022-11-19:
@shenjun7 Look forward to releasing the fix soon.

wishinlife, 2022-11-29:
@zhangyanjiaoesp Hi, When will it be released?

wishinlife, 2022-12-01:
@zhangyanjiaoesp Thanks! This issue has been solved.
There are also 2 minor issues related to mesh networking:
According to the document, the channel can be 1 to 14, but the actual maximum can only be 11.
2. When the CONFIG_SW_COEXIST_ENABLE is turned on, Bluetooth must be started before the WiFi mesh, otherwise the panic_abort will be triggered.

zhangyanjiaoesp, 2022-12-01:
Thanks for the reporting the new issues. Thus this issue has been solved, I think we can close this task. And for the new issues, we can create a new task to track them.

Alvin1Zhang, 2022-12-01:
Thanks for reporting and sharing the updates, for the new issues, please help create a new issue, will close the ticket. Thanks.

closed 2022-12-01 by Alvin1Zhang