espressif/esp-idf #9774: SPI transmit assertion fail?

fpgamaster, 2022-09-12:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

ESP-IDF v4.2.2, ESP32-WROOM-32E

I have two SPI devices connected to VSPI host. Everything is working as expected but from time to time I am getting this:

assert failed: spi_device_transmit spi_master.c:865 (ret_trans == trans_desc)

Backtrace:0x40081fa2:0x3ffb33b00x40092b55:0x3ffb33d0 0x4009941d:0x3ffb33f0 0x4010d281:0x3ffb3510 0x400e03a9:0x3ffb3540 0x400e0474:0x3ffb3590
0x40081fa2: panic_abort at /home/me/esp/esp-idf-v4.4.2/components/esp_system/panic.c:402

0x40092b55: esp_system_abort at /home/me/esp/esp-idf-v4.4.2/components/esp_system/esp_system.c:128

0x4009941d: __assert_func at /home/me/esp/esp-idf-v4.4.2/components/newlib/assert.c:85

0x4010d281: spi_device_transmit at /home/me/esp/esp-idf-v4.4.2/components/driver/spi_master.c:865 (discriminator 1)

0x400e03a9: AMIS30543SPI::enc_transfer16(unsigned short) at /home/me/esp/esp32/my_project/build/../main/myg_driver_AMIS30543.h:159

        uint16_t enc_transfer16(uint16_t data) {
                spi_transaction_t t;
                esp_err_t ret;

                memset(&t, 0, sizeof(t));                       // Zero out the transaction
                t.length = 2 * 8;                               // Len is in bytes, transaction length is in bits.
                t.tx_data[0] = data >> 8;                       // Data
                t.tx_data[1] = data & 0xff;                     // Data
                t.flags = SPI_TRANS_USE_RXDATA | SPI_TRANS_USE_TXDATA;
                ret = spi_device_transmit(spi_enc, &t);         // Transmit!
                ESP_ERROR_CHECK(ret);

                ets_delay_us(3);

                return (((t.rx_data[0] << 8) | t.rx_data[1]) ^ 0xffff);
        }

What could be the root for the issue?

Best regards

KaeLL, 2022-09-12:
[This](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/spi_master.html#driver-features).

fpgamaster, 2022-09-21:
Hi @KaeLL, Thanks for the reply!

I do not access same device from different tasks but I am initializing the SPI bus from one task which adds two slave devices and accesses the first device only, the second device is accessed from a task created from the first task.
The access to the first device is stopped as the second task is created.
So, I do not think that this is the issue.
However, I have switched to polling mode to see if there will be any issues. For now all is fine.

Best regards

closed 2025-03-12 by Alvin1Zhang