espressif/esp-idf #9775: espNOW causes esp_transport_connect to fail

nicolas-Philips, 2022-09-13:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.2

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### Development Kit.

ESP-WROVER-KIT

### Power Supply used.

USB

### What is the expected behavior?

Hello,

I have a project with:

- Wifi in station mode
- espNOW
- azureIot 

I created a new minimum code project (https://github.com/nicolas-Philips/espNowIssue) with:

- wifi station example code: https://github.com/espressif/esp-idf/tree/d6d8324ad9b957f85ecb323ec0db12c4d03a93c4/examples/wifi/getting_started/station
- esp now example code: https://github.com/espressif/esp-idf/tree/d6d8324ad9b957f85ecb323ec0db12c4d03a93c4/examples/wifi/espnow
- The minimum code extracted from azureIot, just 46 lines of code

When calling esp_now_send, data is still received on the lwip socket by azureIot, but not all data, causing azureIot to timeout.
The expected behaviour is that  esp_now_send does not influence lwip sockets.

### What is the actual behavior?

Once the function esp_now_send is called, the esp32 always times out when trying to connect to azureIot. 
The output with embedtls debugging info enabled when the connection attempt times out:
```
I (5031) connTest: Start connTest example
I (5071) mbedtls: ssl_tls.c:5904 => handshake

I (5071) mbedtls: ssl_cli.c:4483 client state: 0

I (5071) mbedtls: ssl_msg.c:2102 => flush output

I (5081) mbedtls: ssl_msg.c:2114 <= flush output

I (5081) mbedtls: ssl_cli.c:4483 client state: 1

I (5091) mbedtls: ssl_msg.c:2102 => flush output

I (5091) mbedtls: ssl_msg.c:2114 <= flush output

I (5101) mbedtls: ssl_cli.c:999 => write client hello

I (5111) mbedtls: ssl_msg.c:2542 => write handshake message

I (5111) mbedtls: ssl_msg.c:2701 => write record
[... 194 lines cut ...]

		esp_transport_close( xTransport );

		ESP_LOGI(TAG, "Finished connTest example");

		vTaskDelay(1000 / portTICK_PERIOD_MS);
	}
}

void startConnTest()
{

	transportList = esp_transport_list_init();

	/* create new transport to the transport list */
	esp_transport_handle_t ssl = esp_transport_ssl_init();
	if(ssl == NULL)
	{
		esp_restart();
	}
	esp_transport_list_add(transportList, ssl, "_conntest");

	xTaskCreate(&conntest_task, "test_task", 8192, NULL, 5, NULL);
}
```

nicolas-Philips, 2022-09-21:
In short: using espNOW send causes LWIP socket to stop functioning correctly. It is not just connecting connecting as mentioned in the example code that starts failing, **but everything using LWIP sockets**.

zhangyanjiaoesp, 2022-10-09:
@nicolas-Philips Have you enable the SPIRAM option in the menuconfig?

nicolas-Philips, 2022-10-10:
@zhangyanjiaoesp Since anything not specifically reconfigured in the SDK is the default config (see step 2 of Steps to reproduce), the SPIRAM option was not set. 

I added two wireshark captures, one without espNow, where the connetion just keeps being created and closed (rather short capture). And one where an ESP packet is sent after about 16s (in the wireshark capture time). From then on, the ESP does not send any packet anymore and azure sends multiple [FIN, ACK] requests, trying to close 
the connection, without a response of the ESP. From then on the ESP just does not send out any packet anymore.

[wireshareCaptures.zip](https://github.com/espressif/esp-idf/files/9745793/wireshareCaptures.zip)

zhangyanjiaoesp, 2022-10-17:
@nicolas-Philips 
I have tested locally using your example, I can reproduce the problem on v4.4.2 branch, but when I change to master branch and [v5.0 branch](https://github.com/espressif/esp-idf/tree/v5.0-beta1), the test pass. 

> I (674) wifi:Init data frame dynamic rx buffer num: 32
I (684) wifi:Init management frame dynamic rx buffer num: 32
I (684) wifi:Init management short buffer num: 32
I (694) wifi:Init dynamic tx buffer num: 32
I (694) wifi:Init static rx buffer size: 1600
I (704) wifi:Init static rx buffer num: 10
I (704) wifi:Init dynamic rx buffer num: 32
I (704) wifi_init: rx ba win: 6
I (714) wifi_init: tcpip mbox: 32
I (714) wifi_init: udp mbox: 6
I (714) wifi_init: tcp mbox: 6
I (724) wifi_init: tcp tx win: 5744
I (724) wifi_init: tcp rx win: 5744
I (734) wifi_init: tcp mss: 1440
I (734) wifi_init: WiFi IRAM OP enabled
I (744) wifi_init: WiFi RX IRAM OP enabled
I (744) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07
I (844) wifi:mode : sta (30:ae:a4:80:15:c8)
I (844) wifi:enable tsf
I (844) wifi station: wifi_init_sta finished.
I (864) wifi:new:<6,0>, old:<1,0>, ap:<255,255>, sta:<6,0>, prof:1
I (864) wifi:state: init -> auth (b0)
I (874) wifi:state: auth -> assoc (0)
I (884) wifi:state: assoc -> run (10)
I (904) wifi:connected with test_zyj, aid = 2, channel 6, BW20, bssid = ac:84:c6:7e:31:2b
I (904) wifi:security: WPA2-PSK, phy: bgn, rssi: -22
I (904) wifi:pm start, type: 1

> W (914) wifi:<ba-add>idx:0 (ifx:0, ac:84:c6:7e:31:2b), tid:0, ssn:0, winSize:64
I (924) wifi:AP's beacon interval = 102400 us, DTIM period = 1
I (1634) esp_netif_handlers: sta ip: 192.168.0.191, mask: 255.255.255.0, gw: 192.168.0.1
I (1634) wifi station: got ip:192.168.0.191
I (1634) wifi station: connected to ap SSID:test_zyj password:12345678
I (1644) ESPNOW: espnow [version: 1.0] init
I (1644) connTest: Start connTest example
W (2094) wifi:<ba-add>idx:1 (ifx:0, ac:84:c6:7e:31:2b), tid:1, ssn:0, winSize:64
I (3924) connTest: Connected to server
I (3924) connTest: Finished connTest example
I (4934) connTest: Start connTest example
I (6644) espnow_example: Start sending broadcast data
I (6644) espnow_example: Receive 2th broadcast data from: 30:ae:a4:0c:34:ec, len: 10
I (6644) espnow_example: Receive 6th broadcast data from: 30:ae:a4:0c:34:ec, len: 10
I (7114) connTest: Connected to server
I (7114) connTest: Finished connTest example
I (7664) espnow_example: send data to ff:ff:ff:ff:ff:ff
I (7664) espnow_example: Receive 7th broadcast data from: 30:ae:a4:0c:34:ec, len: 10
I (8114) connTest: Start connTest example
I (8694) espnow_example: send data to ff:ff:ff:ff:ff:ff
I (9694) espnow_example: send data to ff:ff:ff:ff:ff:ff
I (9694) espnow_example: Receive 9th broadcast data from: 30:ae:a4:0c:34:ec, len: 10
I (9694) espnow_example: Start sending unicast data
I (9694) espnow_example: send data to 30:ae:a4:0c:34:ec
I (10154) connTest: Connected to server
I (10154) connTest: Finished connTest example
I (10754) espnow_example: send data to 30:ae:a4:0c:34:ec
I (11164) connTest: Start connTest example
I (11754) espnow_example: send data to 30:ae:a4:0c:34:ec
I (12824) espnow_example: send data to 30:ae:a4:0c:34:ec
I (13444) connTest: Connected to server
I (13444) connTest: Finished connTest example
I (13934) espnow_example: send data to 30:ae:a4:0c:34:ec
I (14444) connTest: Start connTest example
I (15024) espnow_example: send data to 30:ae:a4:0c:34:ec
I (16044) espnow_example: send data to 30:ae:a4:0c:34:ec
I (16504) connTest: Connected to server
I (16514) connTest: Finished connTest example

zhangyanjiaoesp, 2024-03-05:
@nicolas-Philips 
Can you update your IDF branch and have a try?

Sherry616, 2024-04-24:
Thanks for reporting, will close due to short of feedback, feel free to reopen with more updates.

closed 2024-04-24 by Sherry616