espressif/esp-idf #9776: include "esp_adc/adc_continuous.h" no such file or directory

belongtothenight, 2022-09-13:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1-dev-736-gd6d8324ad9

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

CMD

### What is the expected behavior?

There should be no problem when including "esp_adc/adc_continuous.h".

### What is the actual behavior?

When editing using vscode with "c_cpp_properties.json" correctly setup, all other includes have no error but "esp_adc/adc_continuous.h".
When building with ESP-IDF CMD, it returns "fatal error: esp_adc/adc_continuous.h: No such file or directory".

### Steps to reproduce.

1. Open the entire esp-idf repository with GitHub Desktop.
2. In "esp-idf/examples/peripherals/adc/ folder, duplicate folder "continuous_read" as "continuous_read_test".
3. In ESP-IDF CMD, cd to the same directory, enter "idf.py set-target esp32", enter "idf.py menuconfig" and save, enter "idf.py build"

### Build or installation Logs.

```plain
Setting PYTHONNOUSERSITE, was not set
Using Python in C:\Users\dachu\esp\.espressif\python_env\idf4.4_py3.8_env\Scripts\
Python 3.8.7
Using Git in C:\Users\dachu\esp\.espressif\tools\idf-git\2.34.2\cmd\
git version 2.34.1.windows.1
Setting IDF_PATH: C:\Users\dachu\esp\espidf

Adding ESP-IDF tools to PATH...
    C:\Users\dachu\esp\.espressif\tools\xtensa-esp32-elf\esp-2021r2-patch3-8.4.0\xtensa-esp32-elf\bin
    C:\Users\dachu\esp\.espressif\tools\xtensa-esp32s2-elf\esp-2021r2-patch3-8.4.0\xtensa-esp32s2-elf\bin
    C:\Users\dachu\esp\.espressif\tools\xtensa-esp32s3-elf\esp-2021r2-patch3-8.4.0\xtensa-esp32s3-elf\bin
    C:\Users\dachu\esp\.espressif\tools\riscv32-esp-elf\esp-2021r2-patch3-8.4.0\riscv32-esp-elf\bin
    C:\Users\dachu\esp\.espressif\tools\esp32ulp-elf\2.28.51-esp-20191205\esp32ulp-elf-binutils\bin
    C:\Users\dachu\esp\.espressif\tools\esp32s2ulp-elf\2.28.51-esp-20191205\esp32s2ulp-elf-binutils\bin
    C:\Users\dachu\esp\.espressif\tools\cmake\3.23.1\bin
    C:\Users\dachu\esp\.espressif\tools\openocd-esp32\v0.11.0-esp32-20220411\openocd-esp32\bin
    C:\Users\dachu\esp\.espressif\tools\ninja\1.10.2\
    C:\Users\dachu\esp\.espressif\tools\idf-exe\1.0.3\
    C:\Users\dachu\esp\.espressif\tools\ccache\4.3\ccache-4.3-windows-64
    C:\Users\dachu\esp\.espressif\tools\dfu-util\0.9\dfu-util-0.9-win64
    C:\Users\dachu\esp\espidf\tools

Checking if Python packages are up to date...
Python requirements from C:\Users\dachu\esp\espidf\requirements.txt are satisfied.

Done! You can now compile ESP-IDF projects.
Go to the project directory and run:

  idf.py build

[... 233 lines cut ...]
[88/97] Linking C static library esp-idf\efuse\libefuse.a
[89/97] Linking C static library esp-idf\bootloader_support\libbootloader_support.a
[90/97] Linking C static library esp-idf\spi_flash\libspi_flash.a
[91/97] Linking C static library esp-idf\micro-ecc\libmicro-ecc.a
[92/97] Linking C static library esp-idf\soc\libsoc.a
[93/97] Linking C static library esp-idf\hal\libhal.a
[94/97] Linking C static library esp-idf\main\libmain.a
[95/97] Linking C executable bootloader.elf
[96/97] Generating binary image from built executable
esptool.py v3.3.2-dev
Creating esp32 image...
Merged 1 ELF section
Successfully created esp32 image.
Generated C:/Users/dachu/esp/esp-idf/examples/peripherals/adc/continuous_read_test/build/bootloader/bootloader.bin
[97/97] cmd.exe /C "cd /D C:\Users\dachu\esp\esp-idf\examples\peripherals\adc\continuous_read_test\build\bootloader\esp-idf\esptool_py && C:\Users\dachu\esp\.espressif\python_env\idf4.4_py3.8_env\Scripts\python.exe C:/Users/dachu/esp/espidf/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x1000 C:/Users/dachu/esp/esp-idf/examples/peripherals/adc/continuous_read_test/build/bootloader/bootloader.bin"
Bootloader binary size 0x6330 bytes. 0xcd0 bytes (11%) free.
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

C:\Users\dachu\esp\esp-idf\examples\peripherals\adc\continuous_read_test>
```

### More Information.

I can't find similar issues online, and I've also confirmed that the file "adc_continuous.h" exists, so I have no idea how to solve this.

suda-morris, 2022-09-14:
Hi @belongtothenight  From what print in your CMD `Using Python in C:\Users\dachu\esp\.espressif\python_env\idf4.4_py3.8_env\Scripts\` , I guess you're trying to compile an example that comes from ESP-IDF master branch with your local ESP-IDF v4.4 SDK. Note that, the `esp_adc` component was introduced since esp-idf 5.0, so it doesn't exist in your 4.4 SDK.

belongtothenight, 2022-09-14:
> Hi @belongtothenight  From what print in your CMD `Using Python in C:\Users\dachu\esp\.espressif\python_env\idf4.4_py3.8_env\Scripts\` , I guess you're trying to compile an example that comes from [...]

So what I should do now is to remove esp-idf 4.4 and install 5.0 version of it right? I will test it asap when I'm available, thanks!

belongtothenight, 2022-09-14:
> Hi @belongtothenight 
After I installed 5.0 version the problem is solved, thank you for your help!

closed 2022-09-14 by belongtothenight