espressif/esp-idf #9778: To send wifi and password received from ble as json

Aishwarya4598, 2022-09-13:
### Is your feature request related to a problem?

I am trying to connect to wifi through ble...i am sending json data from ble i.e ssid and psk .i dont know how to pass these variables to wifi

### Describe the solution you'd like.

i want to know how to pass variables recieved as json from ble.
This is my program
#include <stdio.h>
#include<string.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/event_groups.h"
#include "esp_wifi.h"
#include "esp_event.h" //"esp_event_loop.h"
#include "nvs_flash.h"
#include "esp_log.h"
#include "esp_nimble_hci.h"
#include "lwip/err.h"
#include "lwip/sys.h"
#include "nimble/nimble_port.h"
#include "nimble/nimble_port_freertos.h"
#include "host/ble_hs.h"
#include "services/gap/ble_svc_gap.h"
#include "services/gatt/ble_svc_gatt.h"
#include "cJSON.h"
#include "sdkconfig.h"

#define DEVICE_INFO_SERVICE_UUID 0x180A

#define MANUFACTURER_NAME_CHAR 0xFEF4

#define CREDENTIALS_CHAR 0xDEAD

#define BLE_NAME "IoT-BLE"
#define EXAMPLE_ESP_MAXIMUM_RETRY  1
char *TAG = "BLE-CONNECTION";

uint8_t ble_addr_type;

const char *wifi_ssid;
const char *wifi_psk;
static EventGroupHandle_t s_wifi_event_group;

/* The event group allows multiple bits for each event, but we only care about two events:
 * - we are connected to the AP with an IP
 * - we failed to connect after the maximum amount of retries */
#define WIFI_CONNECTED_BIT BIT0
#define WIFI_FAIL_BIT      BIT1

static const char *TAG1 = "wifi station";

static int s_retry_num = 0;

static void event_handler(void* arg, esp_event_base_t event_base,
                                int32_t event_id, void* event_data)
{
    if (event_base == WIFI_EVENT && event_id == WIFI_EVENT_STA_START) {
        esp_wifi_connect();
    } else if (event_base == WIFI_EVENT && event_id == WIFI_EVENT_STA_DISCONNECTED) {
        if (s_retry_num < EXAMPLE_ESP_MAXIMUM_RETRY) {
            esp_wifi_connect();
            s_retry_num++;
            ESP_LOGI(TAG1, "retry to connect to the AP");
        } else {
            xEventGroupSetBits(s_wifi_event_group, WIFI_FAIL_BIT);
        }
        ESP_LOGI(TAG1,"connect to the AP fail");
    } else if (event_base == IP_EVENT && event_id == IP_EVENT_STA_GOT_IP) {
        ip_event_got_ip_t* event = (ip_event_got_ip_t*) event_data;
[... 169 lines cut ...]
    ble_app_advertise();                     //start advertising the services -->
}

void host_task(void *param)
{
    nimble_port_run();
}

//BLE
void app_main()
{
    nvs_flash_init();
    esp_nimble_hci_and_controller_init();                   //initialize bluetooth controller.
    nimble_port_init();                                     //nimble library initialization.
    ESP_ERROR_CHECK(ble_svc_gap_device_name_set(BLE_NAME)); //set BLE name.
    ble_svc_gap_init();                                     //initialize the gap service.
    ble_svc_gatt_init();                                    //initailize the gatt service.
    ble_gatts_count_cfg(gatt_svcs);                         // config all the gatt services that wanted to be used.
    ble_gatts_add_svcs(gatt_svcs);                          // queues all services.
    ble_hs_cfg.sync_cb = ble_app_on_sync;
    nimble_port_freertos_init(host_task);
}
output:
retry to connect ap
failed to connect

nachiketkukade, 2022-09-19:
@Aishwarya4598 , you can use strlcpy to safely copy parameters into wifi_config. Check code used in this example - https://github.com/espressif/esp-idf/blob/master/examples/wifi/iperf/main/cmd_wifi.c#L166

nachiketkukade, 2022-10-04:
Closing due to no response

closed 2022-10-04 by nachiketkukade