espressif/esp-idf #9779: Inccorect readings from i2c slave with master branch, works ok with v4.4.2

FAksel, 2022-09-13:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1-dev-736-gd6d8324ad9

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### Development Kit.

ESP32-S3-WROOM-1-N8 custom board

### Power Supply used.

USB

### What is the expected behavior?

I'm reading a ms5611 pressure sensor via i2c using an esp32-s3 custom board and the i2c.h driver in master mode. The code that works on esp-idf v4.4.2 doesnt work with the master branch. 

Only two commands from the i2c driver are used:

i2c_master_write_to_device(...)  to write a 1 byte command to trigger adc conversion
i2c_master_write_read_device() to read 3 bytes from the device (temperature or pressure)

These are correct data obtained from the sensor:

I (4840) ms5611: bytes 0 to 2: 128, 151, 160
I (4850) ms5611: bytes 0 to 2: 137, 216, 36
I (4850) ms5611: raw P: 8427424, raw T: 9033764
I (4850) ms5611-example: Pressure: 96512 Pa, Temperature: 30.59 C

### What is the actual behavior?

The same code with the master branch results in garbage data

I (3820) ms5611: para 0 to 2: 0, 145, 16
I (3830) ms5611: para 0 to 2: 1, 28, 24
I (3830) ms5611: raw P: 37136, raw T: 72728
I (3830) ms5611-example: Pressure: -18442 Pa, Temperature: 20.21 C

Howver, checking with a logic analyzer, the bytes sent by the slave are correct and the i2c communcation works fine. See here:

[image]

### Steps to reproduce.

**i2c configuration:**

```
    i2c_config_t conf = {
    .mode = I2C_MODE_MASTER,
    .sda_io_num = CONFIG_I2C_MASTER_SDA,         // select GPIO specific to your project
    .sda_pullup_en = GPIO_PULLUP_DISABLE,
    .scl_io_num = CONFIG_I2C_MASTER_SCL,         // select GPIO specific to your project
    .scl_pullup_en = GPIO_PULLUP_DISABLE,
    .master.clk_speed = CONFIG_I2C_MASTER_FREQUENCY,  // select frequency specific to your project
    .clk_flags = 0,                          // you can use I2C_SCLK_SRC_FLAG_* flags to choose i2c source clock here
};
    
    ESP_ERROR_CHECK(i2c_param_config(I2C_NUM_0, &conf));
    ESP_ERROR_CHECK(i2c_driver_install(I2C_NUM_0, conf.mode, 0, 0, 0));
```

**send command and read command**

```
static inline esp_err_t send_command(uint8_t cmd)
{
    // return i2c_dev_write(&dev->i2c_dev, NULL, 0, &cmd, 1);
    return i2c_master_write_to_device(I2C_PORT, CONFIG_MS5611_ADDRESS, &cmd, 1, pdMS_TO_TICKS(5));
}

static inline esp_err_t read_reg(uint8_t reg, uint8_t *readBuf, size_t read_size)
{

    return i2c_master_write_read_device(I2C_PORT, CONFIG_MS5611_ADDRESS, &reg, 1, readBuf, read_size, pdMS_TO_TICKS(5));
}
```

mythbuster5, 2022-09-14:
Any additional information? I tested on my local slave, still works fine on master. Did you add pull-up resistors?

FAksel, 2022-09-14:
Yes, 2x 10k. But again, checking the bus with a logic analyzer shows that the communication works fine and all sent and received data are correct. But somehow the values store in the read buffer are not correct and do not correspond.

There is another evidence of the issue [here](https://stackoverflow.com/questions/72290978/esp-idf-driver-i2c-h-reading-data-from-ccs811-registers-is-not-working-correclty).

Running this code [here ](https://github.com/UncleRus/esp-idf-lib/tree/master/examples/ms5611/default)shows the same behavior. Works with 4.4.2 but not with latest branch.

Any suggestion where to dig? Thank you

FAksel, 2022-09-14:
I've uploaded the complete code to a repo:

[https://github.com/FAksel/ms5611](https://github.com/FAksel/ms5611)

o-marshmallow, 2022-09-14:
Hi @FAksel ,

I just tried on my side with an ESP32-S3 module, and I can write and read properly an I2C device (I/O expander). I took back your functions to perform the test:
[image]

Output from the code:
[image]

(Writing a single byte is not supported on the I2C device I am using)

I tried at both 200KHz and 400KHz, both succeeds. I am using pin 35 for SDA, 36 for SCL, and 2kOhm resistors.

Which frequency have you setup? Which pins are you using? How big are your pull-up resistors between SDA-3.3V and SCL-3.3V?

FAksel, 2022-09-14:
Well, it now works with 2k pull-ups. Sorry for opening the issue, I guess you can close it.
Would still be intersted to know what changed between 4.4.2 and the current master though!

Thank you for the support

o-marshmallow, 2022-09-14:
@FAksel Good to know!
No big change in the I2C was made between v4.4.2 and master from what I aware of. Some commits were made to fix the effective SCL frequency more accurate, but they were also backported to v4.4.x, so I am sure it's related.
@mythbuster5 does anything come to your mind?

In an case, keep in mind that I2C is highly dependent on the pull-up resistor value and your circuit capacitance. 10kOhm is usually a good value for 100KHz and less, but as soon as you go above it, much smaller values are needed.

closed 2022-09-29 by FAksel