espressif/esp-idf #9780: tx power adv

netanbi, 2022-09-14:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi Espressif engineer ,
i work with the example of "gatts_demo" -ble
board: evk esp32-c3-devkitm-1

I succeed in changing the tx power with the func but if I see with NRF app what is the advertisement of the tx power is always 9dbm
even if change the to 18dbm or 0 dBm and with my spectrum analyzer I verify that it's really transforming the power tx but in the app I always see 9dm
thanks

netanbi, 2022-09-18:
Hi, any update or some instructions?

esp-cjh, 2022-09-18:
Hi, @netanbi.

For gatts demo, the tx power value is included in scan response data. The param 'power level` of 'esp_ble_tx_power_set()` is from 0 ~ 15. But in BTM the adv tx power is from 0 ~ 7. So if the power level is greater than 7, it will be set to 7 (corresponds to 9dBm) when config scan response data. The value you set (18dBm -> 14, 0dBm -> 8) is greater than 7.

Additionally, you should call `esp_ble_gap_config_adv_data(&scan_rsp_data)` after setting tx power. Otherwise the tx power included in scan response data will not update.

We will fix it ASAP.

netanbi, 2022-09-18:
can you instruct me to fix the manually in my program?

esp-cjh, 2022-09-19:
OK, this is the patch.
[0001-fix-ble-adv-tx-power-map.zip](https://github.com/espressif/esp-idf/files/9595996/0001-fix-ble-adv-tx-power-map.zip)

netanbi, 2022-09-20:
[fix files.zip](https://github.com/espressif/esp-idf/files/9604454/fix.files.zip)

I fix the files according to your instructions,
first, it's not changed the tx power acording the value the we enter , also your table is strange and not the same like the parameters that need to enter in 
esp_ble_tx_power_set(ESP_BLE_PWR_TYPE_DEFAULT, ESP_PWR_LVL_N6);

typedef enum {
    ESP_PWR_LVL_N27 = 0,              /*!< Corresponding to -27dbm */
    ESP_PWR_LVL_N24 = 1,              /*!< Corresponding to -24dbm */
    ESP_PWR_LVL_N21 = 2,              /*!< Corresponding to -21dbm */
    ESP_PWR_LVL_N18 = 3,              /*!< Corresponding to -18dbm */
    ESP_PWR_LVL_N15 = 4,              /*!< Corresponding to -15dbm */
    ESP_PWR_LVL_N12 = 5,              /*!< Corresponding to -12dbm */
    ESP_PWR_LVL_N9  = 6,              /*!< Corresponding to  -9dbm */
    ESP_PWR_LVL_N6  = 7,              /*!< Corresponding to  -6dbm */
    ESP_PWR_LVL_N3  = 8,              /*!< Corresponding to  -3dbm */
    ESP_PWR_LVL_N0  = 9,              /*!< Corresponding to   0dbm */
    ESP_PWR_LVL_P3  = 10,             /*!< Corresponding to  +3dbm */
    ESP_PWR_LVL_P6  = 11,             /*!< Corresponding to  +6dbm */
    ESP_PWR_LVL_P9  = 12,             /*!< Corresponding to  +9dbm */
    ESP_PWR_LVL_P12 = 13,             /*!< Corresponding to  +12dbm */
    ESP_PWR_LVL_P15 = 14,             /*!< Corresponding to  +15dbm */
    ESP_PWR_LVL_P18 = 15,             /*!< Corresponding to  +18dbm */
    ESP_PWR_LVL_INVALID = 0xFF,         /*!< Indicates an invalid value */

The TX power values should be from -27 dBm to +18 dBm.
The patch has the value range of [-24, +21] dBm.

esp-cjh, 2022-09-20:
Yes, you should changed `BTM_BLE_ADV_TX_POWER` according to the tx power table you used.

The `power_type` of `esp_ble_tx_power_set()` should set to `ESP_BLE_PWR_TYPE_ADV` when advertising.

Do you call `esp_ble_gap_config_adv_data(&scan_rsp_data)` after set tx power?

netanbi, 2022-09-20:
Hi,
by  changing the order of  esp_ble_gap_config_adv_data(&scan_rsp_data) after esp_ble_tx_power_set() its work 
and also you need to change the table to : 
  
#define BTM_BLE_ADV_TX_POWER {-27,-24, -21, -18, -15, -12, -9, -6, -3, 0, 3, 6, 9, 12, 15, 18}
in the patch file, the table was wrong  👍 

it's needs to be in correlation with  

ESP_PWR_LVL_N27 = 0, /*!< Corresponding to -27dbm /
ESP_PWR_LVL_N24 = 1, /!< Corresponding to -24dbm /
ESP_PWR_LVL_N21 = 2, /!< Corresponding to -21dbm /
ESP_PWR_LVL_N18 = 3, /!< Corresponding to -18dbm /
ESP_PWR_LVL_N15 = 4, /!< Corresponding to -15dbm /
ESP_PWR_LVL_N12 = 5, /!< Corresponding to -12dbm /
ESP_PWR_LVL_N9 = 6, /!< Corresponding to -9dbm /
ESP_PWR_LVL_N6 = 7, /!< Corresponding to -6dbm /
ESP_PWR_LVL_N3 = 8, /!< Corresponding to -3dbm /
ESP_PWR_LVL_N0 = 9, /!< Corresponding to 0dbm /
ESP_PWR_LVL_P3 = 10, /!< Corresponding to +3dbm /
ESP_PWR_LVL_P6 = 11, /!< Corresponding to +6dbm /
ESP_PWR_LVL_P9 = 12, /!< Corresponding to +9dbm /
ESP_PWR_LVL_P12 = 13, /!< Corresponding to +12dbm /
ESP_PWR_LVL_P15 = 14, /!< Corresponding to +15dbm /
ESP_PWR_LVL_P18 = 15, /!< Corresponding to +18dbm /
ESP_PWR_LVL_INVALID = 0xFF, /!< Indicates an invalid value */

thanks!

Alvin1Zhang, 2024-02-23:
Feel free to reopen if the issue still happens, thanks.

closed 2024-02-23 by Alvin1Zhang