espressif/esp-idf #9782: assert failed: tcpip_send_msg_wait_sem IDF/components/lwip/lwip/src/api/tcpip.c:455 (Invalid mbox)

DreamPWJ, 2022-09-14:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

4.4

### Operating System used.

Windows

### How did you build your project?

Command line with CMake

### Development Kit.

ESP32-C3

### Power Supply used.

USB

### What is the expected behavior?

NB-IOT   use  esp_https_ota  

### What is the actual behavior?

assert failed: tcpip_send_msg_wait_sem IDF/components/lwip/lwip/src/api/tcpip.c:455 (Invalid mbox)

### Steps to reproduce.

        esp_http_client_config_t config = {
                .url = file_url.c_str(),
                .cert_pem = (char *) server_cert_pem_start,
                .timeout_ms = 600000,
                //.crt_bundle_attach =  esp_crt_bundle_attach,
                .keep_alive_enable = true,
        };
        esp_err_t ret = esp_https_ota(&config);

### Debug Logs.

```plain
assert failed: tcpip_send_msg_wait_sem IDF/components/lwip/lwip/src/api/tcpip.c:455 (Invalid mbox)
Core  0 register dump:
MEPC    : 0x40381bc2  RA      : 0x40384fea  SP      : 0x3fca02a0  GP      : 0x3fc8ba00  
TP      : 0x3fc7eddc  T0      : 0x37363534  T1      : 0x7271706f  T2      : 0x33323130  
S0/FP   : 0x3fca0432  S1      : 0x00000065  A0      : 0x3fca0304  A1      : 0x3fc8bc45  
A2      : 0x00000001  A3      : 0x00000029  A4      : 0x00000001  A5      : 0x3fc90000  
A6      : 0x7a797877  A7      : 0x76757473  S2      : 0x3fca02f8  S3      : 0x00000001  
S4      : 0x3fca02f8  S5      : 0x4201c0cc  S6      : 0x00000000  S7      : 0x00000000  
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000  
T3      : 0x6e6d6c6b  T4      : 0x6a696867  T5      : 0x66656463  T6      : 0x62613938  
MSTATUS : 0x00001801  MTVEC   : 0x40380001  MCAUSE  : 0x00000007  MTVAL   : 0x00000000  
MHARTID : 0x00000000  

Stack memory:
```

DreamPWJ, 2022-09-20:
How to do OTA differential upgrade for esp32

xueyunfei998, 2022-09-26:
hi @DreamPWJ 
Do you have this assert after ota fails? After ota succeeds, you need to call this function `esp_restart`.

DreamPWJ, 2022-09-26:
@xueyunfei998  I did a reboot, I don't think we're talking about a problem, please read the title and content section

xueyunfei998, 2022-10-08:
hi @DreamPWJ 
Can I reproduce this problem locally, or which example in idf do you use to reproduce this bug?

DreamPWJ, 2022-10-09:
I wonder if the esp_https_ota method only works for wifi networks, not for nb-iot networks?  @xueyunfei998

xueyunfei998, 2022-10-10:
@DreamPWJ 
I don't know about NB_IOT network. Do NB_iot devices have ip addresses? Do need tcpip protocol stack?

xueyunfei998, 2022-10-10:
@DreamPWJ 
If NBIOT network has no ip address probability, ota function should not be supported.

DreamPWJ, 2022-10-12:
@xueyunfei998  NB-IOT has ip

xueyunfei998, 2022-10-12:
@DreamPWJ 
Please upload the complete test log.

xueyunfei998, 2022-10-12:
@DreamPWJ 
How is the NBiot device connected with espC3? Can you simply draw a circuit diagram?

DreamPWJ, 2022-10-23:
> @DreamPWJ How is the NBiot device connected with espC3? Can you simply draw a circuit diagram?

uart    AT command

ustccw, 2022-10-24:
@DreamPWJ 你好，请问你是基于 [ESP-AT](https://github.com/espressif/esp-at) 框架开发的，还是自己写的 AT 框架呢?
如果使用的是 [ESP-AT](https://github.com/espressif/esp-at) 框架，那 ESP-IDF 版本应该要用自带的 v4.3 版本，而不是描述里的 v4.4 版本。

另外，可以打开 menuconfig 里的 debug log，再提供下完整的日志，谢谢。
```
python build.py menuconfig -> Component config -> Log output -> Debug
```

closed 2022-12-27 by DreamPWJ

marcosterland, 2023-01-03:
@DreamPWJ 
Have you found a solution? I'm getting the same error message on an ESP32 with SIM800 GSM module.

rossog93, 2023-01-06:
Hi. I'm having the same problem using ESP32 with the ENC28J60 module and InfluxDB client.

I am able to see this warning when compiling:

`warning: "ntohs" redefined`

I don't know really if this warning can be related to the bug discussed here.

Thanks,
Gabriel.

ceren16, 2023-01-08:
> know really if this warning can be related to

> @DreamPWJ Have you found a solution? I'm getting the same error message on an ESP32 with SIM800 GSM module.
hello , ı faced same error how can I solve ? did you solve ?

ceren16, 2023-01-09:
> @DreamPWJ Have you found a solution? I'm getting the same error message on an ESP32 with SIM800 GSM module.

@marcosterland Have you found a solution? I'm getting the same error message on an ESP32 with SIM800 GSM module.

MacielTurchi, 2023-01-31:
> > @DreamPWJ¿Has encontrado una solución? Recibo el mismo mensaje de error en un ESP32 con módulo SIM800 GSM.
> 
> @marcosterland¿Has encontrado una solución? Recibo el mismo mensaje de error en un ESP32 con módulo SIM800 GSM.

Hola, pudiste resolverlo? tengo el mismo inconveniente

superf-lewis, 2023-02-11:
Yep, I'm getting this error when trying to open a socket. I'm using an az-delivery dev kit V4. No complicating NB-IoT or SIMCom module issues.

Short background: Using PlatformIO I have another ESP32 project where httpClient retrieves ~2kb of JSON reliably then crashes after ~630 consecutive requests - every time. Memory leak perhaps? Error messages suggested issue was in WiFiClient.cpp so I want to write my own tcp client to see if I can beat this 630 request limit. Simply creating a socket threw this error:
**assert failed: tcpip_send_msg_wait_sem IDF/components/lwip/lwip/src/api/tcpip.c:455 (Invalid mbox)**

Here's what you need to reproduce the problem:

#include <lwip/sockets.h>

/* Insert this to run only once in loop() */
if ((m_sockfd = socket(AF_INET, SOCK_STREAM, 0)) < 0) {
    return;
}
setsockopt(m_sockfd, SOL_SOCKET, SO_KEEPALIVE, 0, 0);

Code compiles fine in the Expressive 6.0.1 but crashes on the ESP32 when it hits **if ((m_sockfd = socket(AF_INET, SOCK_STREAM, 0)) < 0) {**

I don't try connecting to any tcp server - no point if I can't get past socket creation. Any suggestions to get past this are most welcome.

superf-lewis, 2023-02-11:
Oops. Didn't have the WiFi up. Now the socket opens. Maybe you guys are having issues because your NB-IoT & SIM800 modules have not connected to the network?

ceren16, 2023-02-11:
@superf-lewis  i dont understand , can you solve , if you solve , can you write solution please

superf-lewis, 2023-02-11:
@ceren16 I don't think I solved anything. I made a mistake for my case. If the tcp/ip stack is not initialised, then there is nothing to open a socket on. Do something that opens a socket and, "boom", **assert failed: tcpip_send_msg_wait_sem IDF/components/lwip/lwip/src/api/tcpip.c:455 (Invalid mbox)** I assumed that creating a socket without the stack initialised might cause some grumbles and warnings but not a full on-abort and reboot.

In my case, I had not started WiFi. Starting WiFi initialises the tcp/ip stack. So, if I get the same error as you guys when I do tcp/ip stuff before the stack is initialised, perhaps the same thing is happening to you? If your project has not triggered something that initialises the tcp/ip stack and it tries to do some socket-related operation, then it crashes. Do you start the WiFi before doing anything on the SIM800?

My experience with a SIM800 was on a full Linux OS. The pppd issued AT commands over the serial interface to start a ppp session, then it magically tied the data stream into the tcp/ip stack (I wish I knew how that mechanism worked). The ESP32 is a lot less than a full Linux OS. The SIM800 has it's own tcp/ip stack which you can drive with AT commands (kind of like steering from the back seat). I have not tried this but it's how I assume you'd do it with something like an ESP32. ...Unless there is some driver to connect/integrate the SIM800 data stream to the ESP32 tcp/ip stack that I don't know about (there's plenty I don't know about).

Sorry, this is no solution. I can only suggest you look at what your application is doing and make sure it doesn't do any tcp/ip stuff before the stack has been started. Check and/or rearrange the order you start things in Setup(). Good luck!

CW-B-W, 2023-06-28:
> Oops. Didn't have the WiFi up. Now the socket opens. Maybe you guys are having issues because your NB-IoT & SIM800 modules have not connected to the network?

Thanks! I was suffering from the same issue, and found that I didn't setup the Wi-Fi either.
After Wi-Fi is setup, it works fine.

cbake6807, 2023-06-30:
I had this issue, came here, and it turned out to be that I set too short of a password. Once i changed it from 5 chars to 8 chars, it worked fine. Is there some WPA protocol thing that must be dependent?

Magnatto, 2023-09-12:
Had a similar issue, it was caused by a memory leak, after some time my code didn`t have any heap avaiable

DiracFermi1411, 2023-12-14:
How did you solve it?

OmegaLeWolf, 2024-06-25:
Sorry for the bump, I had a similar issue with my ESP32-Cam and found a solution to the issue. After a bit of research it seemed the problem comes from a lack of a TCPIP stack init. The way I solved this was by initializing the WiFi library and setting the modem mode to station (in my application).

I hope this helps!

MACKADRIS, 2024-07-06:
I had the same issue today, and moved the WiFi init further up in the setup routine to initialize sooner.  Apparently, I was calling something that needed the internet connection, and it was the ntp service calling for the time before wifi was actually connected.