espressif/esp-idf #9784: esp_partition_write (erase), esp_partition_read  serious issue when working from different tasks

ifedotov1984, 2022-09-14:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

esp-idf-v5.0-beta1

### Operating System used.

Windows

### How did you build your project?

Eclipse IDE

### Development Kit.

proprietary

### Power Supply used.

External 3.3V

### What is the expected behavior?

esp_partition_write, esp_partition_erase, esp_partition_read need to work from different tasks simultaneously (thread safety)

### What is the actual behavior?

One of the task randomly freezes on flash erase or flash write.
It is happens only if CONFIG_SPI_FLASH_SHARE_SPI1_BUS=y (additionally setting CONFIG_SPI_FLASH_YIELD_DURING_ERASE=y make  freezes more frequent)

### Steps to reproduce.

1. Start test project (only main flash and no any other devices  (psram, etc) connected to SPI1)
2. if CONFIG_SPI_FLASH_SHARE_SPI1_BUS=y - after few seconds from start, will see only reads from main task, or only writes from second
3. if CONFIG_SPI_FLASH_SHARE_SPI1_BUS is not set - in rotation flash reads and flash writes (works as as expected)

test ode (test project in attachment):

`/* Finding Partitions Example
   This example code is in the Public Domain (or CC0 licensed, at your option.)

   Unless required by applicable law or agreed to in writing, this
   software is distributed on an "AS IS" BASIS, WITHOUT WARRANTIES OR
   CONDITIONS OF ANY KIND, either express or implied.
*/
#include <string.h>
#include <assert.h>
#include "esp_partition.h"
#include "esp_log.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"

static const char *TAG = "example";

const esp_partition_t *partition=NULL;

TaskHandle_t logAddThreadHandle;

static void logAdd_thread(void *arg)
{
    static char write_data[50];
    while(true)
    {
    	ESP_ERROR_CHECK(esp_partition_erase_range(partition, 0, 4096));
        ESP_ERROR_CHECK(esp_partition_write(partition, 0, write_data, sizeof(write_data)));
        ESP_LOGW("esp_partition_write","write");
    	vTaskDelay(1);
    }
}

void app_main(void)
{
    /*
    * This example uses the partition table from ../partitions_example.csv. For reference, its contents are as follows:
    *
    *  nvs,        data, nvs,      0x9000,  0x6000,
    *  phy_init,   data, phy,      0xf000,  0x1000,
    *  factory,    app,  factory,  0x10000, 1M,
    *  storage,    data, ,             , 0x40000,
    */

    // Find the partition map in the partition table
    partition = esp_partition_find_first(ESP_PARTITION_TYPE_DATA, ESP_PARTITION_SUBTYPE_ANY, "storage");
    assert(partition != NULL);

    static char read_data[50];

	if (xTaskCreatePinnedToCore(logAdd_thread, "lodAdd", 3072, NULL, 4, &logAddThreadHandle, 1) != pdPASS)
	{
		/* Failed to launch task */
		ESP_LOGE(TAG,"Failed to launch Log task");
		abort();
	}

	while(1)
	{
		ESP_ERROR_CHECK(esp_partition_read(partition, 0, read_data, sizeof(read_data)));
		ESP_LOGW("esp_partition_read","read");
	}
}
`

Log output when CONFIG_SPI_FLASH_SHARE_SPI1_BUS=y
[image]

when CONFIG_SPI_FLASH_SHARE_SPI1_BUS is not set  (works as expected)
[image]

I insert debugging messages in spi/flash/cache_utils.c
`void IRAM_ATTR spi_flash_disable_interrupts_caches_and_other_cpu(void)
{
    assert(esp_task_stack_is_sane_cache_disabled());

    ESP_DRAM_LOGD("cache_utils","spi_flash_disable_interrupts_caches_and_other_cpu");
    spi_flash_op_lock();
...
`
and got output:
[image]

Looks like cache corrupted. And after that works only one thread.

Seems, than some error in the SPI1 driver (I assume that in the file spi_bus_lock.c)

Test project (a bit modified from examples):
[partition_ops.zip](https://github.com/espressif/esp-idf/files/9564309/partition_ops.zip)

igrr, 2022-09-14:
Looks like this might be related to https://github.com/espressif/esp-idf/issues/7580. This was fixed on master in https://github.com/espressif/esp-idf/commit/aea2fe08162e1ba86ad0189178c1f1cf29238bbd, and was backported to release/v5.0 a few hours ago. The fix on release/v5.0 branch is not on Github yet, should be available there soon.

ifedotov1984, 2022-09-14:
> Looks like this might be related to #7580. This was fixed on master in [aea2fe0](https://github.com/espressif/esp-idf/commit/aea2fe08162e1ba86ad0189178c1f1cf29238bbd), and was backported to release/v5.0 a few hours ago. The fix on release/v5.0 branch is not on Github yet, should be available there soon.

tested. No, patch did'nt helps at all. Issue not in **spi_flash_disable_interrupts_caches_and_other_cpu**, because if **CONFIG_SPI_FLASH_SHARE_SPI1_BUS is not set**, then **spi_flash_disable_interrupts_caches_and_other_cpu** still calls from spi_flash_os_func_app.c 
`static IRAM_ATTR esp_err_t spi1_start(void *arg)
{
#if CONFIG_SPI_FLASH_SHARE_SPI1_BUS
    //use the lock to disable the cache and interrupts before using the SPI bus
    return spi_start(arg);
#else
    //directly disable the cache and interrupts when lock is not used
    **cache_disable**(NULL);
    on_spi1_acquired((spi1_app_func_arg_t*)arg);
    return ESP_OK;
#endif
}`
but **spi_start(arg)** (**spi_bus_lock_acquire_start**) not called in this case. And all wokrs fine.
So, my opinion is that the error is somewhere inside **spi_bus_lock.c**

ifedotov1984, 2022-10-04:
is there any updates from anyone about this issue?

ifedotov1984, 2022-11-16:
> Okay, thanks for checking! I'll mark that this issue is related to SPI driver, then.

A few month passed..
Is there any solution to the bug?

ifedotov1984, 2022-11-24:
Checked at esp-idf v5.0-rc1
[image]

Same issue after 15-30 seconds of work.
test project: 
[partition_ops.zip](https://github.com/espressif/esp-idf/files/10084004/partition_ops.zip)

ginkgm, 2022-11-24:
Hi @ifedotov1984 ,

Your example is very good, it helped me to reproduced the issue effectively! I have also reproduced this on latest release/v4.3 branch. I think the issue has been there since IDF v4.0.

Actually I'm very curious about your use case - I haven't seen this feature being used in real products. May I know your use case? In other words, what device do you connect to the SPI0/1 bus? How do you feel about the performance of the system when accessing the other SPI device?

Cause of the issue: 

The lock is implemented in a way how other drivers are implemented - assume one device will only be used by one user. If the caller violates the rule, they need to add their own lock. This is to avoid unnecessary overhead caused by the lock when the user doesn't need to access the device concurrently (in several tasks/ISR).

In previous flash driver implementation (or when the bus lock is not used), the cache disabling utils happen to ensure the atomicity of flash access. But when the bus lock is used, the lock code get executed first, and it doesn't take concurrent access into consideration. To workaround this issue, if you don't need the bus lock, you may directly disable it. Otherwise, you may wrap all API calls to the main flash with a semaphore.

ifedotov1984, 2022-11-24:
Hello!

 The issue was found while researching esp-idf. I'm new to esp32 and creating my first project on it. When checking the functionality of the controller, i found some issues (these one and for example https://github.com/espressif/esp-idf/issues/9908)
 During tests,  I set the spi bus sharing flag ( I thought that it was needed for psram to work... ) In addition, I want additionally use fram memory to periodically write a small amount of data (about 100bytes, but many many times), and there is no need to use a separate spi port (to save pins) for this. And since the amount of data is small and writing to fram very fast, this will not critically affect to performance