espressif/esp-idf #9785: The size of Queue_t and StaticQueue_t does not match, apparently typedef struct xSTATIC_QUEUE contains an extra pointer.

Clor0xD, 2022-09-14:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

ESP_IDF 4.4.2
The size of Queue_t and StaticQueue_t does not match, apparently typedef struct xSTATIC_QUEUE contains an extra pointer.

"queue.c"

```
typedef struct QueueDefinition /* The old naming convention is used to prevent breaking kernel aware debuggers. */
{

   //there are 2 pointers here
    int8_t * pcHead;           /*< Points to the beginning of the queue storage area. */
    int8_t * pcWriteTo;        /*< Points to the free next place in the storage area. */

    union
    {
        QueuePointers_t xQueue;     /*< Data required exclusively when this structure is used as a queue. */
        SemaphoreData_t xSemaphore; /*< Data required exclusively when this structure is used as a semaphore. */
    } u;

    List_t xTasksWaitingToSend;             /*< List of tasks that are blocked waiting to post onto this queue.  Stored in priority order. */
    List_t xTasksWaitingToReceive;          /*< List of tasks that are blocked waiting to read from this queue.  Stored in priority order. */

    volatile UBaseType_t uxMessagesWaiting; /*< The number of items currently in the queue. */
    UBaseType_t uxLength;                   /*< The length of the queue defined as the number of items it will hold, not the number of bytes. */
    UBaseType_t uxItemSize;                 /*< The size of each items that the queue will hold. */

    volatile int8_t cRxLock;                /*< Stores the number of items received from the queue (removed from the queue) while the queue was locked.  Set to queueUNLOCKED when the queue is not locked. */
    volatile int8_t cTxLock;                /*< Stores the number of items transmitted to the queue (added to the queue) while the queue was locked.  Set to queueUNLOCKED when the queue is not locked. */

    #if ( ( configSUPPORT_STATIC_ALLOCATION == 1 ) && ( configSUPPORT_DYNAMIC_ALLOCATION == 1 ) )
        uint8_t ucStaticallyAllocated; /*< Set to pdTRUE if the memory used by the queue was statically allocated to ensure no attempt is made to free the memory. */
    #endif

    #if ( configUSE_QUEUE_SETS == 1 )
        struct QueueDefinition * pxQueueSetContainer;
    #endif

    #if ( configUSE_TRACE_FACILITY == 1 )
        UBaseType_t uxQueueNumber;
        uint8_t ucQueueType;
    #endif
#ifdef ESP_PLATFORM
    portMUX_TYPE mux;       //Mutex required due to SMP
#endif // ESP_PLATFORM
} xQUEUE;

/* The old xQUEUE name is maintained above then typedefed to the new Queue_t
 * name below to enable the use of older kernel aware debuggers. */
typedef xQUEUE Queue_t;

```

 "FreeRTOS.h"

```
typedef struct xSTATIC_QUEUE
{
   //there are 3 pointers here
    void * pvDummy1[ 3 ];

    union
    {
        void * pvDummy2;
        UBaseType_t uxDummy2;
    } u;

    StaticList_t xDummy3[ 2 ];
    UBaseType_t uxDummy4[ 3 ];
    uint8_t ucDummy5[ 2 ];

    #if ( ( configSUPPORT_STATIC_ALLOCATION == 1 ) && ( configSUPPORT_DYNAMIC_ALLOCATION == 1 ) )
        uint8_t ucDummy6;
    #endif

    #if ( configUSE_QUEUE_SETS == 1 )
        void * pvDummy7;
    #endif

    #if ( configUSE_TRACE_FACILITY == 1 )
        UBaseType_t uxDummy8;
        uint8_t ucDummy9;
    #endif
    portMUX_TYPE xDummy10;
} StaticQueue_t;
typedef StaticQueue_t StaticSemaphore_t;
```

negativekelvin, 2022-09-15:
Why do you want them to be the same size?

Clor0xD, 2022-09-15:
@negativekelvin
So this is the same descriptor in the rtos pipeline. Just for static memory allocation, it had to be moved to the public header and obfuscate. Here is an official comment from freertos with guarantees of size and alignment.

```
/*
 * In line with software engineering best practice, especially when supplying a
 * library that is likely to change in future versions, FreeRTOS implements a
 * strict data hiding policy.  This means the Queue structure used internally by
 * FreeRTOS is not accessible to application code.  However, if the application
 * writer wants to statically allocate the memory required to create a queue
 * then the size of the queue object needs to be know.  The StaticQueue_t
 * structure below is provided for this purpose.  Its sizes and alignment
 * requirements are guaranteed to match those of the genuine structure, no
 * matter which architecture is being used, and no matter how the values in
 * FreeRTOSConfig.h are set.  Its contents are somewhat obfuscated in the hope
 * users will recognise that it would be unwise to make direct use of the
 * structure members.
 */
```

I apologize for my poor knowledge of English

negativekelvin, 2022-09-15:
This exists in upstream freertos so maybe better to raise the issue there?
https://github.com/FreeRTOS/FreeRTOS-Kernel/blob/ff88fc8b6ce9f0c53b06dc46238e08a99ca2c55a/include/FreeRTOS.h#L1322

Clor0xD, 2022-09-15:
The size was broken by modifying the file "queue.c" precisely in the ESP-IDF fork. In the vanilla version, everything just fits together.

negativekelvin, 2022-09-16:
Vanilla version still has 3 pointer vs 2 pointer issue.  IDF added a mux to both structs.

https://github.com/FreeRTOS/FreeRTOS-Kernel/blob/ff88fc8b6ce9f0c53b06dc46238e08a99ca2c55a/queue.c#L102

Clor0xD, 2022-09-20:
@negativekelvin Sorry, I really looked at the wrong version of FreeRTOS when checking.
 FreeRTOS V9.0.0
 
```
 typedef struct QueueDefinition
{
	int8_t *pcHead;					/*< Points to the beginning of the queue storage area. */
	int8_t *pcTail;					/*< Points to the byte at the end of the queue storage area.  Once more byte is allocated than necessary to store the queue items, this is used as a marker. */
	int8_t *pcWriteTo;				/*< Points to the free next place in the storage area. */

	union							/* Use of a union is an exception to the coding standard to ensure two mutually exclusive structure members don't appear simultaneously (wasting RAM). */
	{
		int8_t *pcReadFrom;			/*< Points to the last place that a queued item was read from when the structure is used as a queue. */
		UBaseType_t uxRecursiveCallCount;/*< Maintains a count of the number of times a recursive mutex has been recursively 'taken' when the structure is used as a mutex. */
	} u;

	List_t xTasksWaitingToSend;		/*< List of tasks that are blocked waiting to post onto this queue.  Stored in priority order. */
	List_t xTasksWaitingToReceive;	/*< List of tasks that are blocked waiting to read from this queue.  Stored in priority order. */

	volatile UBaseType_t uxMessagesWaiting;/*< The number of items currently in the queue. */
	UBaseType_t uxLength;			/*< The length of the queue defined as the number of items it will hold, not the number of bytes. */
	UBaseType_t uxItemSize;			/*< The size of each items that the queue will hold. */

	volatile int8_t cRxLock;		/*< Stores the number of items received from the queue (removed from the queue) while the queue was locked.  Set to queueUNLOCKED when the queue is not locked. */
	volatile int8_t cTxLock;		/*< Stores the number of items transmitted to the queue (added to the queue) while the queue was locked.  Set to queueUNLOCKED when the queue is not locked. */

	#if( ( configSUPPORT_STATIC_ALLOCATION == 1 ) && ( configSUPPORT_DYNAMIC_ALLOCATION == 1 ) )
		uint8_t ucStaticallyAllocated;	/*< Set to pdTRUE if the memory used by the queue was statically allocated to ensure no attempt is made to free the memory. */
	#endif

	#if ( configUSE_QUEUE_SETS == 1 )
		struct QueueDefinition *pxQueueSetContainer;
	#endif

	#if ( configUSE_TRACE_FACILITY == 1 )
		UBaseType_t uxQueueNumber;
		uint8_t ucQueueType;
	#endif

} xQUEUE;
```
In version 9 there was int8_t *pcTail;

Alvin1Zhang, 2023-01-12:
Thanks for reporting and sharing the updates, fix on release/4.4 is available at https://github.com/espressif/esp-idf/commit/06cad10ec0375e9c993a19d61ce2bdacfb053bdc. Feel free to reopen.

closed 2023-01-12 by Alvin1Zhang

xentec, 2023-03-09:
Backporting this to v5.0 broke the following code:

```c
static struct {
	TaskHandle_t task;
	SemaphoreHandle_t sem;
	StaticSemaphore_t _sem;
	bool enabled;

	const char *something;
} state;

void init(const char* something) {
	state.something = something;
	state.sem = xSemaphoreCreateMutexStatic(&state._sem);
	// state.something is overwritten here
}
```

After `init(...)` is done, `state.something` becomes something else instead of the passed argument.

xentec, 2023-03-10:
Disregard that comment as the issue only happens with FreeRTOS kernel from Amazon.
Sorry for false alarm.