espressif/esp-idf #9786: adc_info command not working on Windows 10

krupis, 2022-09-14:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello, I use vscode esp-idf on Windows 10. I am trying to execute following command to get ESP32 adc calibration information 

```
espefuse.py --port  COM36 adc_info
```

When I execute this command, python editor (Thonny) opens up:
[image]

I tried to execute different command:
[image]

[image]

[image]

I cannot wrap my head around how can I get this command to work. I would appreciate any suggestions

dobairoland, 2022-09-15:
`espefuse.py` should be available in a properly set-up ESP-IDF Terminal environment (Windows Command Line, Power Shell). I'll ask a Vscode expert to take a look at this.

brianignacio5, 2022-09-15:
Hi @krupis how are you executing the `espefuse.py` script in VSCode ? Are you using the `ESP-IDF: Open ESP-IDF Terminal` command ? This command will load these scripts in PATH. Did you configure your extension  properly?

You should open GitHub Issue in the [vscode esp-idf extension repository](https://github.com/espressif/vscode-esp-idf-extension/issues) instead of here and provide any required troubleshooting information.

krupis, 2022-09-16:
Hello @brianignacio5 
Yes I use ```ESP-IDF: Open ESP-IDF Terminal```
I think I have all the paths setup correctly (Python path, IDF_PATH path, as well as tools path.

I can execute commands such as:
[image]

I think I dont understand properly how to call python scripts from Windows machine.

As I mentioned before, when I execute the command:
```
espefuse.py --port  COM36 adc_info
```
 The espefuse.py script opens up in Python editor that  use (Thonny). I think I am missing something in my command to actually run it

brianignacio5, 2022-09-19:
I think the issue here is because we don't provide DOSKEY in the extension yet for autocomplete.

You need to specify full path of each script and executable `C:\path\to\python.exe C:\path\to\espefuse.py --port COM36 adc_info`

krupis, 2023-11-21:
@brianignacio5 
Hello again. We have discussed about this issue a while ago. I actually forgot about it and created a new ticket by mistake:
https://github.com/espressif/esp-idf/issues/12641

It is really frustrating to specify full paths everytime. It makes it especially annoying when our during mass production our device needs to use some of the idf.py command line tools for testing such as measuring adc_info, merging bin files and etc...

Are there any plans to simplify this and get it to work without the full paths?

krupis, 2023-11-28:
Any updates regarding this?

krupis, 2023-12-11:
@brianignacio5 
Any updates?

brianignacio5, 2024-12-06:
Sorry for late late reply. Fixed in https://github.com/espressif/vscode-esp-idf-extension/pull/1348 for temporal fix in v1.9.0 release but will be using a new fix with EIM integration in next major release very soon.