espressif/esp-idf #9789: NimBLE: Build fails with only Observer role enabled

joaochainho, 2022-09-14:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v.4.4.2

### Operating System used.

macOS

### How did you build your project?

VS Code IDE

### What is the expected behavior?

Build should be successful with no errors.

### What is the actual behavior?

Build returns errors and doesn't complete.

### Steps to reproduce.

1. Create a sample project or test on an existing project
2. Enable BT and select NimBLE
3. Disable al roles but **Observer** (menuconfig)
4. Build the project

### Build or installation Logs.

```plain
...
/Users/raver/espressif/esp-idf-v4.4.2/components/bt/host/nimble/nimble/nimble/host/src/ble_gap.c: In function 'ble_gap_disc_report':
/Users/raver/espressif/esp-idf-v4.4.2/components/bt/host/nimble/nimble/nimble/host/src/ble_gap.c:1007:5: error: implicit declaration of function 'ble_gap_master_extract_state'; did you mean 'ble_gap_slave_extract_cb'? [-Werror=implicit-function-declaration]
     ble_gap_master_extract_state(&state, 0);
     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
     ble_gap_slave_extract_cb
/Users/raver/espressif/esp-idf-v4.4.2/components/bt/host/nimble/nimble/nimble/host/src/ble_gap.c:1008:9: error: implicit declaration of function 'ble_gap_has_client'; did you mean 'ble_gap_mtu_event'? [-Werror=implicit-function-declaration]
     if (ble_gap_has_client(&state)) {
         ^~~~~~~~~~~~~~~~~~
         ble_gap_mtu_event
/Users/raver/espressif/esp-idf-v4.4.2/components/bt/host/nimble/nimble/nimble/host/src/ble_gap.c: In function 'ble_gap_reset_state':
/Users/raver/espressif/esp-idf-v4.4.2/components/bt/host/nimble/nimble/nimble/host/src/ble_gap.c:1424:5: error: implicit declaration of function 'ble_gap_master_failed'; did you mean 'ble_gap_adv_finished'? [-Werror=implicit-function-declaration]
     ble_gap_master_failed(reason);
     ^~~~~~~~~~~~~~~~~~~~~
     ble_gap_adv_finished
/Users/raver/espressif/esp-idf-v4.4.2/components/bt/host/nimble/nimble/nimble/host/src/ble_gap.c: In function 'ble_gap_disc_cancel_no_lock':
/Users/raver/espressif/esp-idf-v4.4.2/components/bt/host/nimble/nimble/nimble/host/src/ble_gap.c:4420:5: error: implicit declaration of function 'ble_gap_master_reset_state'; did you mean 'ble_gap_slave_reset_state'? [-Werror=implicit-function-declaration]
     ble_gap_master_reset_state();
     ^~~~~~~~~~~~~~~~~~~~~~~~~~
     ble_gap_slave_reset_state
/Users/raver/espressif/esp-idf-v4.4.2/components/bt/host/nimble/nimble/nimble/host/src/ble_gap.c: In function 'ble_gap_disc':
/Users/raver/espressif/esp-idf-v4.4.2/components/bt/host/nimble/nimble/nimble/host/src/ble_gap.c:4748:9: error: implicit declaration of function 'ble_gap_master_set_timer'; did you mean 'ble_gap_master_timer'? [-Werror=implicit-function-declaration]
         ble_gap_master_set_timer(duration_ticks);
         ^~~~~~~~~~~~~~~~~~~~~~~~
         ble_gap_master_timer
At top level:
/Users/raver/espressif/esp-idf-v4.4.2/components/bt/host/nimble/nimble/nimble/host/src/ble_gap.c:835:1: warning: 'ble_gap_adv_finished' defined but not used [-Wunused-function]
 ble_gap_adv_finished(uint8_t instance, int reason, uint16_t conn_handle,
 ^~~~~~~~~~~~~~~~~~~~
cc1: some warnings being treated as errors
[1073/1128] Building CXX object esp-idf/asio/CMakeFiles/__idf_asio.dir/asio/asio/src/asio.cpp.obj
ninja: build stopped: subcommand failed.
```

### More Information.

Build fails with other IDF versions such as v4.3.4.
No build errors with v4.2.4 or v5.0-beta1.

joaochainho, 2022-09-14:
###  IDF and NimBLE versions:
esp-idf-v4.2.4: 16 July 2019  - Apache NimBLE v1.2.0
esp-idf-v4.3.4: 18 March 2020 - Apache NimBLE v1.3.0
esp-idf-v4.4.2: 18 March 2020 - Apache NimBLE v1.3.0
esp-idf-v5.0-beta1: 24 March 2021 - Apache NimBLE v1.4.0

rahult-github, 2022-09-15:
Hi @joaochainho  ,

ESP-IDF nimble code is forked clone of upstream apache mynewt-nimble codebase, and we try to maintain the code as equal as possible to uptream nimble code as it makes it easy for maintenance purpose and also end users to switch to ESP-IDF in case their solution already uses Nimble.

The issue you mentioned has been fixed upstream via https://github.com/apache/mynewt-nimble/pull/913 .  However, if you observe, the change was not pushed to nimble-1.3 branch . It was pushed on master branch and then nimble-1.4 got created. Hence, you observe the failure on nimble-1.3 branch and not on nimble-1.4  .  nimble-1.2 is fairly old branch which did not have code under stringent flag checks and hence it compiles.

One suggestion is to cherry-pick those commits and check if it helps resolve your issue.  

Thanks,
 Rahul

AxelLin, 2022-09-15:
> One suggestion is to cherry-pick those commits and check if it helps resolve your issue.

@rahult-github 
I think this should be fixed by ESP-IDF nimble code if esp-idf 4.3/4.4 branches are still in maintain period.

joaochainho, 2022-10-04:
Hi @rahult-github , many thanks for providing the patches!  Build no longer fails with only **Broadcaster** and **Observer** modes enabled.
I've been testing the patches, and I found an issue with scanning. It hangs right at the beginning. No issues if I enable all four modes.

```
D (2461) NimBLEAdvertising: << Advertising start
D (2481) NimBLEAdvertising: >> stop
I (2481) NimBLE: GAP procedure initiated: stop advertising.
D (2481) NimBLEAdvertising: << stop
D (2501) NimBLEAdvertising: >> Advertising start: customAdvData: 1, customScanResponseData: 0
I (2501) NimBLE: GAP procedure initiated: advertise; 
I (2501) NimBLE: disc_mode=2
I (2501) NimBLE:  adv_channel_map=0 own_addr_type=0 adv_filter_policy=0 adv_itvl_min=0 adv_itvl_max=0
I (2511) NimBLE: 
D (2521) NimBLEAdvertising: << Advertising start
D (2541) NimBLEAdvertising: >> stop
I (2541) NimBLE: GAP procedure initiated: stop advertising.
D (2541) NimBLEAdvertising: << stop
I (2561) Main: ------- Starting scan -------
D (2561) NimBLEScan: >> start: duration=1
I (2561) NimBLE: GAP procedure initiated: discovery; 
I (2561) NimBLE: own_addr_type=0 filter_policy=0 passive=0 limited=0 filter_duplicates=1 
I (2571) NimBLE: duration=1000ms
I (2571) NimBLE: 
D (2581) NimBLEScan: << start()
```
Is there anything I can try to troubleshoot it?
Thanks!

rahult-github, 2022-11-29:
Hi @joaochainho ,

Sorry for delayed reply. I used the existing blecent application, applied the above patch and modified the example to not go for GATT ( since in observer / broadcaster role , GAP / GATT connections cannot be made). I do see devices being found , which is expected.  Can you please share your sdkconfig , so i can try the same settings as yours ?

joaochainho, 2022-11-29:
Hi @rahult-github , no worries! 👍 Pls find below my sdkconfig. Running the latest esp-idf v4.4.3.
[sdkconfig.txt](https://github.com/espressif/esp-idf/files/10113032/sdkconfig.txt)

rahult-github, 2022-11-29:
Hi @joaochainho , 

Thanks. Took your sdkconfig and ran the example at my end. I do see discovery works with device being found. Debug prints as below: 

`<pre><font color="#4E9A06">I (2009) NimBLE: GAP procedure initiated: discovery; </font>
<font color="#4E9A06">I (2019) NimBLE: own_addr_type=0 filter_policy=0 passive=1 limited=0 filter_duplicates=1 </font>
<font color="#4E9A06">I (2029) NimBLE: duration=forever</font>
<font color="#4E9A06">I (2029) NimBLE: </font>

D (2029) NimBLE: ble_hs_hci_cmd_send: ogf=0x08 ocf=0x000b len=7

D (2039) NimBLE: 0x0b 
D (2039) NimBLE: 0x20 
D (2039) NimBLE: 0x07 
D (2049) NimBLE: 0x00 
D (2049) NimBLE: 0x30 
D (2049) NimBLE: 0x00 
D (2049) NimBLE: 0x30 
D (2059) NimBLE: 0x00 
D (2059) NimBLE: 0x00 
D (2059) NimBLE: 0x00 
D (2059) NimBLE: 

D (2069) NimBLE: ble_hs_hci_cmd_send: ogf=0x08 ocf=0x000c len=2

D (2069) NimBLE: 0x0c 
D (2069) NimBLE: 0x20 
D (2079) NimBLE: 0x02 
D (2079) NimBLE: 0x01 
D (2079) NimBLE: 0x01 
D (2079) NimBLE: 

D (2119) NimBLE:     mfg_data=
D (2119) NimBLE: 0x4c
D (2119) NimBLE: :0x00
D (2119) NimBLE: :0x12
D (2119) NimBLE: :0x02
D (2119) NimBLE: :0x35
D (2119) NimBLE: :0x01
D (2129) NimBLE: 

D (2149) NimBLE:     flags=0x02

D (2269) NimBLE:     flags=0x02

D (2869) NimBLE:     flags=0x06

D (2869) NimBLE:     mfg_data=
D (2869) NimBLE: 0x57
</pre>`

So i think, functionality is ok, but let's  debug your case.

1. In your case, can you enable nimble debug logs ? 
Component Config  --> Log Output --> Default Log Verbosity --> Debug
Component Config --> Bluetooth --> NimBLE Options --> NimBLE Host log verbosity --> Debug Logs 

 When advertising is started,  debug prints can confirm that indeed devices are present in the vicinity.  BLE_GAP_EVENT_DISC event would be posted when devices are found. If you are handling this event inside your application, i suggest may be putting some print in this event can help. 

2. Assuming you have prints enabled, and still don't observe anything, can you run the existing blecent application in IDF v.4.4 release and confirm once ?  This way, we can be sure that indeed observation is same.  You can attached patch which i created to disable GAP / GATT connection activity temporarily in the blecent example. 

[example_blecent_changes.txt](https://github.com/espressif/esp-idf/files/10113696/example_blecent_changes.txt)

joaochainho, 2022-11-30:
Hi @rahult-github , I believe blecent requires the **central** role. I'm trying to compile just with **observer** and/or **broadcaster** roles. Still getting the same build errors as before.

```
# CONFIG_BT_NIMBLE_ROLE_CENTRAL is not set
# CONFIG_BT_NIMBLE_ROLE_PERIPHERAL is not set
CONFIG_BT_NIMBLE_ROLE_BROADCASTER=y
CONFIG_BT_NIMBLE_ROLE_OBSERVER=y
# CONFIG_NIMBLE_ROLE_CENTRAL is not set
# CONFIG_NIMBLE_ROLE_PERIPHERAL is not set
CONFIG_NIMBLE_ROLE_BROADCASTER=y
CONFIG_NIMBLE_ROLE_OBSERVER=y
```

rahult-github, 2022-12-01:
Hi @joaochainho 

>  I believe blecent requires the central role. 

Yes. correct. But to use only the observer / broadcaster role, the application needs some modification ( to not go for central role related activities). Hence the patch i gave [here](https://github.com/espressif/esp-idf/issues/9789#issuecomment-1330698798) to overcome this . 

>  I'm trying to compile just with observer and/or broadcaster roles. Still getting the same build errors as before.

Per [this](https://github.com/espressif/esp-idf/issues/9789#issuecomment-1266997771) comment, the previous change to compile for observer / broadcaster role was working for you. May i know, if the [first](https://github.com/espressif/esp-idf/issues/9789#issuecomment-1266745278) patch too is included at your end , when you try to compile blecent application ?

rahult-github, 2023-02-03:
Hi @joaochainho , 

May i ask if you have got this working at your end with the patch provided ?

joaochainho, 2023-02-03:
Hi @rahult-github, apologies for the lack of feedback.
The previous change didn't completely fix the issue. Scanning hangs here, if only Broadcaster and Observer roles enabled.

```
...
I (2561) Main: ------- Starting scan -------
D (2561) NimBLEScan: >> start: duration=1
I (2561) NimBLE: GAP procedure initiated: discovery; 
I (2561) NimBLE: own_addr_type=0 filter_policy=0 passive=0 limited=0 filter_duplicates=1 
I (2571) NimBLE: duration=1000ms
I (2571) NimBLE: 
D (2581) NimBLEScan: << start()
I (3382) NimBLEScan: New advertiser: xx:xx:xx:xx:xx:xx
D (3382) NimBLEScan: erase device: xx:xx:xx:xx:xx:xx
D (3662) NimBLE: ble_hs_hci_cmd_send: ogf=0x08 ocf=0x000c len=2
D (3662) NimBLE: 0x0c 
D (3662) NimBLE: 0x20 
D (3662) NimBLE: 0x02 
D (3662) NimBLE: 0x00 
D (3662) NimBLE: 0x00 
D (3662) NimBLE: 
```

On the latest patch to the blecent example, it compiles and runs, but it hangs here. The unpatched blecent example works according to the documentation.
```
...
I (900) NimBLE_BLE_CENT: BLE Host Task Started
I (940) NimBLE: GAP procedure initiated: discovery; 
I (940) NimBLE: own_addr_type=0 filter_policy=0 passive=1 limited=0 filter_duplicates=1 
I (940) NimBLE: duration=forever
I (940) NimBLE: 
```

rahult-github, 2023-02-04:
Hi @joaochainho ,

 There looks to be difference in parameters passed from application while initiating discovery. 
 Non - working case:  
  
  I (2571) NimBLE: duration=1000ms

Working case:
  
 I (940) NimBLE: duration=forever 

 Please check if your use case can increase the discovery time from 1000ms to a larger value . 

Thanks,
 Rahul

Alvin1Zhang, 2023-02-16:
@joaochainho Thanks for reporting, would you please help share if any further updates for the issue? Thanks.

rahult-github, 2023-04-25:
Closing this issue. Feel free to reopen in case of further information

closed 2023-04-25 by rahult-github