espressif/esp-idf #9790: MQTT: Connection refused error: 0x3(Connection refused, server unavailable)

ABHITHLALC, 2022-09-15:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Operating System used. - Windows
Development Kit   -ESP32-WROOM-32D -16M flash

Hi,
I am currently working in ESP-IDF version v4.4.1. I need to change it into stable version v4.4.2. But changing same project form v4.4.1 to v4.4.2 I am facingMQTT error. As:
```
W (43457) MQTT_CLIENT: Connection refused, server unavailable
E (43457) MQTT: MQTT_EVENT_ERROR
E (43457) FATAL: FATAL loop = 2
E (43457) MQTT: Connection refused error: 0x3
E (43467) MQTT_CLIENT: MQTT connect failed
E (43467) MQTT: MQTT_EVENT_DISCONNECTED
```
Can anyone help me with this? What changes should I make during transition form one version to another?
[image]

ABHITHLALC, 2022-09-15:
**Detailed log**

```
I (22791) UART: Initialising UART at 9600 baudrate
I (22801) MQTT: MQTT initialising
I (22801) MQTT: Other event id:7
I (22801) http: HTTP_log
I (22801) RTC: DS-3231 got time
I (22811) RTC: Timestamp--->2022-09-15T06:46:11Z
E (22801) SD_CARD: No sd card detected..!!!!
I (23371) mbedtls: ssl_tls.c:5904 => handshake

I (23371) mbedtls: ssl_cli.c:4483 client state: 0

I (23371) mbedtls: ssl_msg.c:2102 => flush output

I (23371) mbedtls: ssl_msg.c:2114 <= flush output

I (23381) mbedtls: ssl_cli.c:4483 client state: 1

I (23381) mbedtls: ssl_msg.c:2102 => flush output

I (23391) mbedtls: ssl_msg.c:2114 <= flush output

I (23391) mbedtls: ssl_cli.c:999 => write client hello

I (23401) mbedtls: ssl_msg.c:2542 => write handshake message

I (23411) mbedtls: ssl_msg.c:2701 => write record

I (23411) mbedtls: ssl_msg.c:2102 => flush output

I (23421) mbedtls: ssl_msg.c:2122 message length: 267, out_left: 267

I (23431) mbedtls: ssl_msg.c:2127 ssl->f_send() returned 267 (-0xfffffef5)

I (23431) mbedtls: ssl_msg.c:2155 <= flush output

I (23441) mbedtls: ssl_msg.c:2870 <= write record

I (23441) mbedtls: ssl_msg.c:2678 <= write handshake message

I (23451) mbedtls: ssl_cli.c:1467 <= write client hello

I (23461) mbedtls: ssl_msg.c:1886 => fetch input

I (23461) mbedtls: ssl_msg.c:2043 in_left: 0, nb_want: 5

I (23671) mbedtls: ssl_msg.c:2068 in_left: 0, nb_want: 5

I (23671) mbedtls: ssl_msg.c:2069 ssl->f_recv(_timeout)() returned 5 (-0xfffffffb)

I (23681) mbedtls: ssl_msg.c:2089 <= fetch input

I (23681) mbedtls: ssl_cli.c:4483 client state: 2

I (23691) mbedtls: ssl_msg.c:2102 => flush output

I (23691) mbedtls: ssl_msg.c:2114 <= flush output

I (23701) mbedtls: ssl_cli.c:2082 => parse server hello

I (23701) mbedtls: ssl_msg.c:3941 => read record

I (23711) mbedtls: ssl_msg.c:1886 => fetch input

I (23711) mbedtls: ssl_msg.c:2043 in_left: 5, nb_want: 5

I (23721) mbedtls: ssl_msg.c:2089 <= fetch input

[... 361 lines cut ...]

I (26001) mbedtls: ssl_msg.c:1144 => decrypt buf

I (26011) mbedtls: ssl_msg.c:1741 <= decrypt buf

I (26021) mbedtls: ssl_msg.c:4015 <= read record

I (26021) mbedtls: ssl_msg.c:5693 <= read

I (26031) mbedtls: ssl_msg.c:5399 => read

I (26031) mbedtls: ssl_msg.c:5693 <= read

I (26031) mbedtls: ssl_msg.c:5399 => read

I (26041) mbedtls: ssl_msg.c:5693 <= read

W (26041) MQTT_CLIENT: Connection refused, server unavailable
E (26051) MQTT: MQTT_EVENT_ERROR
E (26051) FATAL: FATAL loop = 1
E (26061) MQTT: Connection refused error: 0x3
E (26061) MQTT_CLIENT: MQTT connect failed
E (26071) MQTT: MQTT_EVENT_DISCONNECTED
E (26071) MQTT: Disconnected from ONESENSE
```

ESP-YJM, 2022-09-21:
@ABHITHLALC It seems the MQTT broker refuse the connection with CONNACK, which response code is `0x03 Connection Refused, Server unavailable`. http://docs.oasis-open.org/mqtt/mqtt/v3.1.1/os/mqtt-v3.1.1-os.html#_Toc398718033

ABHITHLALC, 2022-09-24:
Thnak you for the replay.
I will check this. Now  Iam move back to old version v4..4.1

AxelLin, 2022-11-21:
> @ABHITHLALC It seems the MQTT broker refuse the connection with CONNACK, which response code is `0x03 Connection Refused, Server unavailable`. http://docs.oasis-open.org/mqtt/mqtt/v3.1.1/os/mqtt-v3.1.1-os.html#_Toc398718033

I don't get your point.
That does not explain why it works with v4.4.1 but fails with v4.4.2.

ESP-YJM, 2022-11-21:
@AxelLin Yes, looking from the logs, it is true that the server rejected the connection. Need to check whether it is casued by IDF update or user code logic.

closed 2022-12-07 by Alvin1Zhang