espressif/esp-idf #9792: ESP32C2,Use the bluefi interface, enable nimble, and use esp_bt_controller_disable() to turn off bt, a halt occurs

lyh20190329, 2022-09-15:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0-beta1-206-gccdeb43cc7-dirt

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### Development Kit.

esp32c2

### Power Supply used.

External 3.3V

### What is the expected behavior?

Bluetooth is off, freeing up memory

### What is the actual behavior?

esp32c2 crashes or restarts 

### Steps to reproduce.

1、start BT like blow:
  ret = esp_bt_controller_init(&bt_cfg);
    if (ret) {
        ESP_LOGE(TAG,"%s initialize bt controller failed: %s\n", __func__, esp_err_to_name(ret));
    }

    ret = esp_bt_controller_enable(ESP_BT_MODE_BLE);
    if (ret) {
        ESP_LOGE(TAG,"%s enable bt controller failed: %s\n", __func__, esp_err_to_name(ret));
        return;
    }

    ret = esp_blufi_host_and_cb_init(&blufi_callbacks);
    if (ret) {
        ESP_LOGE(TAG,"%s initialise failed: %s\n", __func__, esp_err_to_name(ret));
        return;
    }

2. when BT connect and recive data ,disable as:
 ESP_ERROR_CHECK(esp_bt_controller_disable());   
    esp_bt_controller_deinit();      
   ESP_ERROR_CHECK(esp_bt_controller_mem_release(ESP_BT_MODE_CLASSIC_BT));
........
........

it is run ok in bluedoid but faild in nimblue,please help me,thanks!

### Debug Logs.

```plain
(104954) BT: Recv Custom Data len63

W (104994) BLE_INIT: esp_bt_controller_mem_release not implemented, return OK
I (105004) app_wifi: set_wifi_info_ble:rx_task mini heap=55496,totalheap=94980
Guru Meditation Error: Core  0 panic'ed (Load access fault). Exception was unhandled.

Core  0 register dump:
Stack dump detected
MEPC    : 0x40028246  RA      : 0x4202a40e  SP      : 0x3fcce190  GP      : 0x3fcad640  
0x4202a40e: ble_hs_mbuf_gen_pkt at /home/ubuntu/work/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_hs_mbuf.c:33

TP      : 0x3fca3fec  T0      : 0x4202a400  T1      : 0x4038a3f8  T2      : 0x7373227b  
0x4202a400: ble_hs_mbuf_gen_pkt at /home/ubuntu/work/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_hs_mbuf.c:28

0x4038a3f8: gpspi_flash_ll_set_read_mode at /home/ubuntu/work/esp-idf/components/hal/esp32c2/include/hal/gpspi_flash_ll.h:214
 (inlined by) spi_flash_hal_gpspi_configure_host_io_mode at /home/ubuntu/work/esp-idf/components/hal/spi_flash_hal_common.inc:128

S0/FP   : 0x00001000  S1      : 0x3fcdf250  A0      : 0x00000008  A1      : 0x00000000  
A2      : 0x0000000c  A3      : 0x00000001  A4      : 0x00000012  A5      : 0x00001000  
A6      : 0x00000018  A7      : 0x55514522  S2      : 0x3fce0000  S3      : 0x0000004d  
S4      : 0x3fcdf20c  S5      : 0x00000000  S6      : 0x00000000  S7      : 0x00000000  
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000  
T3      : 0x70222c22  T4      : 0x6946696e  T5      : 0x555f5345  T6      : 0x55514522  
MSTATUS : 0x00001881  MTVEC   : 0x40380001  MCAUSE  : 0x00000005  MTVAL   : 0x00000944  
0x40380001: _vector_table at ??:?

MHARTID : 0x00000000  

Backtrace:

0x40028246 in ?? ()
#0  0x40028246 in ?? ()
#1  0x4202a40e in ble_hs_mbuf_gen_pkt (leading_space=<optimized out>) at /home/ubuntu/work/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_hs_mbuf.c:32
#2  0x00000000 in ?? ()
Backtrace stopped: frame did not save the PC
ELF file SHA256: ac8e35383bf75820

CPU halted.
```

lyh20190329, 2022-09-15:
my config like this:

#
# Automatically generated file. DO NOT EDIT.
# Espressif IoT Development Framework (ESP-IDF) Project Configuration
#
CONFIG_SOC_ADC_SUPPORTED=y
CONFIG_SOC_DEDICATED_GPIO_SUPPORTED=y
CONFIG_SOC_GDMA_SUPPORTED=y
CONFIG_SOC_BT_SUPPORTED=y
CONFIG_SOC_ESP_NIMBLE_CONTROLLER=y
CONFIG_SOC_WIFI_SUPPORTED=y
CONFIG_SOC_ASYNC_MEMCPY_SUPPORTED=y
CONFIG_SOC_SUPPORTS_SECURE_DL_MODE=y
CONFIG_SOC_EFUSE_CONSISTS_OF_ONE_KEY_BLOCK=y
CONFIG_SOC_TEMP_SENSOR_SUPPORTED=y
CONFIG_SOC_SHA_SUPPORTED=y
CONFIG_SOC_ECC_SUPPORTED=y
CONFIG_SOC_FLASH_ENC_SUPPORTED=y
CONFIG_SOC_SECURE_BOOT_SUPPORTED=y
CONFIG_SOC_SYSTIMER_SUPPORTED=y
CONFIG_SOC_XTAL_SUPPORT_26M=y
CONFIG_SOC_XTAL_SUPPORT_40M=y
CONFIG_SOC_ADC_DIG_CTRL_SUPPORTED=y
CONFIG_SOC_ADC_FILTER_SUPPORTED=y
CONFIG_SOC_ADC_MONITOR_SUPPORTED=y
CONFIG_SOC_ADC_PERIPH_NUM=1
CONFIG_SOC_ADC_MAX_CHANNEL_NUM=5
CONFIG_SOC_ADC_ATTEN_NUM=4
CONFIG_SOC_ADC_DIGI_CONTROLLER_NUM=1
CONFIG_SOC_ADC_PATT_LEN_MAX=8
CONFIG_SOC_ADC_DIGI_MIN_BITWIDTH=12
CONFIG_SOC_ADC_DIGI_MAX_BITWIDTH=12
CONFIG_SOC_ADC_DIGI_FILTER_NUM=2
CONFIG_SOC_ADC_DIGI_MONITOR_NUM=2
CONFIG_SOC_ADC_SAMPLE_FREQ_THRES_HIGH=83333
CONFIG_SOC_ADC_SAMPLE_FREQ_THRES_LOW=611
CONFIG_SOC_ADC_RTC_MIN_BITWIDTH=12
CONFIG_SOC_ADC_RTC_MAX_BITWIDTH=12
CONFIG_SOC_RTC_SLOW_CLOCK_SUPPORT_8MD256=y
CONFIG_SOC_BROWNOUT_RESET_SUPPORTED=y
CONFIG_SOC_SHARED_IDCACHE_SUPPORTED=y
CONFIG_SOC_CPU_CORES_NUM=1
CONFIG_SOC_CPU_INTR_NUM=32
CONFIG_SOC_CPU_HAS_FLEXIBLE_INTC=y
CONFIG_SOC_CPU_BREAKPOINTS_NUM=2
CONFIG_SOC_CPU_WATCHPOINTS_NUM=2
CONFIG_SOC_CPU_WATCHPOINT_SIZE=0x80000000
CONFIG_SOC_CPU_IDRAM_SPLIT_USING_PMP=y
CONFIG_SOC_GDMA_GROUPS=1
CONFIG_SOC_GDMA_PAIRS_PER_GROUP=1
CONFIG_SOC_GDMA_TX_RX_SHARE_INTERRUPT=y
CONFIG_SOC_GPIO_PORT=1
CONFIG_SOC_GPIO_PIN_COUNT=21
CONFIG_SOC_GPIO_SUPPORTS_RTC_INDEPENDENT=y
CONFIG_SOC_GPIO_SUPPORT_FORCE_HOLD=y
CONFIG_SOC_GPIO_SUPPORT_DEEPSLEEP_WAKEUP=y
CONFIG_SOC_GPIO_DEEP_SLEEP_WAKE_VALID_GPIO_MASK=0
CONFIG_SOC_GPIO_SUPPORT_SLP_SWITCH=y
CONFIG_SOC_DEDIC_GPIO_OUT_CHANNELS_NUM=8
CONFIG_SOC_DEDIC_GPIO_IN_CHANNELS_NUM=8
CONFIG_SOC_DEDIC_PERIPH_ALWAYS_ENABLE=y
CONFIG_SOC_I2C_NUM=1
CONFIG_SOC_I2C_FIFO_LEN=32
CONFIG_SOC_I2C_SUPPORT_HW_CLR_BUS=y
CONFIG_SOC_I2C_SUPPORT_XTAL=y
CONFIG_SOC_I2C_SUPPORT_RTC=y
CONFIG_SOC_LEDC_SUPPORT_PLL_DIV_CLOCK=y
CONFIG_SOC_LEDC_SUPPORT_XTAL_CLOCK=y
CONFIG_SOC_LEDC_CHANNEL_NUM=6
[... 1466 lines cut ...]
CONFIG_TCP_SND_BUF_DEFAULT=5744
CONFIG_TCP_WND_DEFAULT=5744
CONFIG_TCP_RECVMBOX_SIZE=6
CONFIG_TCP_QUEUE_OOSEQ=y
CONFIG_TCP_OVERSIZE_MSS=y
# CONFIG_TCP_OVERSIZE_QUARTER_MSS is not set
# CONFIG_TCP_OVERSIZE_DISABLE is not set
CONFIG_UDP_RECVMBOX_SIZE=6
CONFIG_TCPIP_TASK_STACK_SIZE=3072
CONFIG_TCPIP_TASK_AFFINITY_NO_AFFINITY=y
# CONFIG_TCPIP_TASK_AFFINITY_CPU0 is not set
CONFIG_TCPIP_TASK_AFFINITY=0x7FFFFFFF
# CONFIG_PPP_SUPPORT is not set
CONFIG_ESP32_PTHREAD_TASK_PRIO_DEFAULT=5
CONFIG_ESP32_PTHREAD_TASK_STACK_SIZE_DEFAULT=3072
CONFIG_ESP32_PTHREAD_STACK_MIN=768
CONFIG_ESP32_PTHREAD_TASK_CORE_DEFAULT=-1
CONFIG_ESP32_PTHREAD_TASK_NAME_DEFAULT="pthread"
CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ABORTS=y
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_FAILS is not set
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ALLOWED is not set
CONFIG_SUPPRESS_SELECT_DEBUG_OUTPUT=y
CONFIG_SUPPORT_TERMIOS=y
CONFIG_SEMIHOSTFS_MAX_MOUNT_POINTS=1
# End of deprecated options

RoshanESP, 2022-09-15:
Hi @lyh20190329, can you please share the sample application to reproduce this issue?

lyh20190329, 2022-09-16:
example like this:
[main.zip](https://github.com/espressif/esp-idf/files/9580215/main.zip)

lyh20190329, 2022-09-16:
When turning on bluetooth (calling start_blufi()) does nothing, then turning off bluetooth (calling sop_blufi()), everything is normal, but after bluetooth receives data, after connecting to wifi successfully and then turning off bluetooth, a crash occurs, I don't know if it is The reason for the wrong usage of the set_wifi_info_ble function or the reason for the destruction of the wifitask, thank you for your guidance

lyh20190329, 2022-09-16:
hi @RoshanESP ,thank you for your guidance,pls help me

RoshanESP, 2022-09-16:
Hi @lyh20190329, 

I think sequence you are trying to deinit is not working. Please find attached functions that you can use instead and achieve the intended functionality. 

I tested these changes on blufi app by modifying code and invoked the new API.  I am attaching the patch file for your reference. 

[blufi_patch.txt](https://github.com/espressif/esp-idf/files/9581780/blufi_patch.txt)

lyh20190329, 2022-09-17:
when i test this ,still crash,the BT_5.0_FEATURE=Y:
  blufi_security_deinit();
    esp_blufi_adv_start();
    esp_blufi_ble_stop();

log:
I (66982) BT: BLUFI ble disconnect

Guru Meditation Error: Core  0 panic'ed (Load access fault). Exception was unhandled.

Stack dump detected
Core  0 register dump:
MEPC    : 0x42020dca  RA      : 0x4201f0e4  SP      : 0x3fccbc00  GP      : 0x3fcaae40  
0x42020dca: ble_npl_mutex_pend at /home/ubuntu/work/esp-idf/components/bt/porting/npl/freertos/include/nimble/nimble_npl_os.h:223
 (inlined by) ble_hs_lock_nested at /home/ubuntu/work/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_hs.c:175

0x4201f0e4: ble_hs_id_copy_addr at /home/ubuntu/work/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_hs_id.c:263

TP      : 0x3fca9c2c  T0      : 0x4201f0d8  T1      : 0x403880d0  T2      : 0x7373227b  
0x4201f0d8: ble_hs_id_copy_addr at /home/ubuntu/work/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_hs_id.c:257

0x403880d0: gpspi_flash_ll_set_command at /home/ubuntu/work/esp-idf/components/hal/esp32c2/include/hal/gpspi_flash_ll.h:289
 (inlined by) spi_flash_hal_gpspi_configure_host_io_mode at /home/ubuntu/work/esp-idf/components/hal/spi_flash_hal_common.inc:118

S0/FP   : 0x3fcca830  S1      : 0x00000000  A0      : 0x3fcb2000  A1      : 0xffffffff  
A2      : 0x00000000  A3      : 0x00000013  A4      : 0x3fcb15b4  A5      : 0x00000000  
A6      : 0x00000080  A7      : 0x55514522  S2      : 0x3fcca850  S3      : 0x00000001  
S4      : 0x00000000  S5      : 0x00000000  S6      : 0x00000000  S7      : 0x00000000  
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000  
T3      : 0x70222c22  T4      : 0x6946696e  T5      : 0x555f5345  T6      : 0x55514522  
MSTATUS : 0x00001881  MTVEC   : 0x40380001  MCAUSE  : 0x00000005  MTVAL   : 0x00000044  
0x40380001: _vector_table at ??:?

lyh20190329, 2022-09-17:
@RoshanESP 
Can you test it on my example? Send custom data through bluetooth, the data contains ssid, passwd. The jason encapsulation of the name, such as: {"ssid":"EQUES","pwd":"EQUES123","user_name":"hgz123456"}, when the application receives this string, it will parse the WiFi data and connect to the WiFi hotspot, and then Turning off bluetooth will crash

RoshanESP, 2022-09-19:
Hi @lyh20190329 , 
I checked your code. Whenever wifi status is 3, the while loop will keep calling the code to stop ble. 

The api I gave yesterday assumed that it would be called only once. Attached is the updated patch where I have modified the stop_blufi function. 

`void stop_blufi()        
{
    if (!blufi_init)
        return;
    blufi_init = 0;
    blufi_ble_stop();
}
`

Please find the complete example attached. I have tested at my end and observed no crash. 
[working_main.zip](https://github.com/espressif/esp-idf/files/9597465/working_main.zip)

lyh20190329, 2022-09-19:
Thank you very much for your support. I can turn off Bluetooth according to your method. Now there is a need to turn it on after turning it off sometimes. I tested it. After turning it off, I directly call the following code to turn it on. like this:
`````````
nimble_port_init();    
    nimble_port_run();
````````````
The bluetooth can never be found. Whether I call it again .Is there a problem with the bluetooth turned on?

RoshanESP, 2022-09-19:
Hi @lyh20190329, 
If the intention is to restart after stopping, why are you not invoking start_blufi() function of your application? 

Your initial request was to stop Bluetooth. Now a new requirement is added. Based on your first request, I suggested the application level changes. If the intention is to restart then ensure that you are also calling esp_blufi_deinit() in the stop_blufi function.

lyh20190329, 2022-09-19:
@RoshanESP
Unfortunately, I re-tested several times, and turned off the bluetooth when the bluetooth was disconnected, or manually turned off the bluetooth when it was connected, and the bluetooth could not be turned off, and sometimes crashed.
```
  case ESP_BLUFI_EVENT_BLE_DISCONNECT:
        BLUFI_INFO("BLUFI ble disconnect\n");
        ble_is_connected = false;       
        blufi_security_deinit();
        esp_blufi_adv_start();
        esp_blufi_ble_stop();
        break;
```

```
void esp_blufi_ble_stop()
{
    int ret = nimble_port_stop();
     ESP_LOGI("BLUFI_EXAMPLE", "esp_blufi_ble_stop=%d-----",ret);
   if (ret == 0) {

       nimble_port_deinit();
   }
}
```

I used the code in the example of ESP latest release, used in the ESP32C2 chip, and the crystal oscillator is 26MHZ (Hardware Settings → Main XTAL Config->Main XTAL frequency (26 MHz) ),It seems that there is still a problem with closing the NIMBLE interface, what should I do?

lyh20190329, 2022-09-19:
> 
if the intention is to restart after stopping, why are you not invoking start_blufi() function of your application?
In the previous test, I tried to call start_blufi() to restart after turning off the bluetooth, but the mobile phone could not find the bluetooth device. From the current consumption, the bluetooth has been turned on. Now test it, the current does not change after turning off the bluetooth, and sometimes it crashes

RoshanESP, 2022-09-19:
Hi @lyh20190329, 

In this case, you are starting advertisement and then attempting to stop ble. 

  case ESP_BLUFI_EVENT_BLE_DISCONNECT:
        BLUFI_INFO("BLUFI ble disconnect\n");
        ble_is_connected = false;       
        blufi_security_deinit();
        esp_blufi_adv_start();
        esp_blufi_ble_stop();
        break;
        
        I suggest referring to documentation for esp apis which will assist you to understand what the API does. 
        
        At my end, I tried to stop and start Bluetooth/blufi and find its working. Please try the attached patch. These are mostly application-level changes. 
[app_changes.txt](https://github.com/espressif/esp-idf/files/9598883/app_changes.txt)

lyh20190329, 2022-09-19:
[blufi_example.zip](https://github.com/espressif/esp-idf/files/9599043/blufi_example.zip)
This is what I debugged on the latest official website example. IDF is also the latest one. I modified XTAL to 26MHZ. After receiving the first character 0x30 of the custom data, turn off the Bluetooth. The test found that the current will not decrease, and the Bluetooth is disconnected. Can no longer be scanned, sometimes crashes

RoshanESP, 2022-09-20:
I made the changes in your example to stop blufi when the first character of custom data is received and then started blufi again with the same steps done for the first time. I found this working properly. Below is the working_patch.txt file for your reference.
[working_patch.txt](https://github.com/espressif/esp-idf/files/9604508/working_patch.txt)

lyh20190329, 2022-09-20:
Hi @RoshanESP
I really appreciate your support, I don't know what to do without your support. 
I have integrated your patch and did the following tests: 
test 1. Do nothing, only turn on and turn off Bluetooth:
it works fine, but there will be memory leaks, each time it leaks about 5k.
Keep turning on and off bluetooth, and finally there is no memory available!

 test2. I did a process test, fist time test is ok,but do this again,seem some broken(ESP not sleep,not power off),Found that it was normal to turn on Bluetooth for the second time to receive data. When it was turned off, the log looked normal. When WIFI was connected to the Internet or was about to be connected to the Internet, a crash occurred: 
The test process is as follows: Connect Bluetooth -> Receive WiFi configuration data -> turn off bluetooth -> connect to wifi -> connect to server (websokect connect))-> start BT->Connect Bluetooth -> Receive WiFi configuration data -> turn off bluetooth -> start wifi -> connect to server (websokect connect))(CRASH )

CRASH LOGS:
Guru Meditation Error: Core  0 panic'ed (Load access fault). Exception was unhandled.

Stack dump detected
Core  0 register dump:
MEPC    : 0x40057ff4  RA      : 0x40058800  SP      : 0x3fcc9540  GP      : 0x3fcacf10  
TP      : 0x3fca6d18  T0      : 0x40057e12  T1      : 0xffffffe0  T2      : 0xffffffff  
S0/FP   : 0x3fcb3c14  S1      : 0x3fccae7c  A0      : 0x3fcb3c14  A1      : 0x3fccae84  
A2      : 0x3fccae8c  A3      : 0x00000004  A4      : 0x317a676a  A5      : 0x3fccae7c  
A6      : 0x00000008  A7      : 0xf4240000  S2      : 0x00000000  S3      : 0x3fccae7d  
S4      : 0x3fccae84  S5      : 0x7f995cfc  S6      : 0x00000000  S7      : 0x00000000  
S8      : 0x3fcc38bc  S9      : 0x00000000  S10     : 0x3c09d000  S11     : 0x00000100  
T3      : 0x00000edf  T4      : 0x0000003f  T5      : 0x3338315f  T6      : 0x6c376875  
MSTATUS : 0x00001881  MTVEC   : 0x40380001  MCAUSE  : 0x00000005  MTVAL   : 0x7f995d00  
0x40380001: _vector_table at ??:?
MHARTID : 0x00000000  
Backtrace:

0x40057ff4 in ?? ()
#0  0x40057ff4 in ?? ()
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
ELF file SHA256: 614b8f49c2ad03b6

CPU halted.

The interface called to close Bluetooth is as follows:
```
 esp_blufi_deinit();
 esp_blufi_ble_stop();
```

lyh20190329, 2022-09-21:
Hi @RoshanESP
After continuous testing and tracking, the problem of crash has been solved. 
Now it's the memory leak that needs your support. 
Don't do anything, keep turning on and off bluetooth, use **esp_get_free_heap_size ()**to check the remaining memory, it will get smaller and smaller until the bluetooth fails to turn on.

RoshanESP, 2022-09-22:
Hi @lyh20190329 ,

Thanks for pointing out. Can you please try the attached changes. Except for the first instance where some memory gets reserved, the next attempts for init / deinit should not cause any memory leak. Can you please try attached patch and share your observation. Please put this patch on top of all existing changes. 
[patch.txt](https://github.com/espressif/esp-idf/files/9626086/patch.txt)
At my end I modified application to do init / deinit in a loop and now see constant heap numbers.

lyh20190329, 2022-09-26:
Hi @RoshanESP
Thanks for support,I try the attached changes and working fine. no memory leak.

rahult-github, 2022-10-14:
This [commit](aa03ffdc1beb1b650453f3adfe4a89aab2e24c82) address the issue discussed here.

closed 2022-10-14 by rahult-github