espressif/esp-idf #9793: ISR safe mcpwm_set_duty or mcpwm_comparator_set_compare_value

usbalbin, 2022-09-15:
### Is your feature request related to a problem?

I have an application where I am producing a sine wave by modulating a PWM signal, think something like [this](https://en.wikipedia.org/wiki/Pulse-width_modulation#/media/File:PWM,_3-level.svg). As far as I understand something like the RMT peripheral is not viable since I need feedback to regulate the output.

I have been using MCPWM for generating PWM(the IDF 4.4 version).

For changing duty at regular intervals what I have tried so far is:
* Busy looping - Inconsistent timing
* High Resolution Timer - causes irregularity once every about 205ms, I believe because of some Wifi stuff
* General Purpose Timer - Excelent consitency however, executes code in ISR context, [MCPWM functions are not safe in this context](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/mcpwm.html#thread-safety)(?)

### Describe the solution you'd like.

Just like `adc_oneshot_read_isr` add something similar to mcpwm to allow updating the duty/compare value from ISR context.

### Describe alternatives you've considered.

See above

### Additional context.

I am happy to try implementing this if you think it would be a good idea. I am open to other suggestions as well.

usbalbin, 2022-09-15:
Judging by the call to `portENTER_CRITICAL_SAFE` in [mcpwm_comparator_set_compare_value](https://github.com/espressif/esp-idf/blob/master/components/driver/mcpwm/mcpwm_cmpr.c#L134), is this already ISR safe?

suda-morris, 2022-09-15:
Hi @usbalbin I'm afraid the `mcpwm_comparator_set_compare_value` can't be used in the interrupt context without replacing the `ESP_RETURN_ON_FALSE` into `ESP_GOTO_ON_FALSE_ISR` and place that function into IRAM. However, even we have made that function to be IRAM safe, it still has overhead that we don't like to (too many parameter check)
How do you like it if we allow the user to update the compare value in the `mcpwm_compare_event_cb_t` callback function? Simply by having a `new_compare_val` member in the `mcpwm_compare_event_data_t`.
Then the callback function would look like:

```c
static bool test_compare_on_reach(mcpwm_cmpr_handle_t cmpr, mcpwm_compare_event_data_t *ev_data, void *user_data)
{
    // update to new compare value
   ev_data->new_compare_val = xxx;
    return false;
}
```

usbalbin, 2022-09-15:
> Hi @usbalbin I'm afraid the mcpwm_comparator_set_compare_value can't be used in the interrupt context without replacing the ESP_RETURN_ON_FALSE into ESP_GOTO_ON_FALSE_ISR and place that function into IRAM.[...]

Oh.

> [...] How do you like it if we allow the user to update the compare value in the mcpwm_compare_event_cb_t callback function? Simply by having a new_compare_val member in the mcpwm_compare_event_data_t. [...]

In my use case the PWM will run at atleast 20kHz. When setting the duty I have to run PID and some other basic math to calculate the duty (~40us from my measurements can probably optimize it a bit lower) and an AD read (~50us in my test) so roughly about 90us in total. I believe doing that at 20kHz (50us) might be cutting it a bit tight unless I am missing something... At the same time I would like to try and keep updating the duty somewhere very close to every 100us.

Or am I aproaching this the wrong way?

Btw I am using a ESP32 so I do have access to two cores and dedicating Core1 to this would be fine with me.

usbalbin, 2022-09-15:
Even if this might perhaps not be the case for the majority of users, I intend on using this from the rust language and will most likely try to put this function behind some thin abstraction that using the type system(think something like [esp-idf-hal](https://github.com/esp-rs/esp-idf-hal)) will ensure the user is unable to pass bad paramters. So I would be fine with less checks at run time.

usbalbin, 2022-09-16:
I could probably go down to only updating the duty every 200us in case that would make things easier. The most important thing is that there are no interruptions as large as several (duty setting)periods.

closed 2022-09-23 by espressif-bot

usbalbin, 2022-09-23:
Thank you a lot!