espressif/esp-idf #9796: The bluetooth scanning effect of esp32s3 is very poor compared to esp32

ericnts, 2022-09-16:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.2

### Operating System used.

macOS

### How did you build your project?

VS Code IDE

### Development Kit.

ESP32-S3-Korvo-2

### Power Supply used.

Battery

### What is the expected behavior?

esp32s3 and esp32 can have the same bluetooth broadcast scanning performance

### What is the actual behavior?

Bluetooth broadcast scan of esp32s3 is poor

### Steps to reproduce.

1. Prepare a device that can send out Beacon broadcasts at a fixed frequency
2. Refer to the example (esp-idf/examples/bluetooth/bluedroid/ble/gatt_client/) to write the code (the final code is in the attachment)
3. Compile two codes and run them on ESP32 and ESP32S3 respectively
4. Every time a Beacon broadcast data is received, it will be printed in the log.
5. Looking at the logs, I found that the ESP32S3 lost a lot more data than the ESP32.

[image]
ESP32 on the left, ESP32S3 on the right

### More Information.

`/*
   This example code is in the Public Domain (or CC0 licensed, at your option.)

   Unless required by applicable law or agreed to in writing, this
   software is distributed on an "AS IS" BASIS, WITHOUT WARRANTIES OR
   CONDITIONS OF ANY KIND, either express or implied.
*/

/****************************************************************************
*
* This demo showcases BLE GATT client. It can scan BLE devices and connect to one device.
* Run the gatt_server demo, the client demo will automatically connect to the gatt_server demo.
* Client demo will enable gatt_server's notify after connection. The two devices will then exchange
* data.
*
****************************************************************************/

#include <stdint.h>
#include <string.h>
#include <stdbool.h>
#include <stdio.h>
#include "nvs.h"
#include "nvs_flash.h"
[... 566 lines cut ...]
    if(ret){
        ESP_LOGE(GATTC_TAG, "%s gattc register failed, error code = %x\n", __func__, ret);
        return 0;
    }

    ret = esp_ble_gattc_app_register(PROFILE_A_APP_ID);
    if (ret){
        ESP_LOGE(GATTC_TAG, "%s gattc app register failed, error code = %x\n", __func__, ret);
    }
    esp_err_t local_mtu_ret = esp_ble_gatt_set_local_mtu(500);
    if (local_mtu_ret){
        ESP_LOGE(GATTC_TAG, "set local  MTU failed, error code = %x", local_mtu_ret);
    }

    return 1;

}

void app_main(void){
    gattc_app_main();

    xTaskCreate(gattc_task, "gattcTask", 2048, NULL, 8, NULL);
}

`

Weijian-Espressif, 2022-09-21:
@ericnts  Please provide the esp32 and S3 sdkconfig files you used for testing

ericnts, 2022-09-21:
@Weijian-Espressif This is all the files [https://github.com/ericnts/esp-gatt](https://github.com/ericnts/esp-gatt)

ericnts, 2022-09-21:
@Weijian-Espressif IDF commit is 1b16ef6c

Weijian-Espressif, 2022-09-21:
@ericnts thanks, If this  issue is located, I will notify you as soon as possible

Weijian-Espressif, 2022-09-23:
[diff.patch.zip](https://github.com/espressif/esp-idf/files/9632307/diff.patch.zip)
@ericnts Please use this patch file and retest

Weijian-Espressif, 2022-09-29:
@ericnts Is there any update?

atwoz, 2022-11-05:
@ericnts were you able to test the patch?

Weijian-Espressif, 2023-02-16:
@atwoz  We have fixed this issue in 2022, and have not waited for your reply.
please test the latest IDF, if you want get more adv rsp, please disable scanbackoff in controller menuconfig.

Alvin1Zhang, 2023-02-16:
To be more specific, the fix commit on release/4.4 is https://github.com/espressif/esp-idf/commit/da943339adc84e131c90a87fde823c24ad4e6a69, feel free to reopen with more updates.

closed 2023-02-16 by Alvin1Zhang